Screenly/Anthias #1277: Automatically load video if no connection instead of going to AP mode

sipsontm, 2019-12-10:
I just tested the Screenly OSE build and i was so impressed, it really works great. Congrats to all that put an effort to it.
We are building an Xmass display stand that has a fireplace. We've built the fireplace and put an lcd display (TV) in it that need to play a short fire burning video. - this helped a lot: 

https://www.screenly.io/blog/2018/12/22/virtual-fireplace-raspberry-pi/

[image]
[image]
[image]

On the way i came across the following issue, that i managed to fix:

- We needed to play the video on loop mode so that we don't get the black screen between preloading of video - i found a fix here and it works great: https://forums.screenly.io/t/looping-video-for-a-virtual-fireplace/145/11

**### We will install this in a location with no internet access.
When i boot PI with no internet connection it will go into AP mode and will not display the video.
Is there a fix for this to be able to boot and load the video automatically without internet connection?**

Sincerely,

Ciprian

vpetersson, 2019-12-11:
> When i boot PI with no internet connection it will go into AP mode and will not display the video.
Is there a fix for this to be able to boot and load the video automatically without internet connection?

This is expected. It's because if the Pi has no internet connection, it doesn't know what time it is. If it doesn't know what time it is, it doesn't know what to play.  The technical reason for this is because the Raspberry Pi lacks a Real Time Clock (RTC). This is why Screenly requires internet (or at least LAN) access in order to properly function.

vpetersson, 2019-12-11:
But also - great project! Would love to see some more photos.

closed 2019-12-11 by vpetersson

sipsontm, 2019-12-11:
vpetersson, thank you for your quick reply, it all makes sense now. Apparently, we now have to use 2 displays in the posm stand (one for the fireplace and one to display a video ad). I will be using 2 x RPI Zero W. I already got a portable wifi hotspot 4G from Huawei that we're going to use. Hope it all goes well. It will be displayed in Carrefour Hypermarket in Timisoara/Romania.

[image]
[image]
[image]

djmtc, 2019-12-12:
Hi there @sipsontm! Daniel here with Screenly. I'm the guy who wrote the original article that you found helpful :) We'd love to do an case study on you deployment and get you more exposure with our audience. Can you reach out to dmountcastle@screenly.io and we can talk more about this? Happy holidays!!