Screenly/Anthias #1280: readme link dead

xhozt, 2019-12-25:
link to faq in Readme is dead https://support.screenly.io/hc/en-us/sections/202652366-Frequently-Asked-Questions-FAQ-

vpetersson, 2019-12-28:
This is a known issue. We're overhauling the FAQ section.

jeff-knurek, 2019-12-30:
the links to the 
* Live Demo
* QA Checklist
* API Docs

are also "dead"

vpetersson, 2020-01-13:
* The Live Demo is a known issue. Basically the refactoring to the Docker model broke it (#1251).
* The QA Checklist appears to be broken due to the service used [Forgett](https://www.forgett.com/) has either been shut down or they have forgot to fix their SSL certificate
* API docs is a function of the demo site. It's easy to work around this, as the API docs is also exposed on any local instance of Screenly OSE.

jeff-knurek, 2020-03-13:
* it seems like the FAQ link should be updated to: https://support.screenly.io/hc/en-us/categories/360002614653-FAQs
* yeah, QA Checklist location is obsolete - not sure if there's an alternative: https://twitter.com/rbstrachan/status/1136747590702252032/photo/1
* it looks like the API docs should point here: https://developer.screenlyapp.com/

Eric-Screenly, 2020-03-24:
Hi @jeff-knurek , the links that you posted directly above for the FAQ and API docs are actually for our commercial (Screenly Pro) subscription platform, and not related to OSE. I just wanted to be clear so there wasn't any confusion here.

Thank you!

closed 2020-05-30 by stale[bot]