Screenly/Anthias #1282: Balena install error RP4V

justinkruit, 2019-12-31:
### Overview of the Issue

When I'm trying to install Screenly on my Raspberry PI 4B 2gb it always ends with an error. I've tried both a release and the latest commits, but it still fails. 

Build Overview:
```
[Info]                Building on arm03
[Info]                Pulling previous images for caching purposes...
[Success]             Successfully pulled cache images
[Info]                Starting to build screenly-viewer
[Info]                Starting to build screenly-server
[Info]                Starting to build screenly-celery
[Info]                Starting to build redis
[Info]                Starting to build screenly-websocket
[Success]             Successfully built service redis
[Success]             Successfully built service screenly-websocket
[Info]                Uploading images
[Success]             Successfully uploaded images
[Error]               Some services failed to build:
[Error]                 Service: screenly-server
[Error]                   Error: The command '/bin/sh -c apt-get update &&     apt-get -y install         build-essential         curl         git-core         libffi-dev         libssl-dev         lsb-release         net-tools         nginx-light         omxplayer         psmisc         python-dev         python-gobject         python-imaging         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3 &&     apt-get clean' returned a non-zero code: 100
[Error]                 Service: screenly-viewer
[Error]                   Error: The command '/bin/sh -c apt-get update &&     apt-get -y install         build-essential         curl         git-core         libffi-dev         libssl-dev         matchbox         net-tools         omxplayer         psmisc         python-dev         python-imaging         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3         uzbl         x11-xserver-utils         xserver-xorg &&     apt-get clean' returned a non-zero code: 100
[Error]                 Service: screenly-celery
[Error]                   Error: The command '/bin/sh -c apt-get update &&     apt-get -y install         build-essential         curl         git-core         libffi-dev         libssl-dev         net-tools         omxplayer         psmisc         python-dev         python-gobject         python-imaging         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3 &&     apt-get clean' returned a non-zero code: 100
[Info]                Built on arm03
[Error]               Not deploying release.
```

Error from screenly-server (others have basically the same error)
```
Step 1/8 : FROM balenalib/raspberrypi4-64-debian:jessie
 ---> cd499f545648
Step 2/8 : RUN apt-get update &&     apt-get -y install         build-essential         curl         git-core         libffi-dev         libssl-dev         lsb-release         net-tools         nginx-light         omxplayer         psmisc         python-dev         python-gobject         python-imaging         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3 &&     apt-get clean
 ---> Running in 4b18dcaf6823
Ign http://archive.debian.org jessie InRelease
Get:1 http://archive.debian.org jessie Release.gpg [2420 B]
Get:2 http://archive.debian.org jessie Release [148 kB]
Get:3 http://archive.debian.org jessie/main arm64 Packages [8593 kB]
Fetched 8744 kB in 8s (1043 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package omxplayer
E: Unable to locate package libraspberrypi0

Removing intermediate container 4b18dcaf6823
The command '/bin/sh -c apt-get update &&     apt-get -y install         build-essential         curl         git-core         libffi-dev         libssl-dev         lsb-release         net-tools         nginx-light         omxplayer         psmisc         python-dev         python-gobject         python-imaging         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3 &&     apt-get clean' returned a non-zero code: 100

```

#### Reproduction Steps

Follow the instructions of https://www.balena.io/blog/deploy-free-digital-signage-software-screenly-ose/
Getting the error on step 1.7

### Environment

* Raspberry Pi Hardware Version: 4b 2gb ram
* Screenly OSE Version: 0.18.2 and latest of Master

vpetersson, 2020-01-02:
Interesting. Looks like the `omxplayer` package is missing in that Docker image.  @rusko124 could you look into this when you have a moment?

EpicKris, 2020-01-04:
Receiving the same error.

rusko124, 2020-01-10:
Rpi4 support is the main goal of this sprint.
I will work on this issue after #1239 merging.

But you can try edit `docker/Dockerfile.*.template` dockerfiles.
For example what you can do:
1. Delete libraspberrypi0.
2. Build UZBL in Dockerfile.viewer.template
```
RUN apt -y update && apt -y install \
      make \
      pkg-config \
      gcc \
      libwebkit2gtk-4.0-dev \
      python3-setuptools \
      python3-six \
      libgnutls28-dev \
      python3-wheel \
      python3-pip

RUN git clone git://github.com/uzbl/uzbl.git /uzbl && \
      cd /uzbl && \
      git checkout origin/next && \
      make && \
      make install
```
Code above is just some thoughts about  building.

But I recommend choosing RPi3 or lower for now. Until the issue is resolved.

tariqinaam, 2020-01-16:
I tried above suggested solution, but i get diff error now
`Package webkit2gtk-4.0 was not found in the pkg-config search path.`

I am not sure if I need to replace the existing apt command from Dockerfile.viewer.template
here is my Dockerfile.viewer.template

`FROM balenalib/%%BALENA_MACHINE_NAME%%-debian:jessie

RUN apt -y update && apt -y install \
      make \
      pkg-config \
      gcc \
      libwebkit2gtk-3.0-dev \
      python3-setuptools \
      python3-six \
      libgnutls28-dev \
      python3-wheel \
      python3-pip \
      git

RUN git clone git://github.com/uzbl/uzbl.git /uzbl && \
      cd /uzbl && \
      git checkout origin/next && \
      make && \
      sudo make install

RUN apt-get update && \
    apt-get -y install \
        build-essential \
        curl \
        git-core \
        libffi-dev \
        libssl-dev \
        matchbox \
        net-tools \
        omxplayer \
        psmisc \
        python-dev \
        python-imaging \
        python-netifaces \
        python-simplejson \
        ifupdown \
        sqlite3 \
        uzbl \
        x11-xserver-utils \
        xserver-xorg && \
    apt-get clean

# Install Python requirements
ADD requirements/requirements.txt /tmp/requirements.txt
RUN curl -s https://bootstrap.pypa.io/get-pip.py | python && \
    pip install --upgrade -r /tmp/requirements.txt

COPY ansible/roles/screenly/files/gtkrc-2.0 /data/.gtkrc-2.0

COPY . /tmp/screenly

CMD ["bash", "chmod 777 /dev/vchiq"]
CMD ["bash", "/tmp/screenly/bin/start_balena.sh"]
`

> But I recommend choosing RPi3 or lower for now. Until the issue is resolved.
are you suggesting to use RP3 hardware?

vpetersson, 2020-01-17:
@rusko124 can you take a look at this?

vpetersson, 2020-01-23:
@rusko124 just FYI from Balena:

> WRT the Pi 4 issue, in lieu of spending time to figure out why those containers don't build on 64bit, in order to retain the generic `Dockerfile.template` using `%%BALENA_MACHINE_NAME%%` we could add another `Dockerfile.raspberrypi4-64` or even `Dockerfile.aarch64` to be a bit more generic, and force that one to use `raspberrypi3` base images.

jallen2281, 2020-02-18:
Hello, Same issue here as well. Has there been any progress on this? It's over a month old now.

kn1gh7h4wk, 2020-02-28:
Any progress made on this issue?

netweaver1970, 2020-03-10:
Any news on this issue?

vpetersson, 2020-03-10:
We've had to reallocate some resources to the commercial version, but are looking to bring on a new developer in the near future.

geokscott, 2020-05-26:
Just ordered 4 Pi 4's , hot to trot on getting a Pi 4 build :)

justinkruit, 2020-05-30:
@geokscott I think Screenly still doesn't work on the Pi 4, so you'll have to be patient unfortunately

geokscott, 2020-05-30:
As an alternative for now, I'm going to try using balena-dash for displaying a URL.
I believe it works on the Pi 4 from what I've read.

EpicKris, 2020-07-17:
Just attempted again using the latest release, same issue.

roinone, 2020-07-22:
Confirmed the same issue here with the latest Sprint 10.1 release, v0.18.3

vpetersson, 2020-07-27:
The latest release did unfortunately not address Balena. We will try to look at that in a not too distant future.

elitegoodguy, 2020-08-07:
I have 10 Pi 4's coming.  I didn't realize this issue existed until I was trying to load it up on a Pi 3 with 64bit OS.

francovico, 2020-08-12:
Same problem here! Any solution?
I'm on raspberrypi4-64-2.53.12+rev1-dev-v11.9.9.img

gazzer82, 2020-08-24:
https://forums.balena.io/t/screenly-ose-on-balenacloud/5853/29

> As a temporary workaround I would replace %%BALENA_MACHINE_NAME%% with raspberrypi3 in the docker/Dockerfile.celery.template, 
> docker/Dockerfile.redis.template, 
> docker/Dockerfile.server.template, 
> docker/Dockerfile.viewer.template 
> docker/Dockerfile.websocket.template

javad87, 2020-09-10:
Anyone can port screenly on Raspberry pi 4 successfully? 

I had the same issue, I tried writing OSE image from sprint 10.1 with Etcher and splash screen did not show up

then tried option 2 with NOOBS installation and there was no Option available on installation screen for screenly...

then I tried really DIY optine 3 and installed raspbian os full latest release and then with the command run bash and their script (bash <(curl -sL https://www.screenly.io/install-ose.sh)), tried answering questions with different  NO and Yes combination installation and it did not work either...

I really need to have screenly on Rpi4, any suggestion what should I do?
When the fix will be available? kindly ask contributor to help us, otherwise I have no choice except finding another solution and I think there is no better free on-premise solution better than screenly 

During package installation I get this Error...
[image]

Hardware: 64 GB microSDcard
OS: 2020-08-20-raspios-buster-armhf-full
Screenly: sprint 10.1---2020-07-17-Screenly-OSE.zip

I also checked check-sum of the downloaded file, it was not corrupt.

ealmonte32, 2020-10-02:
Anyone having issues installing on RPi4 pls see my comment with instructions here:
https://github.com/Screenly/screenly-ose/issues/1355#issuecomment-696388611

Remember to select master/dev branch.

ealmonte32, 2020-10-02:
@justinkruit 
wondering if you could git clone my fork and give it a try with balena, I just created a buster docker-compose and made other changes to test buster.. if you try and report back that would help because i am testing with pi zero w and need someone with rpi4 to test..

justinkruit, 2020-10-02:
@ealmonte32 when I first opened this issue 10 months ago I was testing with the rpi4. When I found out it wasn't going to be soon that it'll support it, I returned it and got the 3. So unfortunately, I'm not able to test anything. Although I've decided to keep the issue open as I still see this as a valid issue that still needs to be addressed.

ealmonte32, 2020-10-02:
> @ealmonte32 when I first opened this issue 10 months ago I was testing with the rpi4. When I found out it wasn't going to be soon that it'll support it, I returned it and got the 3. So [...]

I see.. well, I guess we can hopefully have someone else try it, although I already have it working on non-balena, I just wanted to make sure that it also worked in balena with pi4..

ealmonte32, 2020-10-03:
I actually tried it myself in a pi zero w and its not fully working so its fine, i think i need to get it working there first before needing rpi4 testing.

EpicKris, 2020-10-03:
`ealmonte32/screenly-ose` `c14e03f` remote build failed.

```zsh
% balena push Screenly-4
[Info]                Starting build for Screenly-4, user hq
[Info]                Dashboard link: https://dashboard.balena-cloud.com/apps/1575170/devices
[Info]                Building on arm03
[Info]                Pulling previous images for caching purposes...
[Success]             Successfully pulled cache images
[screenly-viewer]     Step 1/10 : FROM balenalib/raspberrypi4-64-debian:buster-run
[screenly-celery]     Step 1/8 : FROM balenalib/raspberrypi4-64-debian:buster-run
[redis]               Step 1/4 : FROM balenalib/raspberrypi4-64-debian:buster-run
[screenly-server]     Step 1/10 : FROM balenalib/raspberrypi4-64-debian:buster-run
[screenly-viewer]      ---> bd9f20472954
[screenly-viewer]     Step 2/10 : RUN apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         git-core         libffi-dev         libssl-dev         matchbox         net-tools         omxplayer         psmisc         python-dev         python-pip-whl         python-pip         python-pil         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3         uzbl         x11-xserver-utils         xserver-xorg
[screenly-celery]      ---> bd9f20472954
[screenly-celery]     Step 2/8 : RUN apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         git-core         libffi-dev         libssl-dev         net-tools         omxplayer         psmisc         python-dev         python-pip-whl         python-pip         python-pil         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3
[screenly-server]      ---> bd9f20472954
[screenly-server]     Step 2/10 : RUN apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         git-core         libffi-dev         libssl-dev         lsb-release         net-tools         nginx-light         omxplayer         psmisc         python-dev         python-pip-whl         python-pip         python-pil         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3
[redis]                ---> bd9f20472954
[redis]               Step 2/4 : RUN apt-get update && apt-get install -y redis-server
[screenly-viewer]      ---> Running in 07c320534b67
[screenly-celery]      ---> Running in 899730f7805d
[screenly-server]      ---> Running in 9816c1f4d753
[redis]                ---> Running in c3548912da07
[screenly-websocket]  Step 1/7 : FROM balenalib/raspberrypi4-64-debian:buster-run
[screenly-websocket]   ---> bd9f20472954
[screenly-websocket]  Step 2/7 : RUN apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         python-dev         python-pip-whl         python-pip         libffi-dev         libssl-dev         whois
[screenly-websocket]   ---> Running in 4d3ae7f3adc5
[screenly-viewer]     Here are a few details about this Docker image (For more information please visit https://www.balena.io/docs/reference/base-images/base-images/): 
[screenly-viewer]     Architecture: ARM v8 
[screenly-viewer]     OS: Debian Buster 
[screenly-viewer]     Variant: run variant 
[screenly-viewer]     Default variable(s): UDEV=off 
[screenly-viewer]     Extra features: 
[screenly-viewer]     - Easy way to install packages with `install_packages <package-name>` command 
[screenly-viewer]     - Run anywhere with cross-build feature  (for ARM only) 
[screenly-viewer]     - Keep the container idling with `balena-idle` command 
[screenly-viewer]     - Show base image details with `balena-info` command
[screenly-viewer]     Get:1 http://security.debian.org/debian-security buster/updates InRelease [65.4 kB]
[redis]               Here are a few details about this Docker image (For more information please visit https://www.balena.io/docs/reference/base-images/base-images/): 
[redis]               Architecture: ARM v8 
[redis]               OS: Debian Buster 
[redis]               Variant: run variant 
[redis]               Default variable(s): UDEV=off 
[redis]               Extra features: 
[redis]               - Easy way to install packages with `install_packages <package-name>` command 
[redis]               - Run anywhere with cross-build feature  (for ARM only) 
[redis]               - Keep the container idling with `balena-idle` command 
[redis]               - Show base image details with `balena-info` command
[screenly-server]     Here are a few details about this Docker image (For more information please visit https://www.balena.io/docs/reference/base-images/base-images/): 
[screenly-server]     Architecture: ARM v8 
[screenly-server]     OS: Debian Buster 
[screenly-server]     Variant: run variant 
[screenly-server]     Default variable(s): UDEV=off 
[screenly-server]     Extra features: 
[screenly-server]     - Easy way to install packages with `install_packages <package-name>` command 
[screenly-server]     - Run anywhere with cross-build feature  (for ARM only) 
[screenly-server]     - Keep the container idling with `balena-idle` command 
[screenly-server]     - Show base image details with `balena-info` command
[redis]               Get:1 http://security.debian.org/debian-security buster/updates InRelease [65.4 kB]
[... cut ...]
[Error]                   Error: The command '/bin/sh -c apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         git-core         libffi-dev         libssl-dev         net-tools         omxplayer         psmisc         python-dev         python-pip-whl         python-pip         python-pil         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3' returned a non-zero code: 100
[Info]                Built on arm03
[Error]               Not deploying release.
Remote build failed
```

ealmonte32, 2020-10-03:
@EpicKris 
Thanks for the logs, yeah I was in the middle of constant testing so you probably pulled a failed release.
I am still testing and waiting for the latest successful image to load onto the pi zero w which happens at snail speed.

[image]

ealmonte32, 2020-10-03:
@EpicKris & others..
I just finished testing, that release c27347b works on rpi3+ and below.
So now I just need some users with rpi4 to test this fork: https://github.com/ealmonte32/screenly-ose
and report back.

EpicKris, 2020-10-04:
`ealmonte32/screenly-ose` `dd4ba8f` remote build failed.

```zsh
screenly-ose-master % balena push Screenly-4 
[Info]                Starting build for Screenly-4, user hq
[Info]                Dashboard link: https://dashboard.balena-cloud.com/apps/1575170/devices
[Info]                Building on arm01
[Info]                Pulling previous images for caching purposes...
[Success]             Successfully pulled cache images
[screenly-websocket]  Step 1/8 : FROM balenalib/raspberrypi4-64-debian:buster-run
[screenly-viewer]     Step 1/11 : FROM balenalib/raspberrypi4-64-debian:buster-run
[redis]               Step 1/6 : FROM balenalib/raspberrypi4-64-debian:buster-run
[screenly-websocket]   ---> bd9f20472954
[screenly-websocket]  Step 2/8 : RUN apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         python2.7         python2.7-dev         python-pip-whl         python-wheel         python-pip         libffi-dev         libssl-dev     	&& apt-get clean && rm -rf /var/lib/apt/*
[screenly-viewer]      ---> bd9f20472954
[screenly-viewer]     Step 2/11 : RUN apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         git-core         libffi-dev         libssl-dev         libcairo2-dev         libgirepository1.0-dev         matchbox         net-tools         omxplayer         psmisc         python-dev     	python-gobject         python-gi         python2.7         python2.7-dev         python-pip-whl         python-pip         python-pil         python-wheel         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3         uzbl         x11-xserver-utils         xserver-xorg     	&& apt-get clean && rm -rf /var/lib/apt/*
[screenly-websocket]   ---> Running in 35ed6cf6f33f
[redis]                ---> bd9f20472954
[redis]               Step 2/6 : RUN apt-get update && apt-get install -y redis-server 	&& apt-get clean && rm -rf /var/lib/apt/*
[screenly-celery]     Step 1/9 : FROM balenalib/raspberrypi4-64-debian:buster-run
[screenly-viewer]      ---> Running in ed2c4c5e26af
[redis]                ---> Running in 9003439fb499
[screenly-server]     Step 1/11 : FROM balenalib/raspberrypi4-64-debian:buster-run
[screenly-celery]      ---> bd9f20472954
[screenly-celery]     Step 2/9 : RUN apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         git-core         libffi-dev         libssl-dev         libgirepository1.0-dev         libcairo2-dev         python-gobject         python-gi         python-dev         python2.7         python2.7-dev         python-pip-whl         python-wheel         python-pip     	&& apt-get clean && rm -rf /var/lib/apt/*
[screenly-celery]      ---> Running in c7244987d8c9
[screenly-server]      ---> bd9f20472954
[screenly-server]     Step 2/11 : RUN apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         git-core         libffi-dev         libssl-dev         libcairo2-dev         libgirepository1.0-dev         lsb-release         net-tools         nginx-light         omxplayer         psmisc         python-dev         python-gobject         python-gi         python2.7         python2.7-dev         python-pip-whl         python-wheel         python-pip         python-pil         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3    && apt-get clean && rm -rf /var/lib/apt/*
[screenly-server]      ---> Running in 740cc3035b70
[screenly-websocket]  Here are a few details about this Docker image (For more information please visit https://www.balena.io/docs/reference/base-images/base-images/): 
[screenly-websocket]  Architecture: ARM v8 
[screenly-websocket]  OS: Debian Buster 
[screenly-websocket]  Variant: run variant 
[screenly-websocket]  Default variable(s): UDEV=off 
[screenly-websocket]  Extra features: 
[screenly-websocket]  - Easy way to install packages with `install_packages <package-name>` command 
[screenly-websocket]  - Run anywhere with cross-build feature  (for ARM only) 
[screenly-websocket]  - Keep the container idling with `balena-idle` command 
[screenly-websocket]  - Show base image details with `balena-info` command
[screenly-websocket]  Get:1 http://deb.debian.org/debian buster InRelease [121 kB]
[screenly-websocket]  Get:2 http://security.debian.org/debian-security buster/updates InRelease [65.4 kB]
[screenly-websocket]  Get:3 http://deb.debian.org/debian buster-updates InRelease [51.9 kB]
[screenly-viewer]     Here are a few details about this Docker image (For more information please visit https://www.balena.io/docs/reference/base-images/base-images/): 
[screenly-viewer]     Architecture: ARM v8 
[screenly-viewer]     OS: Debian Buster 
[screenly-viewer]     Variant: run variant 
[screenly-viewer]     Default variable(s): UDEV=off 
[screenly-viewer]     Extra features: 
[screenly-viewer]     - Easy way to install packages with `install_packages <package-name>` command 
[screenly-viewer]     - Run anywhere with cross-build feature  (for ARM only) 
[screenly-viewer]     - Keep the container idling with `balena-idle` command 
[screenly-viewer]     - Show base image details with `balena-info` command
[... cut ...] -gobject         python-gi         python2.7         python2.7-dev         python-pip-whl         python-pip         python-pil         python-wheel         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3         uzbl         x11-xserver-utils         xserver-xorg     	&& apt-get clean && rm -rf /var/lib/apt/*' returned a non-zero code: 100
[Info]                Built on arm01
[Error]               Not deploying release.
Remote build failed
```

ealmonte32, 2020-10-04:
@EpicKris 
Thank you for the feedback and reporting, just a suggestion, can you instead of pasting all logs here, next time make a text file and paste there instead? just makes things easier and cleaner on this.

I think I see the issue, I was trying to use this suggestion from balena docs:
[image]

But used `rm -rf /var/lib/apt/*'` instead, thinking that it will be better to remove the entire lists directory.
When you can again, give it a run and let me know via an uploaded log.txt file.

EpicKris, 2020-10-04:
@ealmonte32 I'm unable to upload the command output at the moment, but I've hidden my previous comments to clear things up a bit.

`ealmonte32/screenly-ose` `6f3a5cc` remote build failed.

```zsh
screenly-ose-master % balena push Screenly-4
[Info]                Starting build for Screenly-4, user hq
[Info]                Dashboard link: https://dashboard.balena-cloud.com/apps/1575170/devices
[Info]                Building on arm01
[Info]                Pulling previous images for caching purposes...
[Success]             Successfully pulled cache images
[screenly-viewer]     Step 1/11 : FROM balenalib/raspberrypi4-64-debian:buster-run
[screenly-websocket]  Step 1/8 : FROM balenalib/raspberrypi4-64-debian:buster-run
[redis]               Step 1/6 : FROM balenalib/raspberrypi4-64-debian:buster-run
[screenly-server]     Step 1/11 : FROM balenalib/raspberrypi4-64-debian:buster-run
[screenly-viewer]      ---> bd9f20472954
[screenly-viewer]     Step 2/11 : RUN apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         git-core         libffi-dev         libssl-dev         libcairo2-dev         libgirepository1.0-dev         matchbox         net-tools         omxplayer         psmisc         python-dev     	python-gobject         python-gi         python2.7         python2.7-dev         python-pip-whl         python-pip         python-pil         python-wheel         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3         uzbl         x11-xserver-utils         xserver-xorg     	&& apt-get clean && rm -rf /var/lib/apt/lists/*
[screenly-websocket]   ---> bd9f20472954
[screenly-websocket]  Step 2/8 : RUN apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         python2.7         python2.7-dev         python-pip-whl         python-wheel         python-pip         libffi-dev         libssl-dev     	&& apt-get clean && rm -rf /var/lib/apt/lists/*
[screenly-server]      ---> bd9f20472954
[screenly-server]     Step 2/11 : RUN apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         git-core         libffi-dev         libssl-dev         libcairo2-dev         libgirepository1.0-dev         lsb-release         net-tools         nginx-light         omxplayer         psmisc         python-dev         python-gobject         python-gi         python2.7         python2.7-dev         python-pip-whl         python-wheel         python-pip         python-pil         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3    && apt-get clean && rm -rf /var/lib/apt/lists/*
[screenly-viewer]      ---> Running in 9b779b04a7c3
[screenly-websocket]   ---> Running in e0caafbbae56
[redis]                ---> bd9f20472954
[redis]               Step 2/6 : RUN apt-get update && apt-get install -y redis-server 	&& apt-get clean && rm -rf /var/lib/apt/lists/*
[screenly-server]      ---> Running in 3def49743712
[redis]                ---> Running in 1b5c11dd35f5
[screenly-celery]     Step 1/9 : FROM balenalib/raspberrypi4-64-debian:buster-run
[screenly-celery]      ---> bd9f20472954
[screenly-celery]     Step 2/9 : RUN apt-get update && apt-get install -y --no-install-recommends         build-essential         ca-certificates         git-core         libffi-dev         libssl-dev         libgirepository1.0-dev         libcairo2-dev         python-gobject         python-gi         python-dev         python2.7         python2.7-dev         python-pip-whl         python-wheel         python-pip     	&& apt-get clean && rm -rf /var/lib/apt/lists/*
[screenly-celery]      ---> Running in 1f6437264ea6
[screenly-celery]     Here are a few details about this Docker image (For more information please visit https://www.balena.io/docs/reference/base-images/base-images/): 
[screenly-celery]     Architecture: ARM v8 
[screenly-celery]     OS: Debian Buster 
[screenly-celery]     Variant: run variant 
[screenly-celery]     Default variable(s): UDEV=off 
[screenly-celery]     Extra features: 
[screenly-celery]     - Easy way to install packages with `install_packages <package-name>` command 
[screenly-celery]     - Run anywhere with cross-build feature  (for ARM only) 
[screenly-celery]     - Keep the container idling with `balena-idle` command 
[screenly-celery]     - Show base image details with `balena-info` command
[screenly-celery]     Get:1 http://deb.debian.org/debian buster InRelease [121 kB]
[screenly-celery]     Get:2 http://security.debian.org/debian-security buster/updates InRelease [65.4 kB]
[screenly-celery]     Get:3 http://deb.debian.org/debian buster-updates InRelease [51.9 kB]
[screenly-websocket]  Here are a few details about this Docker image (For more information please visit https://www.balena.io/docs/reference/base-images/base-images/): 
[screenly-websocket]  Architecture: ARM v8 
[screenly-websocket]  OS: Debian Buster 
[screenly-websocket]  Variant: run variant 
[screenly-websocket]  Default variable(s): UDEV=off 
[screenly-websocket]  Extra features: 
[screenly-websocket]  - Easy way to install packages with `install_packages <package-name>` command 
[screenly-websocket]  - Run anywhere with cross-build feature  (for ARM only) 
[screenly-websocket]  - Keep the container idling with `balena-idle` command 
[... cut ...] ct         python-gi         python2.7         python2.7-dev         python-pip-whl         python-pip         python-pil         python-wheel         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3         uzbl         x11-xserver-utils         xserver-xorg     	&& apt-get clean && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
[Info]                Built on arm01
[Error]               Not deploying release.
Remote build failed
```

ealmonte32, 2020-10-04:
@EpicKris thank you, quick question which architecture are you using on your balena application?
I have been testing with rpi instead of armv7hf.. i might have to switch to that to test..

[image]

EpicKris, 2020-10-04:
@ealmonte32 Raspberry Pi 4 uses `aarch64`.

ealmonte32, 2020-10-04:
>  Raspberry Pi 4 uses `aarch64`.

True, I keep forgetting I dont have the pi4 for testing this.. maybe you could just locally edit your Dockerfiles and troubleshoot and report back if you figured out why the command fails?

EpicKris, 2020-10-04:
> True, I keep forgetting I dont have the pi4 for testing this.. maybe you could just locally edit your Dockerfiles and troubleshoot and report back if you figured out why the command fails?

I'm afraid I don't know enough about Docker to do this.

ealmonte32, 2020-10-04:
Well, would you be willing to try and learn a little?
It is probably one of the packages that is not compatible with aarch64, basically you git cloned my repo somewhere right? ok so in that folder, go into there and into the docker folder and with your text-editor just open and edit the Dockerfile for viewer and for server:
[image]

you simply have to start removing some of the lines and packages that might be the culprit, maybe `omxplayer`, also maybe remove that line i added `&& rm -rf /var/lib/apt/lists/*` , i think omxplayer might be the one though because from a quick search it seems it is not 64-bit capable.. _(dont quote me on that lol)_

vpetersson, 2020-12-20:
This should be fixed now.

closed 2020-12-20 by vpetersson