Screenly/Anthias #1283: Static Dates

knapjongen, 2020-01-02:
#511 ## Overview of the Issue

Our screens change every month with new monthly offers, however, the time ranges are always the same Start at 12 a.m. End at 12 a.m.  New screen start at the beginning of the next month at precisely the same time.  Currently, each screen requires a new date and time.  Is there a way to set a default.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version:
* Screenly OSE Version:

vpetersson, 2020-01-03:
More sophisticated scheduling is currently reserved for the commercial version of Screenly. You can of course automate this yourself using the API, but it is not something that we have on the roadmap for OSE (unless someone contributes this code).

closed 2020-01-03 by vpetersson

knapjongen, 2020-01-03:
Thank you for the clarification.

*Raymond Martherus*
Marketing

*e* raymond.martherus@lassens.com
*w* lassens.com

[image: Facebook]   [image: Twitter]   [image: Pinterest]   [image:
Instagram]   [image: Tumblr]
[image: Youtube]
562-444-8670

*Click here to for donation requests*
<https://docs.google.com/forms/d/e/1FAIpQLSeaZGcBGfD_LdqpCYakPcMDxphtmXK8SbvO46YphdOk8JFlFQ/viewform?usp=sf_link>