Screenly/Anthias #1284: Single image upload to replicate for multiple locations

knapjongen, 2020-01-02:
#### Overview of the Issue

We have 7 locations with 2 sets of screens  4 locations with 3 screens and 3 locations with 3 screens.  Each of the screens are the same for the 2 sets, i.e., each screen 1, 2 and 3 are the same for all 4 location, yet we have to upload image 1 to screen 1 at location 1 and then image 1 to screen 1 at location 2, etc.  

Is it possible to load one image that will replicate to all of the screens in that group.  And we have 2 groups. 

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version:
* Screenly OSE Version:

vpetersson, 2020-01-03:
Hi @knapjongen. Thanks for reaching out.

You have two options:

* Program something that consumes the API to automate the process
* Switch to the commercial version that already supports this (see [this page](https://www.screenly.io/screenly-ose-vs-pro/) for a comparison)

closed 2020-01-03 by vpetersson

knapjongen, 2020-01-03:
Thank you for your response.

*Raymond Martherus*
Marketing

*e* raymond.martherus@lassens.com
*w* lassens.com

[image: Facebook]   [image: Twitter]   [image: Pinterest]   [image:
Instagram]   [image: Tumblr]
[image: Youtube]
562-444-8670

*Click here to for donation requests*
<https://docs.google.com/forms/d/e/1FAIpQLSeaZGcBGfD_LdqpCYakPcMDxphtmXK8SbvO46YphdOk8JFlFQ/viewform?usp=sf_link>