Screenly/Anthias #1291: Can't remove over-scan

justinkruit, 2020-01-17:
#### Overview of the Issue

I try to display a full HD image on my full HD monitor, using a Raspberry PI 3b+ and Balena. No matter what I try, the overscan doesn't get removed. Videos display fine. I've tried adding RESIN_HOST_CONFIG_overscan_* values to the device settings in Balena and rebooted it, but the image still isn't full screen. Also tried uploading an image bigger than full HD, but that didn't work either.

#### Reproduction Steps

* Install Screenly on a RP3b+ using Balena ([Instructions I used](https://www.balena.io/blog/deploy-free-digital-signage-software-screenly-ose/))
* Upload an image as asset
* Display the image

### Environment

* Raspberry Pi Hardware Version: 3b+
* Screenly OSE Version: latest commit 55e9f9f

vpetersson, 2020-01-17:
Maybe a silly question, but is the image 1920x1080?

justinkruit, 2020-01-17:
Yes it was. But I also figured it out already (of course just when you make the issue :D).

I set the `RESIN_HOST_CONFIG_disable_overscan` setting to `1`, and now it's all working fine!

closed 2020-01-17 by justinkruit