Screenly/Anthias #1292: Specific web url doesn't show up since last week

thefyfy, 2020-01-22:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

Since last week, all URL from klipfolio.com doesn't show correctly with Screenly whereas they are ok on PC. I want your help, if it's possible, to know how to debug that case on Screenly and find what can cause this on the web page.

#### Reproduction Steps

I give you one of our URL to test if you want on your side : https://app.klipfolio.com/published/e291c9d0f54a913228f9cb90bfac205c/logistique-dg

### Environment

* Raspberry Pi Hardware Version: 3 B+
* Screenly OSE Version: 0.18.2 (Sprint 10) and with experimental branch too (with the new browser?)

Thanks for your amazing job in this project 👍

vpetersson, 2020-01-22:
@thefyfy - There must have been some changes upstream with Klipfolio as we didn't make any changes on our end.

I know @rusko124 is working on fixing up the experimental browser, and we hope to have an update in the near future for this.

vpetersson, 2020-01-22:
Also @thefyfy I had one of our engineers for the commercial version verify that that web page does work there, so if you need to get it to work right away, that's the best path forward. If not, we will be working on it in the near future.

thefyfy, 2020-01-22:
Ok thanks for replying me. Just to know, the experimental browser is working on webkit isn't it ?
For the commercial way, why not, is it possible to pay for "one shot" support ?

vpetersson, 2020-01-22:
> Ok thanks for replying me. Just to know, the experimental browser is working on webkit isn't it ?

Yes, the current experimental branch is using webkit.

> For the commercial way, why not, is it possible to pay for "one shot" support ?

The commercial version is a completely different version and does not share code base (see [this page](https://www.screenly.io/screenly-ose-vs-pro/) for details). Hence it's an entirely different product, but support is included.

thefyfy, 2020-01-22:
Ok and just to be sure: your engineer test my web page on commercial version and it worked fine? Or it's just a new way to explore for me ?

vpetersson, 2020-01-22:
> Ok and just to be sure: your engineer test my web page on commercial version and it worked fine? Or it's just a new way to explore for me ?

Yes, we tested this earlier today and it appears to be rendering fine: 
[image]

thefyfy, 2020-01-22:
Ok cool, I'm going to explore this way if we want to go fast on the resolution of this problem.

closed 2020-03-29 by stale[bot]