Screenly/Anthias #1293: Grafana failed to load data

ruben-herold, 2020-01-24:
hi,

we have here some screenly OSE systems to show grafana dashboards. After the last grafana update we get an error on loading the dashboards:

prometheus-src plugin failed
Error: Loading chunk 44 failed. (missing:
https://grafanaserver.local/public/build/prometheusPlugin.<some hex vaules>.js)

with normal desktop browsers we don't see any problems.

vpetersson, 2020-01-24:
Could you try switching to the experimental version to see if that works?

ruben-herold, 2020-01-24:
@vpetersson is there an image downloadable?
My last test was with: https://github.com/Screenly/screenly-ose/releases/download/0.18.2/2019-09-25-Screenly-OSE.zip

vpetersson, 2020-01-24:
No, you need to manually install it using the [installer](https://github.com/Screenly/screenly-ose#installing-on-raspbian). That said, you can still use that as the base image.

Alternatively, you could try the commercial version of Screenly, which is known to work with Grafana (see [this page](https://www.screenly.io/screenly-ose-vs-pro/) for details).

ruben-herold, 2020-01-24:
what I have done no was:

Settings --> Upgrade Screenly --> Latest. 

after this reboot. But nothing changed.

ruben-herold, 2020-01-24:
@vpetersson version shows now 

Screenly Version | master@55e9f9f
-- | --

vpetersson, 2020-01-24:
No, you need to SSH into the Pi, run the upgrade script and answer yes to the question if you want to run the experimental version.

ruben-herold, 2020-01-24:
@vpetersson tried this. After reboot the pi hangs now in rc-local.service. I can access the new webinterface but no browser on the screen only the Screenly Logo with rc-local.service on it.

vpetersson, 2020-01-24:
Can you SSH in an get some logs?

ruben-herold, 2020-01-24:
@vpetersson no cause card already new flashed with an  Raspbian Lite. Its updating in the moment, after this I will try again.

ruben-herold, 2020-01-24:
@vpetersson first try with Raspbian Lite. failed:

  Booted it, 
  logged in , 
  update system, 
  start your script. 

Ended in a error message installing docker-ce 
[image]

Now I will try again via ssh.

ruben-herold, 2020-01-24:
@vpetersson got it running now. But same error.

ruben-herold, 2020-01-24:
@vpetersson on my desktop all grafana stuff runs fine with firefox, chromium, google chome, and even with midori (small webkit based browser)

vpetersson, 2020-01-24:
What Raspberry Pi hardware are you running on?

ruben-herold, 2020-01-27:
@vpetersson Raspberry Pi 3 Model B V1.2

vpetersson, 2020-01-27:
Please take a look at #1282 to see if that helps.

ruben-herold, 2020-01-27:
@vpetersson I'm not sure if we talk about different stuff.  I got experimental running, please see above. If I understand #1282 right it's about an install problem. 

But on experimental I get the same error like on current.  The browser in experimental also said:

prometheus-src plugin failed
Error: Loading chunk 44 failed. (missing:
https://grafanaserver.local/public/build/prometheusPlugin..js)

On my desktop all grafana stuff runs fine with firefox, chromium, google chome, and even with midori (small webkit based browser)

equinoxefr, 2020-01-29:
Hi,

Same issue here, i'm running experimental version with grafana. Since last version (6.6.0) of grafana i have an error: "An unexpected error as happened"

Opening a grafana screen without any graph works.

Is there a log to check ?

ruben-herold, 2020-01-29:
Yes the same her with 6.6.0 with 6.5 I get the error above.

ruben-herold, 2020-02-05:
any news here?

varac, 2020-02-07:
I just installed latest screenly-ose release then the *only* use case for me is to show grafana dashboards, which won't work for me neither.
Hearing the the paid version supports grafana makes me sad because this shows that the ose version doesn't get the support it should. Please fix this ! As @ruben-herold explained above, my grafana dashboards display fine with any browser I tried so far and only fail in screenly-ose.

varac, 2020-02-07:
I'm using grafana v6.6.0 (5c11bbdfb4) for the record.

vpetersson, 2020-02-10:
> Hearing the the paid version supports grafana makes me sad because this shows that the ose version doesn't get the support it should

We're happy to accept contributions. The reality is that the commercial version of Screenly is what enable us to spending money on paid developers to OSE in the first place. So yes, it will always take precedence.

ruben-herold, 2020-02-10:
@vpetersson is it possible to get a build that starts the browser with enabled  js console or something, so that the error can logged to see whats going wrong.

vpetersson, 2020-02-10:
@ruben-herold I'd have to pass that on to @rusko124 as he worked most recently on it. But yes, i think that should be possible.

ealmonte32, 2020-02-24:
@ruben-herold , you can use VNC viewer and control the browser with mouse and get to inspect element (screenshots on bottom).. but, I just tested with a master version of screenly (development), and I get the site to load this:

[image]
[image]

using VNC viewer:
[image]
[image]

ruben-herold, 2020-02-24:
I have done some testing and it happend only if I use prometheus as datasource. 
Non Prometheus based databsources worked.

ealmonte32, 2020-02-24:
Ok, I see what you mean, these errors:

[image]

[image]

Well.. this is as you know most likely the uzbl browser not supporting whatever plugin that is.

ealmonte32, 2020-02-24:
_Self-note: I am noticing that when I leave this asset open for unlimited time (only asset), Xorg (X.service) CPU usage spikes to 100%+ for undefined time, stays like that, and makes Pi hang.
Will need to use cpu limits on this._

ruben-herold, 2020-02-25:
@ealmonte32  also it does not work in experimental. As far as I know that experimental don't use uzbl

ealmonte32, 2020-02-27:
> @ealmonte32 also it does not work in experimental. As far as I know that experimental don't use uzbl

Correct, it uses QT webview, and I think it just needs to be compiled with support for all these plugins.. maybe rusko can chime in about that..

closed 2020-05-04 by stale[bot]

ruben-herold, 2020-05-08:
Still not work. Any progress?

vpetersson, 2020-05-08:
No progress has been made on this, but I can confirm that this is working in the commercial version. That's the only workaround i can offer right now.

ruben-herold, 2020-05-08:
@vpetersson  anything I can do as non developer? Testing? Or something else to speedup

vpetersson, 2020-05-08:
@ruben-herold Yes, it seems like we need to patch the [browser](https://github.com/Screenly/screenly-ose-webview) to support Grafana. We had a similar issue in the commercial version a while back. @sergey-borovkov, perhaps you can chime in here?

sergey-borovkov, 2020-05-08:
> @ruben-herold Yes, it seems like we need to patch the [browser](https://github.com/Screenly/screenly-ose-webview) to support Grafana. We had a similar issue in the commercial version a while back. @sergey-borovkov, perhaps you can chime in here?

We moved to webengine in the commercial version. So that issue resolved itself there.

vpetersson, 2020-05-13:
@ruben-herold I'm not sure what the amount of effort is to move from Webkit to Webengine in the browser to be honest. If you're interested in looking into this, that would be fantastic.