Screenly/Anthias #1295: Image using same mac address

goll420, 2020-01-28:
#### Overview of the Issue

Image uses same MAC address.

#### Reproduction Steps

Steps to reproduce this issue, eg:

When using in an environment which limits traffic on the network via mac address, we had 2 Pis flashed with same image. As it transpired they both have the same mac addresses for wifi. Is there a way to change the mac address of the wifi adaptor?

### Environment

* Raspberry Pi Hardware Version: 3B+
* Screenly OSE Version: 2019.09.25

vpetersson, 2020-01-28:
That should in theory be impossible. The MAC address part of the network hardware. It's not something that is set in the disk image. If the MAC address is the same on two devices on a network, that will have severe implications on a large number of things.

There are ways to change the MAC address, but I'd start by looking into this to ensure it is the actual case. You can use `ipconfig` or `ip` and similar tools to check this.

goll420, 2020-01-28:
I did think it would also be impossible, but my guy on site is telling me he can't add the 2nd pi as the MAC address is the same. He's bringing it back for me to look at further but wondered if it was a known issue; I can tell from your response that it isn't though.

vpetersson, 2020-01-28:
Yeah it should in theory be impossible. Perhaps there is some really strange edge case. Please do look into this as I'm curious to see what you find.

goll420, 2020-01-28:
Got the 2nd Pi back to headquarters. Got this from DHCP and also in the Screenly settings.

[image]

goll420, 2020-01-28:
MAC Address | b8:27:eb:12:d0:99

goll420, 2020-01-28:
Oh, this looks like a newbie mistake by my guy.... Sorry for wasting any time. A simple case of picking up the wrong pi mistaking it for the other....

vpetersson, 2020-01-28:
Haha, all good :)

closed 2020-01-28 by vpetersson