Screenly/Anthias #1296: Password issues I cannot connect to WiFi

netgifted, 2020-02-04:
I tried different Raspberry pi devices, I can't pass by the password it always says I enetered an incorrect password, after 10 attempts and flashing multiple cards I gave up. Is there any way I can manualy insert the WiFi password?

### Environment

* Raspberry Pi Hardware Version: 3b and 2b
* Screenly OSE Version: 10

vpetersson, 2020-02-05:
Hi @netgifted - We've now upgraded to a new release of the network tool.  Could you try to upgrade to the latest development release?

netgifted, 2020-02-05:
Thanks a lot, just to be sure... at https://github.com/screenly/screenly-ose/releases the latest release is still showing 2019. what is the right image download? thanks

vpetersson, 2020-02-05:
No, you need to run the installation script as outlined under [Install Screenly OSE](https://www.screenly.io/ose/).

netgifted, 2020-02-05:
After spending all morning and 3 attempts to install I finally gave up. There are always errors on the Python installation due to deprecation of 2.7 at the end of install after reboot it always goes to login. Thank you for your help I'll look into Screenly in a 1 year or so, this is too buggy for me and though I'm persistent I don't have much time to waste. Thank a  lot once again :)

vpetersson, 2020-02-05:
Were you able to resolve the issue, @netgifted?

netgifted, 2020-02-05:
No I wasn't I tried still to install Raspbian instead of Raspbian Lite. I installed Python and upgraded all as I run `bash <(curl -sL https://www.screenly.io/install-ose.sh)` the script produces python deprected errors, when it gets to the end and I reboot Screenly doesn't start. I spent 5 hours on this...

ealmonte32, 2020-02-21:
@vpetersson I also experienced these wifi "wrong password" issues with 2019 image and resin, but after 3 attempts it lets you in, very weird.. also, a while ago (many months ago, probably an older image issue), I had to change the default locale and set wifi country before i was able to get the proper wifi connected..

@netgifted raspbian lite is better for screenly since it doesnt install all of the unnecessary packages for the GUI since screenly viewer doesnt use it anyway, plus many many other packages, which is why I recommend always either the screenly image, or raspbian lite and then run the bash script.
Getting deprecated errors doesnt prevent the installer from finishing and installing everything, I'm sorry you spend 5 hours on this but it is probably something being overlooked..

Are you using the latest image for raspbian? if so this might be why, I suggest the Stretch version (screenly doesnt supports buster yet) ..

Here is where you can get the version of Raspbian that works perfectly fine:
http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/2019-04-08-raspbian-stretch-lite.zip

netgifted, 2020-02-21:
thanks a lot! Iºll try this Raspbian image today

Tony Silveira
tony@netgifted.com (mailto:tony@netgifted.com)
http://lestudio.tv

netgifted, 2020-02-24:
I installed the image and ran
$ bash <(curl -sL https://www.screenly.io/install-ose.sh)
After reboot I am back at command prompt screenly doesn't start. I followed installation steps and there where no significant errors (python2.7 deprecation only)
Is there anything I can do? I've used screenly 2 years ago and liked it but it seams now impossible to run on my machines (Pi 2, 3 or 4)
Thanks

Tony Silveira
tony@netgifted.com (https://link.getmailspring.com/link/E62E4FED-6498-4E26-B087-10A941F83124@getmailspring.com/0?redirect=mailto%3Atony%40netgifted.com&recipient=cmVwbHkrQUpHS0RWNDJBTk8yRUlBUU1SM0hZU040TFJWSzNFVkJOSEhDQ1lERTRJQHJlcGx5LmdpdGh1Yi5jb20%3D)
http://lestudio.tv (https://link.getmailspring.com/link/E62E4FED-6498-4E26-B087-10A941F83124@getmailspring.com/1?redirect=http%3A%2F%2Flestudio.tv&recipient=cmVwbHkrQUpHS0RWNDJBTk8yRUlBUU1SM0hZU040TFJWSzNFVkJOSEhDQ1lERTRJQHJlcGx5LmdpdGh1Yi5jb20%3D)

ealmonte32, 2020-02-24:
I am having a hard time figuring out what part of the installation you're doing differently then others for it not to work since these steps are working for everyone else, and I just tested as well.

Can you be a bit more specific, maybe post some screenshots of the steps you're taking when doing this?
Did you flash the SD card with the raspbian lite stretch image using Etcher?
When you inserted this SD card and logged in and were at prompt/console, you logged in with username pi and password raspberry, then typed the bash install script, and then what did you select from the install menu/steps?
when this is all done and it asks you to reboot, and you press y to reboot, what happens after screenly reboots?

netgifted, 2020-02-25:
I burn the card image with Etcher. I follow every step entering raspi-config Raspbian is working fine 100% I then install Screenly and at the end reboot. Raspbian starts and I get to the Username and Password prompt again. I tried this  several times (maybe 8 to 10 times) with different options not upgrading, upgrading, Managing Network with Raspbian, managing with Screenly, whatever I do I am back at Username and Passoword prompt. I can record every single step if you want just one more time but at this point I really spent more time than what I wanted with this and I'm giving up. Thanks for your help, I find odd that anyone can get this to work...

ealmonte32, 2020-02-25:
I'm really perplexed by your situation as I have not seen many users have such difficulty setting up a screenly raspberry pi environment. I have read Rpi4 users have some minor issues, but not on Pi2/3. 

So if you take the screenly sprint 10 image https://github.com/Screenly/screenly-ose/releases/download/0.18.2/2019-09-25-Screenly-OSE.zip and burn it to SD card with etcher, and then insert into Rpi3, and start it up, it gets to the Screenly-OSE access point screen where you have to connect to it from another device (if using Pi on WiFi), and join it to an existing WiFi network, then it restarts the services and should get an IP from your network..
I believe you have been stuck at the part where you try to join the access point it creates and you say the password you enter keeps saying is wrong? if so, the only thing I can tell you is that I tested this many times and even recently and after 3 retries with password it gives on screen, it then accepts it. The only other time I had an issue was when it did not want to see my WiFi SSID, and I had to change the locale and wifi country via `sudo raspi-config` , and then it all worked, but that is more of a one time setup situation because the screenly image from sprint release is not universally setup (sort of can't be, that's why raspbian asks you questions during initial setup so it can customize to your locale, etc), so screenly iso image defaults to UK I believe, while I am in US, and I had to make those changes before that screenly-ose-wifi setup worked.

I want to help you, but I can't seem to replicate your issue.

netgifted, 2020-02-25:
I was finally able to install Screenly on a Raspberry pi 3b The issue must be the USB WiFi on the other boards. Thank you for your help it was really appreciated. I was able to install with an image and yes I had to input the password 4 times :-)

closed 2020-02-25 by netgifted

vpetersson, 2020-02-25:
This is an interesting catch. So for the WiFi adapter to be able able to work for this, it needs to support[AP Mode](https://www.elinux.org/RPi_USB_Wi-Fi_Adapters). I'd think that most does, but there are likely adapters out there that doesn't support it, so it's a good thing to check for in the future when we encounter these things.