Screenly/Anthias #1300: Persistent Data for Dockerized Development Environment

aptonline, 2020-02-11:
I've setup the Dockerized Development Environment and all working well but wanted to know if its possible to keep the data/assets/settings persistent over sessions.

I've adding `-v ~/Desktop/assets:/home/pi/screenly_assets` to mount the assets and this is working but the server is not accessing the saved assets between re-runs on the docker container.

Any help would be grateful.

vpetersson, 2020-02-13:
You'd also need to volume mount the database. The root directory is `/home/pi/screenly` as you can see [here](https://github.com/Screenly/screenly-ose/blob/master/bin/install.sh#L161), so you might want to simply volume mount all of that.

aptonline, 2020-02-13:
Thanks but is that not what I’m doing with the above parameter addition?

vpetersson, 2020-02-19:
Yes, the the `-v` option is for volume mounting the file system into the container. See [the man page](https://docs.docker.com/engine/reference/run/#volume-shared-filesystems) for more details.

closed 2020-04-26 by stale[bot]