Screenly/Anthias #1301: Image assets do not display

NebojsaIlicRookie, 2020-02-14:
Greetings,

I have a mistery to solve with Screenly. When i etch the SD card and boot my Pi, everything works well. The system itself works good. I can access the editor, i can see the default widgets when i first boot it. I see the IP.
BUT, when i start adding assets through the editor, i load in a webpage and this works perfect. As soon as i add a JPEG (exactly the same file that worked before), it starts behaving odd. The JPEG assets when due do not show, but rather a quick white flash occurs, and then just a black screen. After the allocated time (2x10 seconds), the webpage displays perfectly fine for 10 seconds. Then the loop ends and the black screen that occurs at the end of the loop lasts longer than usual.

After some time it starts displaying the assets for a short time, but then goes back to the "broken" loop.

I tried both the previous version and the newest version of the software. Problem remains, and behaves exactly the same. Using the same rasp pi and the same SD card that i used before it stopped functioning properly.

Reproduction steps are simple:

Boot the software
Enter the IP and access the editor
Drag and drop JPEG assets
Enable them
Lose your mind

* Raspberry Pi Hardware Version: Pi 3 B+
* Screenly OSE Version: 0.18.2

Please if some debug logs are needed let me know also how to get them, I am relatively new to 
all of this, but IT literate.

Thanks in advance

frijhee, 2020-02-16:
Is it possible that your JPEG asset was somehow corrupted during the upload/enable process?
Have you checked the integrity of your files in the /home/pi/screenly_assets/ folder? A quick way to see if your JPEG file(s) are recognized as JPEGs would be to run the command:
"file *" in the screenly_assets folder.

I'm not sure if this will help, but IF your image file isn't corrupted, you should be able to see some metadata information about it (indicating its an image file).

Here's an example:
pi@raspberrypi:~/screenly_assets $ file 50a73f8c71bc49cd89b6c2365b4aa87d 

50a73f8c71bc49cd89b6c2365b4aa87d: JPEG image data, Exif standard: [TIFF image data, big-endian, direntries=10, manufacturer=Apple, model=iPad Pro, orientation=upper-left, xresolution=150, yresolution=158, resolutionunit=2, software=12.3.1, datetime=2019:12:18 17:08:59], baseline, precision 8, 4032x3024, frames 3

NebojsaIlicRookie, 2020-02-18:
Thanks for the reply and the interest.

So I accidentally stumbled upon the root of the problem in the meantime.
While I was playing around with the settings, I uploaded a screenshot as a filler, along with the JPEGs in question, and the screenshot would display perfectly fine.
So I backtracked, and it turns out that the JPEGs are fine (not corrupted) but either the screenly app or the rasp Pi has problems with their resolution (7920x4320px). I export these images from a PDF, as it is an editable layout for some organization charts etc, and the export settings were really high after a reinstall without me noticing. After dropping the resolution to 1080p, it works fine. 
Worth keeping in mind!

closed 2020-02-18 by NebojsaIlicRookie