Screenly/Anthias #1302: omxplayer timed out

nosyminotaur, 2020-02-19:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue
Using Screenly with Balena, but keep getting
`omxplayer timed out` issues. The video stops responding after a few seconds and then stops abruptly and plays the next one.
It looks as if the device is very slow or something.

A paragraph or two about the issue you're experiencing.
Here's some logs copied from the Balena Dashboard-
```
19.02.20 11:05:21 (+0000)  screenly-viewer  update_playlist
19.02.20 11:05:21 (+0000)  screenly-viewer  Generating asset-list...
19.02.20 11:05:21 (+0000)  screenly-viewer  generate_asset_list deadline: 2020-01-07 11:15:00
19.02.20 11:05:21 (+0000)  screenly-viewer  get_next_asset after refresh
19.02.20 11:05:21 (+0000)  screenly-viewer  get_next_asset counter 0 returning asset 3 of 16
19.02.20 11:05:21 (+0000)  screenly-viewer  Showing asset In It Together | LinkedIn India (video)
19.02.20 11:05:21 (+0000)  screenly-viewer  Asset URI /data/screenly_assets/4bb485dc98734158a7bee92ee6312751
19.02.20 11:05:21 (+0000)  screenly-viewer  Displaying video /data/screenly_assets/4bb485dc98734158a7bee92ee6312751 for 60 
19.02.20 11:05:21 (+0000)  screenly-viewer  current url is file:///tmp/screenly_html/black_page.html
19.02.20 11:06:42 (+0000)  screenly-viewer  omxplayer timed out
19.02.20 11:06:42 (+0000)  screenly-viewer  Last update: 2020-02-19 04:06:54
19.02.20 11:06:44 (+0000)  screenly-viewer  get_next_asset
19.02.20 11:06:44 (+0000)  screenly-viewer  refresh_playlist
19.02.20 11:06:44 (+0000)  screenly-viewer  refresh: counter: (0) deadline (2020-01-07 11:15:00) timecur (2020-02-19 11:06:44.437303)
19.02.20 11:06:44 (+0000)  screenly-viewer  update_playlist
19.02.20 11:06:44 (+0000)  screenly-viewer  Generating asset-list...
19.02.20 11:06:44 (+0000)  screenly-viewer  generate_asset_list deadline: 2020-01-07 11:15:00
19.02.20 11:06:44 (+0000)  screenly-viewer  get_next_asset after refresh
19.02.20 11:06:44 (+0000)  screenly-viewer  get_next_asset counter 0 returning asset 4 of 16
19.02.20 11:06:44 (+0000)  screenly-viewer  Showing asset 100% Natural Specialty Tea  - Teamonk Teas (video)
19.02.20 11:06:44 (+0000)  screenly-viewer  Asset URI /data/screenly_assets/58e535db35694a00b19a61eea67a8936
19.02.20 11:06:44 (+0000)  screenly-viewer  Displaying video /data/screenly_assets/58e535db35694a00b19a61eea67a8936 for 30 
19.02.20 11:06:44 (+0000)  screenly-viewer  current url is file:///tmp/screenly_html/black_page.html
19.02.20 11:07:35 (+0000)  screenly-viewer  omxplayer timed out
19.02.20 11:07:35 (+0000)  screenly-viewer  Last update: 2020-02-19 04:06:54
19.02.20 11:07:37 (+0000)  screenly-viewer  get_next_asset
19.02.20 11:07:37 (+0000)  screenly-viewer  refresh_playlist
19.02.20 11:07:37 (+0000)  screenly-viewer  refresh: counter: (0) deadline (2020-01-07 11:15:00) timecur (2020-02-19 11:07:37.153326)
19.02.20 11:07:37 (+0000)  screenly-viewer  update_playlist
19.02.20 11:07:37 (+0000)  screenly-viewer  Generating asset-list...
19.02.20 11:07:37 (+0000)  screenly-viewer  generate_asset_list deadline: 2020-01-07 11:15:00
19.02.20 11:07:37 (+0000)  screenly-viewer  get_next_asset after refresh
19.02.20 11:07:37 (+0000)  screenly-viewer  get_next_asset counter 0 returning asset 5 of 16
19.02.20 11:07:37 (+0000)  screenly-viewer  Showing asset Best Android App |  Online Scholarship form  | Scholarship for Indian Students 2018 - 19 (video)
19.02.20 11:07:37 (+0000)  screenly-viewer  Asset URI /data/screenly_assets/3eb53a856bc84a809952b7ef3b185213
19.02.20 11:07:37 (+0000)  screenly-viewer  Displaying video /data/screenly_assets/3eb53a856bc84a809952b7ef3b185213 for 29 
19.02.20 11:07:37 (+0000)  screenly-viewer  current url is file:///tmp/screenly_html/black_page.html
19.02.20 11:07:47 (+0000)  rabbitmq  2020-02-19 11:07:47.792 [error] <0.4191.0> closing AMQP connection <0.4191.0> (172.17.0.1:38652 -> 172.17.0.2:5672):
19.02.20 11:07:47 (+0000)  rabbitmq  missed heartbeats from client, timeout: 60s
19.02.20 11:07:48 (+0000)  rabbitmq  2020-02-19 11:07:48.879 [error] <0.4204.0> closing AMQP connection <0.4204.0> (172.17.0.1:38682 -> 172.17.0.2:5672):
19.02.20 11:07:48 (+0000)  rabbitmq  missed heartbeats from client, timeout: 60s
19.02.20 11:07:59 (+0000)  rabbitmq  2020-02-19 11:07:59.686 [error] <0.4227.0> closing AMQP connection <0.4227.0> (172.17.0.1:38966 -> 172.17.0.2:5672):
19.02.20 11:07:59 (+0000)  rabbitmq  missed heartbeats from client, timeout: 60s
19.02.20 11:08:27 (+0000)  screenly-viewer  omxplayer timed out
19.02.20 11:08:27 (+0000)  screenly-viewer  Last update: 2020-02-19 04:06:54
19.02.20 11:08:28 (+0000)  screenly-viewer  get_next_asset
19.02.20 11:08:28 (+0000)  screenly-viewer  refresh_playlist
19.02.20 11:08:28 (+0000)  screenly-viewer  refresh: counter: (0) deadline (2020-01-07 11:15:00) timecur (2020-02-19 11:08:28.943732)
```
#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version:
Model 3B Revision: 1.2 Ram: 1 GB Sony UK
* Screenly OSE Version:

vpetersson, 2020-02-19:
These strange errors are often related to broken SD cards. Try swapping it out for a new one.

nosyminotaur, 2020-02-20:
I am using this micro SD card along with my Pi ->
https://www.amazon.in/SanDisk-Ultra-microSD-Adapter-SDSQUAR-016G-GN6MA/dp/B073K14CVB/
I bought this thinking this was of high quality. Should I replace this?

vpetersson, 2020-02-20:
Nothing wrong with SanDisk, but SD cards got a limited lifespan. This life span depends on the amount of writes. Sometimes they last for months, sometimes years.

ealmonte32, 2020-02-21:
@nosyminotaur can you tell us more about the assets? without this video, does screenly work fine? are youtube videos/URLs fine? is any other video working fine? how large is the video file you are uploading?

the more details you can give, the better we can isolate what the cause might be.. I have the same SD card you showed..

closed 2020-04-30 by stale[bot]