Screenly/Anthias #1306: Large asset uploads result in orphaned files and no asset entry in database

ftrofin, 2020-02-23:
I've been trying to upload a large video file (3GB) and I'm consistently seeing this:

1. The file is uploaded successfully. I can see a new file in ~/screenly_assets with a name like
932cdb09ad0c4942b45bd5597b4ab0b3 (which tells me that the upload completed because it doesn't have the `.tmp` suffix).
2. there's no entry in the database file:
```
sqlite3 .screenly/screenly.db 
SQLite version 3.16.2 2017-01-06 16:32:41
Enter ".help" for usage hints.

sqlite> select * from assets;

default_4e6404b48a2645598946273490e2bb4d|Screenly Weather Widget|https://weather.srly.io||2019-09-26 08:02:57.216340|2025-09-26 08:02:57.216340|10|webpage|1|0|0|0|0
default_9d4f5258b6964b22adbd3096c12b6371|Screenly Clock Widget|https://clock.srly.io||2019-09-26 08:02:57.216340|2025-09-26 08:02:57.216340|10|webpage|0|0|0|1|0```

(those are the default assets for weather and clock).

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Click "Upload" tab
* select a large file (or use a slow connection).
* A progress bar tracks the upload status. When it reached 100% nothing happens: the dialog remains in place. No errors are shown.
* After I dismiss it no new assets show up in the Schedule Overview.

### Environment

* Raspberry Pi Hardware Version: Model 2B Revision: 1.1 Ram: 1 GB
* Screenly OSE Version: master@ec7387e
(Initially I tried with 2019-09-25 version then I upgrade to latest hoping that it was fixed.

I tried this with both Safari and Chrome browsers. Same problem.

ftrofin, 2020-02-23:
I also want to mention that the upload transfer is abysmal: I get 1.5-2.5 MB/sec between the laptop and the Raspberry Pi running the Screenly server and they are both wired in (no wireless) into the same switch running at 1Gbps.

ealmonte32, 2020-02-24:
Hmm.. I believe this is a limit on apache and nothing to do with Pi or screenly, maybe you can change it and try again..

`To restricts the total size of the HTTP request body sent from the client use LimitRequestBody Directive. You can add this directive using .htaccess file or httpd.conf file under virtual host or directory configuration options. You can set value (in bytes) from 0 (unlimited) to 2147483647 (2GB) that are allowed in a request body.`

closed 2020-05-02 by stale[bot]