Screenly/Anthias #1307: External monitoring

pelicanmedia, 2020-02-25:
Not an issue, more of a query...
I'd like to install Nagios on my Balena install to monitor uptime along with all my other external servers etc.
Out of the services installed (redis, screenly-celery, screenly-server, screenly-viewer & screenly-websocket) what would be the best one to install on that would be persistent on reboot?

Thanks

### Environment

* Raspberry Pi Hardware Version: Pi3 B
* Screenly OSE Version: Latest

vpetersson, 2020-02-25:
Not sure  if you mean a Nagios server or just monitor it using Nagios. I would strongly discourage installing a Nagios server, as we're already operating on limited resources. If you're just looking to monitor the installation using Nagios, the easiest way is to monitor the webservice i presume.

pelicanmedia, 2020-02-25:
Hi, sorry I meant just monitor it using Nagios. By installing the nagios-nrpe-server and nagios-plugins. Mainly to monitor uptime so I can see if/when there have been unplanned power disruptions.

vpetersson, 2020-02-25:
Ah, ok. I'm not sure if it's pull or push (been ages since I touched Nagios). Regardless, I would recommend adding a new container that only has Nagios. Since Docker containers should be considered ephemeral, you can't just install it within an existing one.

pelicanmedia, 2020-02-25:
Ok great, thanks for the quick response. Just wanted to check before I broke something!
I'll have a go at creating a new container.

closed 2020-05-02 by stale[bot]