Screenly/Anthias #1308: Settings removed from URL

Naesstrom, 2020-02-25:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue
I'm trying to setup Screenly OSE with balena cloud but when I add a URL to a published Google Slide it removes all settings like loop, time etc. from the URL

#### Reproduction Steps

* Create a slide in Google Drive and go to File -> Publish to the web and enter settings like time, auto start, auto loop etc. and copy the URL
* Press Add Asset
* Enter the Asset URL ie: (https://docs.google.com/presentation/d/e/googleslidessecretkey/pub?start=true&loop=true&delayms=10000)
* Press "Save"
* Click on "Edit"
* Asset location has now changed to "https://docs.google.com/presentation/d/e/googleslidessecretkey"
* Set it to active and it will rush through the slides 1s/slide and stop at the last one, ignoring settings.

### Environment

* Raspberry Pi Hardware Version: Model 3B Revision: 1.2 Ram: 1 GB Sony UK
* Screenly OSE Version: Not sure where I see the latest but cloned the repo today

ealmonte32, 2020-03-13:
@Naesstrom 
This PR: https://github.com/Screenly/screenly-ose/pull/1321
will fix that issue, if you do not want to wait, take a look at the viewer.py file and edit it to add the `.replace(...)` code I added, and you would have it running properly after that (restart the viewer after editing the file).

abrenoch, 2020-03-17:
This is honestly ridiculous.

Ampersands in the URL will be converted.

So `http://www.test.com?param=1&param2=2`, as entered into the screenly asset URL, will save as `http://www.test.com?param=1&amp;param2=2`

Can we get a fix for this yesterday?

ealmonte32, 2020-03-17:
Can you verify you are running the latest viewer.py? try to `cat` the viewer.py file on your screenly-viewer service to see if the `.replace` has this line:

```
escaped_url = current_browser_url.replace('@', '\\@').replace('&amp;amp;', '\\&')
```

abrenoch, 2020-03-17:
Thanks for the quick reply @ealmonte32 

I can verify that line exists on line# 326:
`escaped_url = current_browser_url.replace('@', '\\@').replace('&amp;amp;', '\\&')`

I would be happy to try out any suggestions!

ealmonte32, 2020-03-17:
@abrenoch This is right now on your current screenly-viewer container?

abrenoch, 2020-03-17:
That is correct

ealmonte32, 2020-03-17:
And was the asset URL deleted and then added again on this new version?

abrenoch, 2020-03-17:
Do you mean when I tried to edit the asset @ealmonte32 ? I do not believe the asset URL can be modified after it is added, but it does update regardless - I can just click the edit button then save and it will exhibit that behavior.

Unless you meant something else?

ealmonte32, 2020-03-17:
> Do you mean when I tried to edit the asset @ealmonte32 ? I do not believe the asset URL can be modified after it is added, but it does update regardless - I can just click the edit button then save and it will exhibit that behavior.
> 
> Unless you meant something else?

Assets cannot be updated or edited, you basically update the name of the asset not the actual URL that was passed onto the viewer. (AFAIK)

Can you just test by deleting the asset and adding it again?

Meaning just create a new Asset and paste the URL as you normally would, and only test that asset.

abrenoch, 2020-03-17:
Deleting then re-adding the same asset doesn't seem to have an effect:

adding:
[image]

added:
[image]

editing:
[image]

edited:
[image]

abrenoch, 2020-03-17:
Investigating this further - it appears `escape` from the `flask` library appears to be causing the issue

I added a log call here (in server.py prepare_asset_v1_2):
[image]

which produces:
[image]

ealmonte32, 2020-03-17:
> which produces:
> [image]
> @ealmonte32 seems how your replace call is in the viewer... Is this expected behavior? If the url is being corrected in the view (by my understanding when the asset is displayed) then there must have been some anticipation of the asset URL saving with `&amp;`, no?
> The problem then compounds when editing since the original '&' is still present in the url, and `flask.escape` just replaces that instances with another `&amp;` all over again

So just a quick question because I honestly dont test much in balena, I do almost all testing with raspbian-lite.. I understand that the log is showing the `&amp;`  as part of the URL, but is the actual URL loading this incorrectly? meaning, is the viewer browser itself displaying the website URL incorrectly?

I dont know flask much but I can sure see what you mean and might be right, and since I dont test on balena much like I mentioned, this might be the reason, I could def take a look and see if that might be the issue where the prepare asset parser itself is not escaping it how the viewer escapes it.

abrenoch, 2020-03-17:
That's a great point... I'll be sure to double check once I get back to the office, but while I was testing; no the provided url did not load. It seems like with your replace line, it might work after a single edit (because `&amp;amp;` appears to be the string being replaced) but I don't recall that being the case.

beemeeup, 2020-04-30:
This seems to be the same behaviour which is mentioned in #1275

abrenoch, 2020-07-02:
I want to bump this - it is a problem that needs to be resolved.

ealmonte32, 2020-07-13:
@abrenoch 
I did fix for this, PR has been requested.

closed 2020-09-20 by stale[bot]

ealmonte32, 2020-09-20:
PR has been done for this and waiting for approval.. https://github.com/Screenly/screenly-ose/pull/1344