Screenly/Anthias #1309: The screen displays the message: "Unable to load page. Problem ocurred while loading the URL"

jcsanchezrb, 2020-02-25:
Sometimes the screen displays the message: 

Unable to load page. Problem occurred while loading the URL: http://xxxxxx.yyy.zzzzz/page.html 

Could not connect: No route to host and a button with the text "Try again".

Obviously, I cannot click the button. 

How could I do that it will automatically try to reload the page?

PD: I am using Screenly OSE in Balena.

ealmonte32, 2020-03-02:
@jcsanchezrb 
Assets that for whatever reason fail to load, (whether network related or not), should reload on the next cycle and try again (if your assets are on a scheduled cycle). Can you take a picture or screenshot and post it? How does your Screeenly Schedule Overview look like? Post the URL so that I can try it. When any of the assets I have fail to load, when the scheduled time passes, and the next asset is scheduled to load, it goes on to load the next asset.

jcsanchezrb, 2020-03-03:
Hi, 

As you tell me, I hoped the performance that you define, but in that situation the only solution is reboot the device from Balena. 

I want remind you I'm using Screenly OSE run on Balena (option 4 defined in https://www.screenly.io/ose/) 

I attach a picture of the screen and also a screenshot of balena logs where you can see how the asset is tried to reload every 20 seconds.
[image]
[image]

jcsanchezrb, 2020-03-04:
Additional information.

I'm testing now option 1, download the custom image (https://github.com/Screenly/screenly-ose/issues/1309) and I get the same error "Unable to load page", when I removed the network cable and then I plug it in again; Screenly is unable to reload the url again.

I realized that if I have another active assets, a video in my case, Screenly is able to load the url when I plug the network cable in again.

ealmonte32, 2020-03-13:
I tried to load the URL you posted and I am getting error website doesnt exist.

jcsanchezrb, 2020-03-14:
Hi Emyll,

That URL was an example. In fact, it's not a public server.

I think that the problem will be the same with any other URL.

Could you please try to test I describe in my previous comment and then give me your feedback?

e6on, 2020-04-16:
Hi

I am expirencing the same issue on my Screenly devices.
Do you have any fix to this problem?

jcsanchezrb, 2020-04-16:
https://support.screenly.io/hc/en-us/articles/360009207314-What-content-types-are-supported-by-Screenly-

Websites
Screenly is capable of rendering most websites. However, if the website uses too much JavaScript, playback performance and rendering are likely to suffer. To get the best possible results, keep the website as lean as possible and optimize it for 1920x1080. 

When you want a URL to refresh each time it displays, you can make that happen by adding the URL asset two times and adding an image asset, (say a black or white screen or any other image asset), in between the URLs.

ealmonte32, 2020-04-21:
> Hi Emyll,
> 
> That URL was an example. In fact, it's not a public server.
> 
> I think that the problem will be the same with any other URL.
> 
> Could you please try to test I describe in my previous comment and then give me your feedback?

Hmm..  so "not a public server" and this is a "balena" device. Can you confirm the following for me, is your device able to reach external/public websites? and is the error Unable to load only for the internal non-public URL?

closed 2020-06-28 by stale[bot]