Screenly/Anthias #1310: Problem with USB Stick use

srops, 2020-02-25:
Hello guys,

I need your help, please.

I connected my hard drive to the Rasperry Pi.
I installed the Screenly software on the Raperry.

Now I would like to play the images wich are on the hard drive.
But still have the ability to quickly change whether it should play folder 1 or folder 2.

Is this possible?

Best regards
Manuel

ealmonte32, 2020-02-26:
Did you click on Settings > USB assets > generate file ?
You have to set it up that way.

[image]

vpetersson, 2020-02-26:
@srops Doesn't sound like a good fit. It would be a lot easier if you simply uploaded the assets and scheduled them accordingly. That said, as @ealmonte32 correctly pointed out, you can play assets from the USB interface (assuming they are mounted correctly).

srops, 2020-02-26:
Hmmm Okey I will check this @ealmonte32 

I can’t upload 5000 pictures? @vpetersson

ealmonte32, 2020-02-26:
> Hmmm Okey I will check this @ealmonte32
> 
> I can’t upload 5000 pictures? @vpetersson

Well, as far as i know, if the total size (MB or GB) of all the pictures is available in the disk drive (SD card), then the sqlite database should be able to handle those entries (rows) without a problem.. remember, uploading the assets from your USB will then be saved into the raspberry pi, so make sure there is enough space.

vpetersson, 2020-02-27:
Yep, there's no technical limitation in adding 5k pictures. It might not be the best user experience using the web interface, but you should in theory be able to do it (assuming the constraints that @ealmonte32 correctly points out).

closed 2020-05-06 by stale[bot]