Screenly/Anthias #1314: install.sh throwing error at Download upgrade_screenly.sh

rchadgray, 2020-03-02:
#### Overview of the Issue

While trying to use:
bash <(curl -sL https://www.screenly.io/install-ose.sh)

I ran into an error at this step:
TASK [screenly : Download upgrade_screenly.sh from github repository] 

An exception occurred during task execution. To see the full traceback, use -vvv. The error was: TypeError: from_buffer() cannot return the address of the raw string within a str or unicode object
fatal: [localhost]: FAILED! => {"changed": false, "module_stderr": "Traceback (most recent call last):\n  File \"/root/.ansible/tmp/ansible-tmp-1583181556.5-74078379847371/AnsiballZ_get_url.py\", line 114, in <module>\n    _ansiballz_main()\n  File \"/root/.ansible/tmp/ansible-tmp-1583181556.5-74078379847371/AnsiballZ_get_url.py\", line 106, in _ansiballz_main\n    invoke_module(zipped_mod, temp_path, ANSIBALLZ_PARAMS)\n  File \"/root/.ansible/tmp/ansible-tmp-1583181556.5-74078379847371/AnsiballZ_get_url.py\", line 49, in invoke_module\n    imp.load_module('__main__', mod, module, MOD_DESC)\n  File \"/tmp/ansible_get_url_payload_QI4Knu/__main__.py\", line 336, in <module>\n  File \"/tmp/ansible_get_url_payload_QI4Knu/ansible_get_url_payload.zip/ansible/module_utils/urls.py\", line 97, in <module>\n  File \"/usr/local/lib/python2.7/dist-packages/urllib3/contrib/pyopenssl.py\", line 46, in <module>\n    import OpenSSL.SSL\n  File \"/usr/local/lib/python2.7/dist-packages/OpenSSL/__init__.py\", line 8, in <module>\n    from OpenSSL import crypto, SSL\n  File \"/usr/local/lib/python2.7/dist-packages/OpenSSL/crypto.py\", line 15, in <module>\n    from OpenSSL._util import (\n  File \"/usr/local/lib/python2.7/dist-packages/OpenSSL/_util.py\", line 152, in <module>\n    with ffi.from_buffer(b\"\"):\nTypeError: from_buffer() cannot return the address of the raw string within a str or unicode object\n", "module_stdout": "", "msg": "MODULE FAILURE\nSee stdout/stderr for the exact error", "rc": 1}

### Environment

* Raspberry Pi Hardware Version:
Raspberry Pi 3 B+
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter)
Ethernet
* Screenly OSE Version:
Current version on GitHub

Here is the log from the terminal of the installation.
[RPiOSE.txt](https://github.com/Screenly/screenly-ose/files/4277832/RPiOSE.txt)

rchadgray, 2020-03-02:
Also while using the install.sh i used the default answers during installation.

Experimental Branch?  No
Development Brand? No
WoTT? No
Manage Network?  Yes
System upgrade? No

dig1talmad, 2020-03-03:
Same issue here, also Pi 3 B+
Started with 2020-02-05-raspbian-buster-lite image
Ran update, upgrade and then Screenly OSE install
Experimental Branch? No
Development Brand? No
WoTT? No
Manage Network? No
System upgrade? No
Let me know if there's something I can provide to help diagnose.

rchadgray, 2020-03-03:
It appears to be a bug in Ansible.

I found this, but i cannot say this is the bug that is happening when OSE uses Ansible.
https://github.com/ansible/ansible/issues/48790

ealmonte32, 2020-03-03:
Guys screenly-ose does is not yet compatible with Raspbian Buster, you need the Stretch version, here:

http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/2019-04-08-raspbian-stretch-lite.zip

rchadgray, 2020-03-03:
I am running stretch.

2019-04-08-raspbian-stretch-lite.img

ealmonte32, 2020-03-04:
> I am running stretch. Please see the terminal log i attached to the original post.
> 
> 2019-04-08-raspbian-stretch-lite.img

Ok, so then that comment goes only for @dig1talmad 

Now, about the error, I will try and reproduce your issue with the same raspbian lite image and production version as you selected.

hertell, 2020-03-04:
I can confirm that with the same hardware (PI 3 B+), the stretch-build as mentioned abow by @ealmonte32 without any upgrades, i get the same error as @rchadgray reported..

ealmonte32, 2020-03-05:
Ok, I confirmed the issue, troubleshooting and will see if I find solution for this:

`runpy.run_module(mod_name='ansible.modules.net_tools.basics.get_url`

ealmonte32, 2020-03-05:
@rchadgray @hertell 

Ok I was able to fix it, but need to do more tests on fresh raspbian installs to make sure that it is ansible.

hertell, 2020-03-05:
@ealmonte32 The /usr/local/sbin directory exists in my rasbian, so it can't be the issue in my case. 

I go now for a new try with master branch.

hertell, 2020-03-05:
No joy.. i ansible ends with this error:

`An exception occurred during task execution. To see the full traceback, use -vvv. The error was: TypeError: from_buffer() cannot return the address of the raw string wit
hin a str or unicode object
fatal: [localhost]: FAILED! => {"changed": false, "module_stderr": "Traceback (most recent call last):\n  File \"/root/.ansible/tmp/ansible-tmp-1583407858.86-75227340134
999/AnsiballZ_apt.py\", line 114, in <module>\n    _ansiballz_main()\n  File \"/root/.ansible/tmp/ansible-tmp-1583407858.86-75227340134999/AnsiballZ_apt.py\", line 106,
in _ansiballz_main\n    invoke_module(zipped_mod, temp_path, ANSIBALLZ_PARAMS)\n  File \"/root/.ansible/tmp/ansible-tmp-1583407858.86-75227340134999/AnsiballZ_apt.py\",
line 49, in invoke_module\n    imp.load_module('__main__', mod, module, MOD_DESC)\n  File \"/tmp/ansible_apt_payload_xS78es/__main__.py\", line 281, in <module>\n  File
\"/tmp/ansible_apt_payload_xS78es/ansible_apt_payload.zip/ansible/module_utils/urls.py\", line 97, in <module>\n  File \"/usr/local/lib/python2.7/dist-packages/urllib3/c
ontrib/pyopenssl.py\", line 46, in <module>\n    import OpenSSL.SSL\n  File \"/usr/local/lib/python2.7/dist-packages/OpenSSL/__init__.py\", line 8, in <module>\n    from
 OpenSSL import crypto, SSL\n  File \"/usr/local/lib/python2.7/dist-packages/OpenSSL/crypto.py\", line 15, in <module>\n    from OpenSSL._util import (\n  File \"/usr/lo
cal/lib/python2.7/dist-packages/OpenSSL/_util.py\", line 152, in <module>\n    with ffi.from_buffer(b\"\"):\nTypeError: from_buffer() cannot return the address of the ra
w string within a str or unicode object\n", "module_stdout": "", "msg": "MODULE FAILURE\nSee stdout/stderr for the exact error", "rc": 1}`

rchadgray, 2020-03-05:
I burned a new image of 2019-04-08-raspbian-stretch-lite.img and that folder does already exist.
/usr/local/sbin

I tried the master branch instead of production and it did say it was using
REPOSITORY=https://github.com/screenly/screenly-ose.git

It did install properly.  No errors.  I will try again with production and let you know.

Attached is my terminal log.
[OSE master branch.txt](https://github.com/Screenly/screenly-ose/files/4293480/OSE.master.branch.txt)

rchadgray, 2020-03-05:
I burned a new image of 2019-04-08-raspbian-stretch-lite.img

THis time i used production.  It still said it was using production
sudo -u pi ansible localhost -m git -a 'repo=https://github.com/screenly/screenly-ose.git dest=/home/pi/screenly version=production

It did error.  Attached is the terminal log.
[OSE production branch.txt](https://github.com/Screenly/screenly-ose/files/4293723/OSE.production.branch.txt)

So there is some difference between master and production.

ealmonte32, 2020-03-05:
I was just about to write that I was going to do the same test.. burning fresh image of raspbian now to check why ansible is running into that error, something probably with a module version.. comparing both of your logs now during installation to check.

scharph, 2020-03-05:
solved it for me [https://github.com/Screenly/screenly-ose/issues/1264#issuecomment-556974845](https://github.com/Screenly/screenly-ose/issues/1264#issuecomment-556974845)

ealmonte32, 2020-03-05:
@scharph doesnt make sense though, both installation methods are calling the same openssl version from requirements.txt.. actually.. I just checked, the problem might be openssl is not being called for install in the production requirements.txt , I will make PR for it , that should fix it.

ealmonte32, 2020-03-05:
I am testing with reinstalling using master branch requirements.txt

Ok, so, when I installed with the master requirements by downloading that requirements file using wget, installing first all those packages with sudo pip install -r requirements.txt, then the production installation was successful, when I then sudo pip uninstall pyopenssl, and ran the install script again, the issue arises.. so yeah, it must be the missing package from requirements in production branch.
Unless I am confused as to what is happening, that should be the culprit.. if you guys find something else let me know.

Thanks guys for bringing it to attention.

hertell, 2020-03-06:
Yep, removing the OpenSSL-directory worked, and i got all installed and Screenly running. Tomorrow i will do a new fresh install, and report back how that went :-)

hertell, 2020-03-08:
I did now a fresh install, and the setup of the development-branch completed succesfully! :-)

closed 2020-05-15 by stale[bot]