Screenly/Anthias #1315: Screenly won't sync time with NTP servers

qbicdesign, 2020-03-03:
#### Overview of the Issue

Installed from custom image 25/11/2019, then upgraded to dev branch
NTP servers will not sync

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Install Screenly
* edit /etc/systemd/timesyncd.conf entering your NTP server
* run systemctl status systemd-timesyncd.service
* raspberrypi systemd-timesyncd[1339]: Timed out waiting for reply from [ntpserver IP]:123.
### Environment

* Raspberry Pi Hardware Version: 3 model B
* Raspberry Pi Network Setup: Ethernet
* Screenly OSE Version: master@f2cbd9f

###Solution
* sudo apt install htpdate

vpetersson, 2020-03-03:
I wouldn't call this a bug necessarily. It sounds like the NTP server is blocked. Using [htpdate](https://linux.die.net/man/8/htpdate) is a great way to circumvent situations where NTP is blocked, but in most environments it isn't.

closed 2020-03-03 by vpetersson

qbicdesign, 2020-03-03:
In that case there's something in Raspbian configuration which is blocking NTP queries by default.
I queried the same NTP server from another (not raspberry) linux server and it responds correctly.
Can someone verify?

vpetersson, 2020-03-03:
No, there shouldn't be, unless you've added a firewall or similar. I don't have a Pi readily available here, but my guess would be that it's something on your network. Many enterprise environments block NTP.