Screenly/Anthias #1316: Sprint 10 no longer works with my RTMP video streams

rchadgray, 2020-03-04:
Sprint 9 (2019-04-27-Screenly-OSE-lite.img) works with my NGINX based video stream:
rtmp://10.1.0.82/live

I just tried Sprint 10 (2019-09-25-Screenly-OSE-lite.img) and it does not work with my NGINX video stream.

I use OBS to stream to an NGINX service.  Then NGINX offers up the live stream as rtmp://<ip of NGINX>/live

VLC media player can view the stream off of NGINX.  So the stream is "Good".

#### Reproduction Steps
I put the rtmp URL in the Screenly GUI and say "skip asset check".

I noticed on Sprint 9 with debugging turned on and viewing the system info tab of the web GUI it shows the command it is running to show the stream.  If i run this on Sprint 9 it works:

pi@raspberrypi:~ $ /usr/bin/timeout 50000 omxplayer rtmp://10.1.0.82/live -o hdmi
Video codec omx-h264 width 1280 height 720 profile 100 fps 30.000000
Error: Unable to open font
Audio codec aac channels 2 samplerate 44100 bitspersample 16
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 1280x720@30.00 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0

Sprint 10 gives back "have a nice day ;)"

pi@raspberrypi:~ $ /usr/bin/timeout 50000 omxplayer rtmp://10.1.0.82/live -o hdmi
have a nice day ;)

### Environment
Raspberry Pi 3 b+
Screenly Provided images for sprint 9 and 10
Ethernet wired

I have two SD cards one with Sprint 9 and one with 10 and i can load each one and replicated it every time. Let me know how i can help provide debugging information.

Chad

rchadgray, 2020-03-04:
ok... MAYBE i cannot reliably replicate it... strange.  I just put the Sprint 10 SD card in and i got the stream to work.

Let me test some more.  Sorry!  Just trying to help out!

Chad

rchadgray, 2020-03-04:
Yes so far RTMP streams work on Sprint 10.  Sorry i jumped the gun.  Another friend of mine is having trouble getting it to work so when i saw Sprint 10 did not like the stream the first time i tried it i assumed it was a bug that he was running into. If there are log files on the Sprint 10 OSE that you want to see why the stream did not work the first time let me know.

ealmonte32, 2020-03-04:
@rchadgray 

Just a suggestion, I'm sure you have your troubleshooting methods in order to try and isolate an issue, but things like checking to make sure every package is updated because remember sometimes a sprint 9 package included version 1.0 of package abcxyz, and sprint 10 included version 1.2, but maybe 1.2 introduced a bug that was recently fixed that no one else ran into because they didn't use the image as much or whatever it may be, so abcxyz releases 1.3, or 1.2.1, and updating using `sudo apt full-upgrade` might fix the issue on sprint 10 (just using examples)..

Basically, as you did the second time, do some basic troubleshooting first, and if you can replicate the issue with consistency, then of course by all means open the issue and thanks for helping out.

rchadgray, 2020-03-05:
i am closing this..

closed 2020-03-05 by rchadgray