Screenly/Anthias #1319: Unable to play MPEG stream Sprint 10

hostheimer, 2020-03-10:
Since I've installed Sprint 10 the some video steams will not run.

#### Overview of the Issue

Since I've installed Sprint10 the .mjpg and .cgi  video streams stop running.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL  http://213.4.39.225:81/mjpg/video.mjpg
* Press "On"
* Wait for the asset to show up on the screen

also url: http://88.26.233.51:8080/mjpg/video.mjpg
Also a site with a video URL: http://87.140.122.56/axis-cgi/mjpg/video.cgi does not work anymore after I installed Sprint10

### Environment

* Raspberry Pi Hardware Version: 3B+
* Raspberry Pi Network Setup:  built-in WiFi and Ethernet
* Screenly OSE Version: Sprint10

ealmonte32, 2020-03-13:
@hostheimer 
Works fine for me, which version of screenly-ose are you using?
Screenshots of my test below.

[image]
[image]
[image]

ealmonte32, 2020-03-13:
So I just realized, having a stream might be a problem/conflict if LimitCPU # of seconds is shorter than how long the stream asset is left "On" for because the uzbl-core process stays in "R" (Running) state for as long as the stream is on...  
Conflict with this PR: https://github.com/Screenly/screenly-ose/pull/1304

closed 2020-05-20 by stale[bot]