Screenly/Anthias #1325: Screenly OSE without Internet connection playing video

priya2212, 2020-04-16:
Hi,
We are making a digital signage with raspberry pi zero W. We have installed screenly OS and this is working great with network connection & is showing the live news update & local video. But when it doesn't have a network connection at boot time then it gets stuck at SSID, password screen, and the local files are not able to play.

Can you let me know if it is possible to play the video & unloaded news feed in offline mode and where is network connectivity, then it can show the live news feed and the video? and How to do this? I need to complete this assignment as early as possible

Your prompt response will be highly appreciated.

Thank you.

vpetersson, 2020-04-20:
@priya2212 Offline mode is not fully supported. Screenly was always designed to be online (or at least have local network connection). Development effort would need to go into getting this to work properly in offline mode.

priya2212, 2020-04-20:
@vpetersson, Many thanks for your kind support. So, I should assume that it's not possible for me with screenly to work in offline mode? What kind of development efforts are needed? I would be happy investing my efforts on it.

vpetersson, 2020-05-13:
> Many thanks for your kind support.
Sure thing.

> So, I should assume that it's not possible for me with screenly to work in offline mode?

Currently, it's not possible to do this, correct.

> What kind of development efforts are needed? I would be happy investing my efforts on it.

Ideally, "full offline mode" would look something like this:

We'd have to introduce a new type of network configuration. Right now we're using Balena's [wifi-connect](https://github.com/balena-io/wifi-connect). This is launched *if* there is no network connection (i.e. if there is no ethernet or wifi configured). The logic for detecting this is available [here](https://github.com/Screenly/screenly-ose/blob/master/lib/utils.py#L111-L149) and called on [here](https://github.com/Screenly/screenly-ose/blob/master/start_resin_wifi.py).

Hence, to add full offline support, we'd need to create a new "mode" that can either be invoked inside wifi-connect, or by modifying some file on disk. Once this mode has been triggered, the WiFi needs to be put into ap-mode (just like wifi-connect does) and run a DHCP server, such that you can connect to it using a phone or laptop.

Now, if the purpose is *only* to work in partial offline mode (i.e. survive outages), that can probably be achieved easier with some kind of configuration option that bypasses the wifi-connect.

closed 2020-07-19 by stale[bot]