Screenly/Anthias #1329: Possibilities to sponsor this project

loleg, 2020-05-13:
In a [recent comment](https://github.com/Screenly/screenly-ose/issues/1187#issuecomment-627827966) on #1187 the financial situation of the project was mentioned as a reason we're not seeing fresh releases of Screenly OSE. Of course, [buying a subscription](https://www.screenly.io/pricing/) is an obvious way to support the project (and something I am considering). Additionally, I would like to suggest that the OSE project looks for alternative financing mechanisms to boost the good work being done here. 

GitHub recently launched a [Sponsors](https://github.com/sponsors) initiative with a button you can easily add to this repo, and I recommend [OpenCollective](https://opencollective.com/) and [Patreon](https://www.patreon.com/) among many other ways out there to connect to transparent funding and a supportive community for an open source project.

vpetersson, 2020-05-13:
Hi @loleg - those are good points. I haven't looked closely at the GitHub Sponsors initiative yet, but it does seem interesting.

vpetersson, 2020-06-26:
@loleg This is now live **and** we have our first Gold sponsor.

More can be found here: https://github.com/sponsors/Screenly

closed 2020-06-26 by vpetersson

loleg, 2020-06-26:
Congratulations! Here's to many more ...