Screenly/Anthias #1331: Does the browser support ES6?

unitedmh, 2020-05-27:
I just have a quick question does the internal browser support ES6?

Is there a way to update the browser to chromium or something newer?

dot1q, 2020-05-29:
From my understanding, no. I think screenly uses an old open source browser that is easy to interface with programmatically, and chrome is much less open and doesn't offer an easy api. I think they have been working on it for a while, but I can't ever get the experimental branch to work correctly. I always end up transpiling my sites into ES5 and having to add polyfill. It requires a bit more testing, but with JS, it seems you can always dumb it down to older versions. It's a hard one because you can't find what console errors are thrown on screenly, so the next best thing, is using IE11 locally to see if it renders and then cross your fingers that it will load on screenly.

unitedmh, 2020-05-29:
That's what I noticed, I ended up rewriting my original code with jQuery and had no issue, kinda sucks that it uses an outdated browser!

vpetersson, 2020-06-26:
This will be solved when we have finalized the integration of the [new browser](https://github.com/Screenly/screenly-ose-webview).

closed 2020-06-26 by vpetersson