Screenly/Anthias #1332: Balenacloud + screenly not working on my Pi A+ (old ones)

richms, 2020-05-30:
I have 3 old pi A+s that are to limited to use for much, so I thought I would try them with screenlyose and balena - it worked for me in the past on a pi3 before going another way for that project.

I still had the screenly pushed to my balena so I iniitally tried just imaging the cards for the pi 1/0/0w onto some 8 gig cards and sticking them into the pis with a cheap known to work ralink USB wifi stick in it.

They come up eventually on the balena cloud but just sit there with current release being factory build.

I downloaded the latest screenly from here, and made a new application in balena called screenly and then did a balena push screenly, and it all went up to it fine, moved the pis over to the new application - still sitting on facrory release and showing the balena logo.

I got my pi4 out, downloaded and flashed the pi 4 image and put that in, and it all worked fine as expected, giving me a screenly page with the IP and I was able to put assets in and show them. But that is a waste of a pi 4 to cycle thru jpegs.

Is the old 512 meg pi A+ too crap to run balena cloud and screenly so its failing?  They do keep coming up online (vpn only) sometimes in the list of pi's in the balena devices list.

trustees629, 2020-06-09:
I have two Pi zeroes that were running on Balena. I made the mistake of upgrading them. Never could get them to work again and wasn't able to figure out a way to go back to the old version that was working. My bad for not making an image first. Anyway, I ended up flashing Rasbian Stretch Lite and installed through the bash command. It's working for me now. I have the IP addresses statically assigned and use port forwarding to access them remotely now.

vpetersson, 2020-06-26:
@richms 

> Is the old 512 meg pi A+ too crap to run balena cloud and screenly so its failing? They do keep coming up online (vpn only) sometimes in the list of pi's in the balena devices list.

it's hard to say without logs. However, the A+ is pretty dated by now.

closed 2020-09-05 by stale[bot]