Screenly/Anthias #1334: Status of the Project/Wifi connection settings

shaqaruden, 2020-06-08:
First and foremost what is the current status of this project. With nothing committed since mid-march and with the current world situation that this project has been put on hold for a while if not indefinitely.

Also I am running some older builds of screen (production@618864d) and I was wondering where screenly configures wifi and how to set a static IP. I am currently moving my devices to a new subnet with static IP's. I have attempted to modify resolv.conf but this did not work.

shaqaruden, 2020-06-08:
Fixed my network issue

vpetersson, 2020-06-26:
Hi @shaqaruden,

We are in the process of hiring a new developer to revamp OSE. We have a lot of work in progress that is not reflected in the current build. We're going to track that work [here](https://github.com/Screenly/screenly-ose/projects/7).

closed 2020-06-26 by vpetersson