Screenly/Anthias #1345: cannot modify asset (sprint 10 default)

pooh22, 2020-07-13:
#### Overview of the Issue

I've just create a fresh image on an 8GB sd card for a rpi3. I used the sprint 10 downloaded zip file.
(This is the second time, to confirm the issue I had the first time, after I modified several defaults)

The problem is that I upload a jpg image (successfully) and then try to modify it (make active, modify the end date, rename), I cannot use the "activity" switch and renaming or changing the end date results in "The operation failed. Please reload the page and try again." at the top of the screen.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On" 

this fails! Actually, anything that modifies the asset database

### Environment

* Raspberry Pi Hardware Version: rpi 3
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter): ethernet cable
* Screenly OSE Version: sprint 10 unmodified

#### Things I've tried

* check permissions of ~pi/.screenly/screenly.db
* modified the permissions to allow any user to modify screenly.db
* configure debug logging on (though I don't know exactly where to find the logs)
* enabled disable example assets (works!)
* rename example asset (works)

ealmonte32, 2020-07-13:
Something is not right here, can you reinstall using the bash command? I usually like running fresh installs from raspbian-lite rather than using the created image since the created image doesnt have the constant updates the master branch has.

pooh22, 2020-07-21:
Thanks for the answer, I tried to install raspberry OS and install using the bash script (production branch). I think it didn't make it without errors to the end, but I haven't yet had time to try something else. It will be a few weeks until I have some more time to test stuff. Should I use the master branch, as you suggest?

If the production branch equals the sprint 10 download, and if I can't get the the production branch to work. I can imagine a lot of people are frustrated after downloading the sprint10 appliance image...

I'll report back when I have tried some more things, but I don't know when that will be...

pooh22, 2020-08-16:
Today I installed the development (master) branch on raspberry OS and it worked as expected, so the problem I reported for sprint 10 didn't appear in master.

For good measure I tried sprint 10.1 as well, and it also didn't have this problem :smiley:

This issue can be closed as far as I'm concerned, thanks for fixing it!

closed 2020-11-01 by stale[bot]