Screenly/Anthias #1347: 502 Bad Gateway after OS update

neilmcintyre, 2020-07-14:
#### Overview of the Issue

OSE screenly fails after OS update

#### Reproduction Steps

Not sure if the content steps are important but here is how i reproduce

Install new player from 2019-09-25-Screenly-OSE-lite.img MD5: 72EF7973319CF3B215B8EA64E5153D8E
Connect player on lan
sign into web
set basic authentication
load content (3x jpg 1xmp4)
use keyboard to trigger console, enable SSH
use SSH to run Rasp-config, run 'update' (not update tool) which updates the underlying OS
after restart I get a connection refused on the HDMI output and a 502 bad gateway connecting to the web management interface. port 8080 is connection refused

I presume that the screenly viewer service is not running but I am not great on Linux. I am happy to gather more information as needed

### Environment

pi@SCREENLY101:~ $ sudo netstat -pln | grep :80
[sudo] password for pi:
tcp        0      0 127.0.0.1:8000          0.0.0.0:*               LISTEN                                                                                          526/nginx: master p
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN                                                                                          526/nginx: master p
pi@SCREENLY101:~ $

pi@SCREENLY101:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@SCREENLY101:~ $

rusko124, 2020-07-14:
Hey @neilmcintyre !
Could you provide logs by a `journalctl` command, please ?

neilmcintyre, 2020-07-14:
-- Logs begin at Tue 2020-07-14 14:04:51 AWST, end at Tue 2020-07-14 16:16:50 AWST. --
Jul 14 14:04:51 SCREENLY101 kernel: Booting Linux on physical CPU 0x0
Jul 14 14:04:51 SCREENLY101 kernel: Linux version 4.19.66-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1253 SMP Thu Aug 15
Jul 14 14:04:51 SCREENLY101 kernel: CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Jul 14 14:04:51 SCREENLY101 kernel: CPU: div instructions available: patching division code
Jul 14 14:04:51 SCREENLY101 kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jul 14 14:04:51 SCREENLY101 kernel: OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
Jul 14 14:04:51 SCREENLY101 kernel: Memory policy: Data cache writealloc
Jul 14 14:04:51 SCREENLY101 kernel: cma: Reserved 8 MiB at 0x33400000
Jul 14 14:04:51 SCREENLY101 kernel: On node 0 totalpages: 211968
Jul 14 14:04:51 SCREENLY101 kernel:   Normal zone: 1863 pages used for memmap
Jul 14 14:04:51 SCREENLY101 kernel:   Normal zone: 0 pages reserved
Jul 14 14:04:51 SCREENLY101 kernel:   Normal zone: 211968 pages, LIFO batch:63
Jul 14 14:04:51 SCREENLY101 kernel: random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
Jul 14 14:04:51 SCREENLY101 kernel: percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
Jul 14 14:04:51 SCREENLY101 kernel: pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
Jul 14 14:04:51 SCREENLY101 kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Jul 14 14:04:51 SCREENLY101 kernel: Built 1 zonelists, mobility grouping on.  Total pages: 210105
Jul 14 14:04:51 SCREENLY101 kernel: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbdepth=32 bcm2
Jul 14 14:04:51 SCREENLY101 kernel: Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Jul 14 14:04:51 SCREENLY101 kernel: Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Jul 14 14:04:51 SCREENLY101 kernel: Memory: 817288K/847872K available (8192K kernel code, 629K rwdata, 2176K rodata, 1024K init, 821K bss, 22392K reserved, 8192K cm
Jul 14 14:04:51 SCREENLY101 kernel: Virtual kernel memory layout:
                                        vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                        fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                                        vmalloc : 0xb4000000 - 0xff800000   (1208 MB)
                                        lowmem  : 0x80000000 - 0xb3c00000   ( 828 MB)
                                        modules : 0x7f000000 - 0x80000000   (  16 MB)
                                          .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                                          .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                                          .data : 0x(ptrval) - 0x(ptrval)   ( 630 kB)
                                           .bss : 0x(ptrval) - 0x(ptrval)   ( 822 kB)
Jul 14 14:04:51 SCREENLY101 kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jul 14 14:04:51 SCREENLY101 kernel: ftrace: allocating 26394 entries in 78 pages
Jul 14 14:04:51 SCREENLY101 kernel: rcu: Hierarchical RCU implementation.
Jul 14 14:04:51 SCREENLY101 kernel: NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jul 14 14:04:51 SCREENLY101 kernel: arch_timer: cp15 timer(s) running at 19.20MHz (phys).
Jul 14 14:04:51 SCREENLY101 kernel: clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Jul 14 14:04:51 SCREENLY101 kernel: sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Jul 14 14:04:51 SCREENLY101 kernel: Switching to timer-based delay loop, resolution 52ns
Jul 14 14:04:51 SCREENLY101 kernel: Console: colour dummy device 80x30
Jul 14 14:04:51 SCREENLY101 kernel: console [tty1] enabled
Jul 14 14:04:51 SCREENLY101 kernel: Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Jul 14 14:04:51 SCREENLY101 kernel: pid_max: default: 32768 minimum: 301
Jul 14 14:04:51 SCREENLY101 kernel: Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
Jul 14 14:04:51 SCREENLY101 kernel: Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
Jul 14 14:04:51 SCREENLY101 kernel: CPU: Testing write buffer coherency: ok
Jul 14 14:04:51 SCREENLY101 kernel: CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Jul 14 14:04:51 SCREENLY101 kernel: Setting up static identity map for 0x100000 - 0x10003c
Jul 14 14:04:51 SCREENLY101 kernel: rcu: Hierarchical SRCU implementation.
Jul 14 14:04:51 SCREENLY101 kernel: smp: Bringing up secondary CPUs ...
Jul 14 14:04:51 SCREENLY101 kernel: CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
Jul 14 14:04:51 SCREENLY101 kernel: CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
Jul 14 14:04:51 SCREENLY101 kernel: CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
Jul 14 14:04:51 SCREENLY101 kernel: smp: Brought up 1 node, 4 CPUs
Jul 14 14:04:51 SCREENLY101 kernel: SMP: Total of 4 processors activated (153.60 BogoMIPS).
lines 1-56

rusko124, 2020-07-14:
@neilmcintyre 
Please try the `journalctl > journal.txt` command and attach `journal.txt` file to this thread.

neilmcintyre, 2020-07-14:
[journal.txt](https://github.com/Screenly/screenly-ose/files/4917493/journal.txt)

neilmcintyre, 2020-07-14:
thanks Rusko

rusko124, 2020-07-14:
Okay, I've found these lines:
```
Jul 14 14:18:37 SCREENLY101 celery[3893]:   File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
Jul 14 14:18:37 SCREENLY101 celery[3893]:     __import__(name)
Jul 14 14:18:37 SCREENLY101 celery[3893]:   File "/home/pi/screenly/server.py", line 10, in <module>
Jul 14 14:18:37 SCREENLY101 celery[3893]:     import psutil
Jul 14 14:18:37 SCREENLY101 celery[3893]: ImportError: No module named psutil
```

I assume that the installation was with some issues.

Do you need to save your content or you can drop it?

neilmcintyre, 2020-07-14:
i can wipe it, i was builidng a replacement for one that did the same thing

rusko124, 2020-07-14:
@neilmcintyre 
You can try these steps:
- Download https://www.balena.io/etcher/
- Download https://github.com/Screenly/screenly-ose/releases/tag/0.18.2(the latest Screenly image)
- Use the Etcher to write the image to SD
- Run new image on your device
- Go to a settings page of UI
- Scroll to Upgrade Screenly
- Try to upgrade it :)

neilmcintyre, 2020-07-14:
ok will try with that image, though I think that's the version I am starting with. 
Will let you know how I go

ealmonte32, 2020-07-14:
> use keyboard to trigger console, enable SSH
> use SSH to run Rasp-config, run 'update' (not update tool) which updates the underlying OS
> after restart I get a connection refused on the HDMI output and a 502 bad gateway connecting to the web management interface. port 8080 is connection refused
> 

Are you trying to update the underlying OS? why not just run `sudo apt update && apt full-upgrade` ?

neilmcintyre, 2020-07-15:
actually i'm trying to replicate a failure that has already happened. My belief is that the initial failure was caused through the web management, by doing an update with the option to 'update system' as well. 

I'll use that method to update again today. The image rusko linked is the same version but has a slightly different modified date, and i don't normally use Balena etcher.

neilmcintyre, 2020-07-15:
'perform full system upgrade' displayed the following warning (might not be relevant)

[33mDEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support[0m

following full system upgrade and reboot, I am back to 502 Bad Gateway

neilmcintyre, 2020-07-15:
It's actually still outputting to the HDMI but the web management is not answering to nginx

rcostain, 2020-08-28:
Was this resolved? I ran into the same problem.

brodiekrause, 2020-08-28:
In my situation, my WiFi network was flaking out.  Once I fixed that, this problem went away.

neilmcintyre, 2020-08-29:
I had problems replicating this a third time. I think it was possibly caused by a WMV asset being loaded, which caused the service to stop responding.

vpetersson, 2020-09-01:
Another plausible reason would be failing SD cards. We've seen that a lot as the root cause for all sorts of errors over the years.

rcostain, 2020-09-16:
In our case, the error has nothing to do with nginx.
We checked the Traceback when starting Screenly, and there were Python 2.7 components missing: **pydbus**, **plutil** & **celery**.

`Traceback (most recent call last):
  File "./server.py", line 19, in <module>
    from celery import Celery
ImportError: No module named celery`

I installed each of them and Screenly came back.

This on a standard Raspian Stretch install with Screen OSE installed using the standard bash/curl script.

ealmonte32, 2020-09-17:
I tested this as well and I saw the services were all running properly, so I knew something was just misconfigured, not that you did anything, but something during this web upgrade task was changed..
@neilmcintyre  can you type this in terminal:
`./screenly/bin/enable_ssl.sh` 

I was getting the web server being forced to connect through https but did not have ssl enabled, after running this command which takes care of all for you, the screenly web server/nginx was back up and running properly with proper configuration.

ealmonte32, 2020-09-17:
> In our case, the error has nothing to do with nginx.
> We checked the Traceback when starting Screenly, and there were Python 2.7 components missing: **pydbus**, **plutil** & **celery**.
>
> `Traceback [...]

I dont understand how those packages got removed to begin with.. someone must have removed them, I have never seen any of the ansible tasks or other form of configuration remove those packages unless explicitly done... and you meant **psutil** right?

vpetersson, 2020-12-20:
This should have been fixed now.

closed 2020-12-20 by vpetersson