Screenly/Anthias #1354: [rpi3+] Raspberry Pi OS (32-bit) Lite / Buster (May 2020) : Supported ?

Noschvie, 2020-07-21:
* Raspberry Pi Hardware Version:
    Raspberry Model 3B+ Rev 1.3 Ram: 1 GB
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter)
   DHCP, IP address assigned by router definition
* Screenly OSE Version:
    latest

`bash <(curl -sL https://www.screenly.io/install-ose.sh)`
running on a rpi3B+ (with Raspberry Pi OS Lite / Buster installed from scratch) doesn't lead to a running Screenly OSE instance.

Running the latest disk image works fine. But how can I login to the OS to install an OpenVPN client ?
Thanks!

Noschvie, 2020-07-22:
`bash <(curl -sL https://www.screenly.io/install-ose.sh)`
running on a rpi3B+ (with Raspbian Stretch / Lite) works fine.

mflage, 2020-09-10:
You're contradicting yourself. Is it working or is it not?

Noschvie, 2020-09-10:
The Screenly OSE version "July 2020" is / was not running with Raspberry Pi OS Lite / Buster.

closed 2020-09-10 by Noschvie

javad87, 2020-09-10:
Anyone can port screenly on Raspberry pi 4 successfully? 

I had the same issue, I tried writing OSE image from sprint 10.1 with Etcher and splash screen did not show up

then tried option 2 with NOOBS installation and there was no Option available on installation screen for screenly...

then I tried really DIY optine 3 and installed raspbian os full latest release and then with the command run bash and their script (bash <(curl -sL https://www.screenly.io/install-ose.sh)), tried answering questions with different  NO and Yes combination installation and it did not work either...

I really need to have screenly on Rpi4, any suggestion what should I do?
When the fix will be available? kindly ask contributor to help us, otherwise I have no choice except finding another solution and I think there is no better free on-premise solution better than screenly 

During package installation I get this Error...
[image]

Hardware: 64 GB microSDcard
OS: 2020-08-20-raspios-buster-armhf-full
Screenly: sprint 10.1---2020-07-17-Screenly-OSE.zip

I also checked check-sum of the downloaded file, it was not corrupt.