Screenly/Anthias #1355: RaPi 4 Model B does not work with Sprint 10.1

Schrammgroup, 2020-07-21:
* Raspberry Pi Hardware Version: 4 Model B

* Screenly OSE Version: 10.1 (latest)

The Pi (4) does not start (no network and no display connection) with the new version. i used the 2020-07-17-Screenly-OSE-lite.img file with Rufus to copy. 
With a Pi Version 3 it works.

What should i do/ change for use with Pi 4?

Noschvie, 2020-07-22:
> With a Pi Version 3 it works.

Which OS version are you running at the RPi 3 ?

Schrammgroup, 2020-07-22:
I only used the Screenly OSE Version: 10.1 (latest). i use an empty micro sd card and copy the 2020-07-17-Screenly-OSE-lite.img file to the sd card with ruphus. This starts in raspi 3 but not in raspi 4

stevemacdonald, 2020-07-24:
Same issue here - finally tried installing on my RPi 4 as it was stated Sprint 10 now supported it, but the RPi 4 doesn't boot at all - display just goes to sleep. This same MicroSD card in my Pi 3 B+ boots fine.

Schrammgroup, 2020-07-24:
I have no solution to the problem, but if you install Raspios buster on the RPi4 and then execute the following command in the console, it works for me.

bash <(curl -sL https://www.screenly.io/install-ose.sh)

Noschvie, 2020-07-24:
> bash <(curl -sL https://www.screenly.io/install-ose.sh)

How did you answer the "y/n" questions on the RPi4 ?

Schrammgroup, 2020-07-24:
if i remember correctly all with "yes" except for the one with the developer version. i think that was the secend question. That should be "no" because i habe no display connection with the other version.

Noschvie, 2020-07-24:
> bash <(curl -sL https://www.screenly.io/install-ose.sh)

tested with RPi3B+ and RPi4 ... works fine with both.
...experimental branch? No
...development (master) branch? Yes
...install the WoTT agent...? No
...manage your network? No

vpetersson, 2020-07-27:
This is very strange. I'm away for the next few weeks, so I can't look at this. What I can say however is that I did test this image on a Raspberry Pi 4 Model B, and it booted without issues. It's a long-shot, but perhaps try re-downloading the disk image and flash it out with [Etcher](https://www.balena.io/etcher/), as it does disk image validation.

jasonkuhr, 2020-08-12:
I think the startfiles for the Pi4 are missing and thats why it won't start. Looks like drivers are missing on the boot partition. I'm just trying to get it to work.

jznofc, 2020-08-12:
I'm just get it started with changes to the boot partition. Used RaspiOS bootfiles and copyed the files from screenly-ose to it. So it starts, but for now the wifi interface it not working. Only the LAN Interface.
So I think for the start Problem @vpetersson have to rebuild the boot partition with those files added:
bcm2711-rpi-4-b
fixup4.dat
fixup4cd.dat
fixup4db.dat
fixup4x.dat
kernel7l.img
kernel8.img
start4.elf
start4cd.elf
start4db.elf
start4x.elf

I'm now trying to fix the wifi issus.

jznofc, 2020-08-12:
So for now i don't know where to implement drivers or so. I'm not that deep into the matter, but I try to install Screenly OSE with bash on RaspOS and it works fine! I think its a driver Issue.

scottdoesstuff, 2020-08-20:
I can confirm bits of this.
1. The sprint image 10.1 still contained stretch rather than buster. And I understand the 4 did5n't like that. Maybe it's just the drivers which are included in the buster images and not in stretch images. (Using a pi3 to upgrade to buster then switching across to a pi3 was not successful for me. Perhaps further indicating the driver issue.)
2. Starting with a plain buster lite image from raspberry foundation, you can successfully build a working system following the advice from **Schrammgroup** and **Noschvie** above. I performed an `sudo apt update` and `sudo apt upgrade` first. The latter essentially got me a kernel and driver update.
_note_ at this point I am using a dedicated ethernet connection but will test the wifi bits soon.

Hope this helps the conversation for others who might like to try a rudimentary setup in the near future. It can be done. The install just takes a long while instead of quicky burning an image.
I just need to figure out how to get two connected displays mirroring each other rather than acting as one big display. (I'm guessing I need to intercept the viewer startup to insert a `xrandr --output HDMI-2 --same-as HDMI-1` before the viewer grabs the X server.) Obviously independent display control would be awesome but I figure that's a rewrite in code that is substantial.

JasonNVSD, 2020-08-31:
Same here.  Pi 4 Model B boots Pi OS version August 2020, but doesn't boot Sprint 10.1.

ambriels, 2020-09-02:
About a month ago, I followed the instructions on this page, and had quite a hard time trying to get Screenly operational on my rpi4.  Continued to work at it, and I got it operational so it was working perfectly.  I decided to mess with it for some crazy reason, and screwed it all up!  However, before I screwed it all up, I made an image of my SD card, so I could burn the working image back if by chance I screwed it up.  Now, even if I burn the previously working image, it boots to the bootloader fail image, as if I didn't have an SD card inserted...  Any thoughts or ideas?

I also tried redoing it from scratch, from the current 2020-08-20 and from what I previously used (2020-07-17), and I can't get it to boot into screenly.

Any help would be appreciated!

javad87, 2020-09-10:
Could anyone  port screenly on Raspberry pi 4 successfully? 

I have the same issue, I tried writing OSE image from sprint 10.1 with balena Etcher and splash screen did not show up...

then tried option 2 with NOOBS installation and there was no Option available on installation screen for screenly...

at the end, I tried really DIY optine 3 and installed raspbian os full latest release and then with the command run bash and the script (bash <(curl -sL https://www.screenly.io/install-ose.sh)), tried answering questions with different  NO and Yes combination installation and it did not work either...

I really need to have screenly on Rpi4, any suggestion what should I do? or any work around?
When the fix will be available? kindly ask contributor to help us; 
otherwise, I have no choice except finding another solution and I think there is no better free on-premise solution better than screenly 

During package installation I get this Error...
[image]

Hardware: 64 GB microSDcard
OS: 2020-08-20-raspios-buster-armhf-full
Screenly: sprint 10.1---2020-07-17-Screenly-OSE.zip

I also checked check-sum of the downloaded file, it was not corrupt.

JasonNVSD, 2020-09-11:
After the trouble I had with the Pi 4 I gave up and replaced with them all with Pi 3's, which work fine.  Until this is resolved, can someone edit the releases page at least?  It will save people a lot of time/money/frustration.

ealmonte32, 2020-09-21:
Can someone with a spare Pi 4 give my installation/fork a try and report back the errors? I don't have a Pi4 to try and would like to finalize some fixes for the installer and pip requirements.

Please burn a fresh raspbian buster lite image from official repo: 
http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip

Then, please run this install script:

`bash <(curl -sL https://bit.ly/3637FqO)`

_The bitly URL simply resolves to https://raw.githubusercontent.com/ealmonte32/screenly-ose/master/bin/install.sh_

JasonNVSD, 2020-09-21:
> Please say Yes the development (master) branch question

This is the same as "experimental branch"?

JasonNVSD, 2020-09-21:
I'm transcribing this from the monitor:

TASK [screenly : Create screenly-server container]
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to import the required Python library (Docker SDK for Python: docker (Python >= 2.7) or docker-py (Python 2.6)) on raspberrypi's Python /usr/bin/python. Please read module documentation and install in appropriate location, for example via `pip install docker` or `pip install docker-py` (Python 2.6).  The error was: No module named ssl_match_hostname"}

PLAY RECAP
local host   : ok=50   changed=33   unreachable=0   failed=1   skipped=5   rescued=0   ignored=0

ealmonte32, 2020-09-21:
> > Please say Yes the development (master) branch question
> 
> @ealmonte32 This is the same as "experimental branch"?

No, this is not, this is just used for experimental testing of features that then goes into the master/development branch..

ealmonte32, 2020-09-21:
> TASK [screenly : Create screenly-server container]
> fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to import the required Python library (Docker SDK for Python: docker (Python >= [...]

Is this error from you selecting the experimental branch?

JasonNVSD, 2020-09-21:
> Is this error from you selecting the experimental branch?

Yes.  I'll start from scratch and do it again saying No to experimental branch.

maximilianotaverna, 2020-09-25:
Same here RPI4B etcher sprint 10.1 image and all I get is a black screen on either HDMI output. Im assuming a working image will be released soon?

ealmonte32, 2020-09-28:
@maximilianotaverna 
please see my comment here https://github.com/Screenly/screenly-ose/issues/1355#issuecomment-696388611

and try it, please select master branch/development when making your selection

JasonNVSD, 2020-09-28:
> Can someone with a spare Pi 4 give my installation/fork a try and report back the errors?

@ealmonte32 Your fork works without errors on a Pi 4.

ok=76 changed=49 unreachable=0 failed=0 skipped=22 rescued=0 ignored=0

Raspberry Model | Model 4B Revision: 1.1 Ram: 4GB Sony UK
Screenly Version | master@34e350c

ealmonte32, 2020-09-28:
> @ealmonte32 Your fork works without errors on a Pi 4.
>
> ok=76 changed=49 unreachable=0 failed=0 skipped=22 rescued=0 ignored=0
>
> Raspberry Model | Model 4B Revision: 1.1 Ram: 4GB Sony UK
> Screenly [...]

@vpetersson 
Pls see above whenever you have a chance.

ealmonte32, 2020-09-28:
> @ealmonte32 Your fork works without errors on a Pi 4.
> 
> ok=76 changed=49 unreachable=0 failed=0 skipped=22 rescued=0 ignored=0
> 
> Raspberry Model | Model 4B Revision: 1.1 Ram: 4GB Sony UK
> Screenly Version | master@34e350c

@JasonNVSD  thank you for reporting back, that's good to hear, I would love for others to try as well to confirm.

lsl061085, 2020-10-04:
> Can someone with a spare Pi 4 give my installation/fork a try and report back the errors? I don't have a Pi4 to try and would like to finalize some fixes for the installer and pip requirements.
> 
> Please burn a fresh raspbian buster lite image from official repo:
> http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip
> 
> Then, please run this install script:
> 
> `bash <(curl -sL https://bit.ly/3637FqO)`
> 
> Please say Yes the development (master) branch question, then No to the rest of questions.
> 
> _The bitly URL simply resolves to https://raw.githubusercontent.com/ealmonte32/screenly-ose/master/bin/install.sh_

Hi，I can use this method to install Screenly-ose normally on Raspberry Pi 4B. But the picture cannot be displayed in full screen when playing the picture through Screenly-ose.I tried to turn off overscan but it didn’t work. I set the default output of the Raspberry Pi to 1920x1080 output, and use 1920x1080 pictures, it still cannot be displayed in full screen

ealmonte32, 2020-10-05:
> Hi，I can use this method to install Screenly-ose normally on Raspberry Pi 4B. But the picture cannot be displayed in full screen when playing the picture through Screenly-ose. I tried to turn off overscan but it didn’t work.「disable_overscan=1 hdmi_group=1 hdmi_mode=16」 I set the default output of the Raspberry Pi to 1920x1080 output, and use 1920x1080 pictures, it still cannot be displayed in full screen

Hmm.. is this an uploaded picture? a video? how did you set up the display? using raspi-config? or boot/config.txt file?

lsl061085, 2020-10-05:
> > 嗨，我可以使用这种方法在Raspberry Pi 4B上正常安装Screenly-ose。但是通过Screenly-ose播放图片时，无法以全屏显示图片。我试图关闭过扫描，但没有成功。“ disable_overscan = 1 hdmi_group = 1 hdmi_mode = 16”我将Raspberry Pi的默认输出设置为1920x1080输出，并使用1920x1080图片，但仍无法全屏显示
> 
> 嗯..这是上传的图片吗？一段录像？您是如何设置显示器的？使用raspi-config？还是boot / config.txt文件？

Yes, this is the 1920x1080 resolution picture I uploaded. Video files can be displayed in full screen automatically. I set up the monitor by editing /boot/config.txt, my monitor is LeEco Super4 X43 Pro. The following is my config.txt configuration information:
disable_overscan=1
framebuffer_width=1920
framebuffer_height=1080
hdmi_group=1
hdmi_mode=16
dtparam=audio=on
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
gpu_mem=256
framebuffer_depth=32
framebuffer_ignore_alpha=1

ealmonte32, 2020-10-05:
I think the resolution you have is conflicting with the raspi-config generated one, for example when i set the resolution via raspi-config, the /boot/config.txt shows this:
```
hdmi_group=2
hdmi_mode=82
```
And it does not use the two framebuffer_width or framebuffer_height you forced..

Try to remove that, or simply run `sudo raspi-config` , then option 7, then option A5, you can see the last DMT Mode is 82, which corresponds to 1920x1080.

lsl061085, 2020-10-05:
> 我认为您拥有的分辨率与raspi-config生成的分辨率冲突，例如，当我通过raspi-config设置分辨率时，/ boot / config.txt显示如下：
> 
> ```
> hdmi_group=2
> hdmi_mode=82
> ```
> 
> 并且它不使用您强制使用的两个framebuffer_width或framebuffer_height。
> 
> 尝试删除它，或者简单地运行`sudo raspi-config`，然后运行选项7，然后运行选项A5，您可以看到最后一个DMT模式是82，对应于1920x1080。
Thank you for your reply. I set DMT Mode is 82 through sudo raspi-config. Then I restarted the Raspberry Pi 4 and re-tested. The 1920x1080 resolution image file still cannot be displayed in full screen. Only the 3840x2160 resolution image can be displayed in full screen. Is it because my LeEco Super4 X43 Pro TV has a 4K resolution?

ealmonte32, 2020-10-05:
@lsl061085 Yes pretty much, but you can try `hdmi_enable_4kp60=1` and custom settings:

Value | Default | Description
-- | -- | --
width | 3840 | width in pixels
height | 2160 | height in pixels
framerate | 60 | framerate in Hz
aspect | 3 | aspect ratio 1=4:3, 2=14:9, 3=16:9, 4=5:4, 5=16:10, 6=15:9
margins | 0 | 0=margins disabled, 1=margins enabled
interlace | 0 | 0=progressive, 1=interlaced
rb | 0 | 0=normal, 1=reduced blanking

Referece: https://www.raspberrypi.org/documentation/configuration/hdmi-config.md
and 
https://www.raspberrypi.org/documentation/configuration/config-txt/video.md#Custom%20Mode

Let me know how it goes.

lsl061085, 2020-10-05:
> @lsl061085 Yes pretty much, but you can try `hdmi_enable_4kp60=1` and custom settings:
>
> Value	Default	Description
> width	3840	width in pixels
> height	2160	height in pixels
> framerate	60	framerate in Hz [...]

Hi, I tried your recommended settings, but the 1920x1080 picture still cannot be displayed in full screen.
[image]
I checked with tvservice and it seems that the Raspberry Pi still outputs 4K by default.
root@raspberrypi:/home/pi# /opt/vc/bin/tvservice -s
state 0xa [HDMI CUSTOM RGB lim 16:9], 3840x2160 @ 30.00Hz, progressive

ealmonte32, 2020-10-05:
> Hi, I tried your recommended settings, but the 1920x1080 picture still cannot be displayed in full screen.
> I checked with tvservice and it seems that the Raspberry Pi still outputs 4K by default. [...]

Hmm.. I guess it is forcing 30hz.. let's try one more, you need to set the hdmi group to 87 i think..
Comment out the hdmi 4kp60=1 line for testing, and add the settings below, but remember check everywhere in the config.txt file for the hdmi_group and hdmi_mode lines and comment them out from the original so that you dont have conflicting modes..

Also according to this, it might be because we are trying to use hdmi_group 2:
https://www.raspberrypi.org/forums/viewtopic.php?t=248111#p1515586
And that guy comment out -> #dtoverlay=vc4-fkms-v3d

try: 
hdmi_cvt=3840 2160 30 3
hdmi_group=1
hdmi_mode=87
hdmi_drive=2

lsl061085, 2020-10-05:
> > 嗨，我尝试了建议的设置，但是1920x1080图片仍然无法全屏显示。
> > 我用tvservice检查过，看来Raspberry Pi默认仍然输出4K。
> > root @ raspberrypi：/ home / pi＃/ opt / vc / bin / tvservice -s
> > 状态0xa [HDMI CUSTOM RGB lim 16：9]，3840x2160 @ 30.00Hz，逐行
> 
> 嗯..我猜是强迫了30hz ..让我们再尝试一次，您需要将hdmi组设置为87，我想是..
> 注释掉hdmi 4kp60 = 1行以进行测试，并添加以下设置，但是请记住检查所有地方在hdmi_group和hdmi_mode行的config.txt文件中，并从原始行中注释掉它们，以免出现冲突的模式。
> 
> 同样根据此，这可能是因为我们试图使用hdmi_group 2：
> [https](https://www.raspberrypi.org/forums/viewtopic.php?t=248111#p1515586) ://www.raspberrypi.org/forums/viewtopic.php [?](https://www.raspberrypi.org/forums/viewtopic.php?t=248111#p1515586)
> t=248111#p1515586那个家伙注释掉了->＃dtoverlay = vc4- fkms-v3d
> 
> 尝试：
> hdmi_cvt = 3840 2160 30 3
> hdmi_group = 1
> hdmi_mode = 87
> hdmi_drive = 2
> 
> **等一下，我刚刚意识到，分辨率工作正常，但这只是图像没有全屏显示，因为图像不是显示器的分辨率。很抱歉，到现在为止，我想我们正在尝试为所有东西提供显示器是4k，而不仅仅是图像...是的，如果图像小于输出分辨率，我们需要对查看器进行编码，以扩展/拉伸显示内容以匹配输出分辨率。我将在某个时候尝试在viewer.py上执行此操作，并对其进行试验。**

Thank you for your patient guidance, currently screenly-ose does not support automatic scaling of pictures to the monitor resolution, right?

ealmonte32, 2020-10-05:
> Thank you for your patient guidance, currently screenly-ose does not support automatic scaling of pictures to the monitor resolution, right?

Correct.
I want to work on that but at the same time think about this, imagine you upload a picture that is smaller than the resolution, and if you force a scale of fit the window, this will make the image very distorted and disproportionate.. think about this example:

your original picture:
 
[image]

now, the scaled resized one:
[image]

Remember that not every image wants to be distorted and resized/stretched..
Like this example of your profile pic:
[image]

then if you make it forced resized/scaled/stretched to fit "tv", you get:
[image]

You get the idea right?
But, I understand what you would want, like a button that as you upload an image it lets you select "auto adjust: fill screen" sort of right?

lsl061085, 2020-10-05:
> > 感谢您的耐心指导，当前屏幕显示-不支持将图像自动缩放到显示器分辨率，对吗？
> 
> 正确。
> 我想解决这个问题，但与此同时，请考虑一下，假设您上传的图片小于分辨率，并且如果强制缩放比例适合窗口，这将使图像非常失真且不成比例。关于这个例子：
> 
> 您的原始图片：
> 
> [image]
> 
> 现在，缩放后的大小调整为：
> [image]
> 
> 请记住，并非每个图像都希望被扭曲和调整大小/拉伸。
> 像您的个人资料照片的示例一样：
> [image]
> 
> 那么，如果您将其强制调整大小/缩放/拉伸以适合“电视”，则会得到：
> [image]
> 
> 您的想法正确吗？
> 但是，我知道您想要什么，就像一个按钮，当您上传图像时，它可以让您选择“自动调整：填充屏幕”，对吗？

Because there are monitors with different resolutions to display the same content, but the monitor resolutions are different, such as: 1920x1080,2560x1440, 3840x2160. My idea is similar to using the office software to play PPT presentation documents on a TV monitor via USB storage, and automatically zoom the full screen display. In this way, there is no need to make two different resolution pictures to display on two monitors respectively. 「But I think it would be great to have this auto-scaling feature for users to choose whether to enable it or not. Thank you for your work」

lsl061085, 2020-10-05:
> 正确。
> 我想解决这个问题，但与此同时，请考虑一下，假设您上传的图片小于分辨率，并且如果强制缩放比例适合窗口，这将使图像非常失真且不成比例。关于这个例子：
>
> 您的原始图片：
>
> [image]
>
> 现在，缩放后的大小调整为：
> [image]
>
> 请记住，并非每个图像都希望被扭曲和调整大小/拉伸。
> 像您的个人资料照片的示例一样：
> [image] [...]

Because there are monitors with different resolutions to display the same content, but the monitor resolutions are different, such as: 1920x1080,2560x1440, 3840x2160. My idea is similar to using the office software to play PPT presentation documents on a TV monitor via USB storage, and automatically zoom the full screen display. In this way, there is no need to make two different resolution pictures to display on two monitors respectively. 「But I think it would be great to have this auto-scaling feature for users to choose whether to enable it or not. Thank you for your work」

bd8392, 2020-10-14:
> Can someone with a spare Pi 4 give my installation/fork a try and report back the errors? I don't have a Pi4 to try and would like to finalize some fixes for the installer and pip requirements. [...]

This is not working on pi 4 (it seems the docker packages can't be found)

bd8392, 2020-10-14:
Okay here comes the strange part...the script does not work when exucted before sudo -i....executed as pi user it works...

bd8392, 2020-10-14:
Website does not show correctly....rendering issue with uzb?

ealmonte32, 2020-10-18:
> Okay here comes the strange part...the script does not work when exucted before sudo -i....executed as pi user it works, but is rather painfull and time-consuiming compared to ready image...

Yes it works best as pi user because the install script was specifically designed (directories, file locations, etc) to be run as pi user and not superuser/root, while still requiring superuser privileges for installing packages/modules, etc.

..

> Website does not show correctly....rendering issue with uzb?

If a website does not currently render properly it is because of UZBL yes, only when future Screenly OSE version with Qt WebView based browser is released then it will render just like other modern browsers.

bd8392, 2020-10-18:
> version with Qt WebView based browser is released

Can we release PR for that? Because I had not a single website displayed correctly...I will help where I can and suggest chromium :-)

vpetersson, 2020-10-18:
> Can we release PR for that? Because I had not a single website displayed correctly

There's still work left for this.

> ...I will help where I can and suggest chromium :-)

No, Chromium won't cut it. There's a longer discussion on why [here](https://github.com/Screenly/screenly-ose/issues/527).

bd8392, 2020-10-18:
Okay cool :-) thanks for the info :-)

vpetersson, 2020-10-20:
Closed in favor of #1389

closed 2020-10-20 by vpetersson