Screenly/Anthias #1357: NetworkManager is always installed

jcsanchezrb, 2020-07-27:
When you click on the upgrade button under Settings tab, even though in the dialog window you do not activate the option: "Do you want Screenly to manage your network?....", at the end of the update NetworkManager has been installed and enabled as you can see in the screenshot.

This occurs with either branch (lastest or production)
[Upgrade Screenly.docx](https://github.com/Screenly/screenly-ose/files/4981487/Upgrade.Screenly.docx)

ealmonte32, 2020-09-22:
@jcsanchezrb 
There is a PR pending that addresses this problem, thank you for bringing to our attention.

@vpetersson the current PR for install.sh which includes many changes, has this included in it..
it is due to the following error in the $WEB_UPGRADE parameters:
```
  if [ "$MANAGE_NETWORK" = false ]; then
    NETWORK="y"
  elif [ "$MANAGE_NETWORK" = true ]; then
    NETWORK="n"
  else
    echo -e "Invalid -n parameter."
    exit 1
```

As you can see, the network manager is set as yes if variable was false, but for some other reason in the install script, setting to yes was causing it to also install, tested current install.sh in this PR https://github.com/Screenly/screenly-ose/pull/1346 and it is working as it should now.

vpetersson, 2020-10-21:
Cool. Thanks @ealmonte32. Closing out this ticket as I've merged your PR.

closed 2020-10-21 by vpetersson