Screenly/Anthias #1358: (Recurring) 502 Bad Gateway - nginx

swebervna, 2020-07-31:
**Overview of the Issue**
My Screenly OSE instances keep encountering this issue around a week after installing Screenly OSE. For the longest time everything worked, but lately this issue keeps popping up, even after countless reinstallations of Screenly OSE (I install via the disk image, onto my 2 Raspberry Pi's).

Here's what I get when I try accessing the web interface:
[image]

**Reproduction Steps**
Very simple for my case: start the Raspberry Pi. Nothing specific from my input causes it, it's just stuck anytime the Pi runs.

**Environment**
Raspberry Pi Hardware Version: Model 3B Revision: 1.2 Ram: 1 GB Sony UK
Raspberry Pi Network Setup: Ethernet
Screenly OSE Version: master@fcf603e

vpetersson, 2020-08-12:
The most common reason for why this would happen would be because of a broken SD card. I'd recommend that you replace it and see if that solves the issue. If that doesn't solve the issue, please commend and I'll reopen the issue.

closed 2020-08-12 by vpetersson

swebervna, 2020-09-15:
@vpetersson I tried a new SD card for one of them and I'm still experiencing the issue.