Screenly/Anthias #1359: Screenly OSE + LED Screen 336x504 px without external scalar

francovico, 2020-08-18:
Hello everybody,
I want to use Screenly OSE with a 504x336 resolution LED screen that is connected directly to a sender without a scalar proccesor.
I need to fit the image in there and start from the coordinate x = o, y = o.
Is there any way to do it?
I cannot change the screen size or the position of the screen.
Any tips for this?

scharph, 2020-08-28:
For RaspberryPi maybe the section **"Custom mode"** in [is](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md) interesting for you

vpetersson, 2020-08-28:
I don't *think* that is possible, @francovico but I could be wrong. To my knowledge, we haven't had anyone doing this.

ealmonte32, 2020-09-17:
@francovico i tested this and worked fine on VNC display.. see screenshot

[image]
[image]
[image]

closed 2020-12-20 by stale[bot]