Screenly/Anthias #1360: Random Screenly Viewer crashes

SvenVanderwegen, 2020-08-20:
### Overview of the Issue

Screenly Viewer crashes randomly every 10-15min. The pi is installed with Balena Cloud and is already updated to the latest OS en supervisor. Screenly itself is also already updated to Sprint 10.1. The problem started before the updating.

### Logs

```
20.08.20 14:09:38 (+0200) <screenly-viewer> browser found dead, restarting
20.08.20 14:09:38 (+0200) <screenly-viewer> Loading browser...
20.08.20 14:09:38 (+0200) <screenly-viewer> Viewer crashed.
20.08.20 14:09:38 (+0200) <screenly-viewer> Traceback (most recent call last):
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "viewer.py", line 593, in <module>
20.08.20 14:09:38 (+0200) <screenly-viewer>     main()
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "viewer.py", line 588, in main
20.08.20 14:09:38 (+0200) <screenly-viewer>     asset_loop(scheduler)
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "viewer.py", line 454, in asset_loop
20.08.20 14:09:38 (+0200) <screenly-viewer>     view_image(uri)
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "viewer.py", line 337, in view_image
20.08.20 14:09:38 (+0200) <screenly-viewer>     cb=lambda b: 'COMMAND_EXECUTED' in b and 'setimg' in b)
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "viewer.py", line 309, in browser_send
20.08.20 14:09:38 (+0200) <screenly-viewer>     load_browser()
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "viewer.py", line 264, in load_browser
20.08.20 14:09:38 (+0200) <screenly-viewer>     if browser:
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 334, in __len__
20.08.20 14:09:38 (+0200) <screenly-viewer>     return len(str(self))
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 370, in __str__
20.08.20 14:09:38 (+0200) <screenly-viewer>     else: return unicode(self).encode(self.call_args["encoding"])
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 373, in __unicode__
20.08.20 14:09:38 (+0200) <screenly-viewer>     if self.process and self.stdout:
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 316, in stdout
20.08.20 14:09:38 (+0200) <screenly-viewer>     self.wait()
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 295, in wait
20.08.20 14:09:38 (+0200) <screenly-viewer>     self._handle_exit_code(self.process.wait())
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 309, in _handle_exit_code
20.08.20 14:09:38 (+0200) <screenly-viewer>     self.process.stderr
20.08.20 14:09:38 (+0200) <screenly-viewer> SignalException_9: 
20.08.20 14:09:38 (+0200) <screenly-viewer> 
20.08.20 14:09:38 (+0200) <screenly-viewer>   RAN: '/usr/bin/uzbl-browser --config=- --uri=None --print-events'
20.08.20 14:09:38 (+0200) <screenly-viewer> 
20.08.20 14:09:38 (+0200) <screenly-viewer>   STDOUT:
20.08.20 14:09:38 (+0200) <screenly-viewer> EVENT [50] INSTANCE_START 50
20.08.20 14:09:38 (+0200) <screenly-viewer> EVENT [50] BUILTINS 'back forward scroll reload reload_ign_cache stop zoom_in zoom_out toggle_zoom_type uri js script toggle_status spawn sync_spawn sync_spawn_exec sh sync_sh exit search search_reverse search_clear dehilight set toggle dump_config dump_config_as_events chain print event request menu_add menu_link_add menu_image_add menu_editable_add menu_separator menu_link_separator menu_image_separator menu_editable_separator menu_remove menu_link_remove menu_image_remove menu_editable_remove hardcopy include show_inspector add_cookie delete_cookie clear_cookies download '
20.08.20 14:09:38 (+0200) <screenly-viewer> EVENT [50] VARIABLE_SET status_format str '<b>@[@TITLE]@</b> - @[@uri]@ - <span foreground="#bbb">@NAME</span>'
20.08.20 14:09:38 (+0200) <screenly-viewer> EVENT [50] VARIABLE_SET sh... (14495 more, please see e.stdout)
20.08.20 14:09:38 (+0200) <screenly-viewer> 
20.08.20 14:09:38 (+0200) <screenly-viewer>   STDERR:
20.08.20 14:09:38 (+0200) <screenly-viewer> 
20.08.20 14:09:38 (+0200) <screenly-viewer> Traceback (most recent call last):
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "viewer.py", line 593, in <module>
20.08.20 14:09:38 (+0200) <screenly-viewer>     main()
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "viewer.py", line 588, in main
20.08.20 14:09:38 (+0200) <screenly-viewer>     asset_loop(scheduler)
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "viewer.py", line 454, in asset_loop
20.08.20 14:09:38 (+0200) <screenly-viewer>     view_image(uri)
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "viewer.py", line 337, in view_image
20.08.20 14:09:38 (+0200) <screenly-viewer>     cb=lambda b: 'COMMAND_EXECUTED' in b and 'setimg' in b)
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "viewer.py", line 309, in browser_send
20.08.20 14:09:38 (+0200) <screenly-viewer>     load_browser()
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "viewer.py", line 264, in load_browser
20.08.20 14:09:38 (+0200) <screenly-viewer>     if browser:
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 334, in __len__
20.08.20 14:09:38 (+0200) <screenly-viewer>     return len(str(self))
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 370, in __str__
20.08.20 14:09:38 (+0200) <screenly-viewer>     else: return unicode(self).encode(self.call_args["encoding"])
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 373, in __unicode__
20.08.20 14:09:38 (+0200) <screenly-viewer>     if self.process and self.stdout:
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 316, in stdout
20.08.20 14:09:38 (+0200) <screenly-viewer>     self.wait()
20.08.20 14:09:38 (+0200) <screenly-viewer>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 295, in wait
20.08.20 14:09:38 (+0200) <screenly-viewer>     self._handle_exit_code(self.process.wait())
[... 88 lines cut ...]
20.08.20 14:09:38 (+0200) <screenly-viewer> EVENT [50] INSTANCE_START 50
20.08.20 14:09:38 (+0200) <screenly-viewer> EVENT [50] BUILTINS 'back forward scroll reload reload_ign_cache stop zoom_in zoom_out toggle_zoom_type uri js script toggle_status spawn sync_spawn sync_spawn_exec sh sync_sh exit search search_reverse search_clear dehilight set toggle dump_config dump_config_as_events chain print event request menu_add menu_link_add menu_image_add menu_editable_add menu_separator menu_link_separator menu_image_separator menu_editable_separator menu_remove menu_link_remove menu_image_remove menu_editable_remove hardcopy include show_inspector add_cookie delete_cookie clear_cookies download '
20.08.20 14:09:38 (+0200) <screenly-viewer> EVENT [50] VARIABLE_SET status_format str '<b>@[@TITLE]@</b> - @[@uri]@ - <span foreground="#bbb">@NAME</span>'
20.08.20 14:09:38 (+0200) <screenly-viewer> EVENT [50] VARIABLE_SET sh... (14495 more, please see e.stdout)
20.08.20 14:09:38 (+0200) <screenly-viewer> 
20.08.20 14:09:38 (+0200) <screenly-viewer>   STDERR:
20.08.20 14:09:38 (+0200) <screenly-viewer> 
20.08.20 14:09:47 (+0200) <screenly-viewer> Still continue...
20.08.20 14:09:48 (+0200) <screenly-viewer> Still continue...
20.08.20 14:09:52 (+0200) <screenly-viewer> /usr/local/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py:163: CryptographyDeprecationWarning: OpenSSL version 1.0.1 is no longer supported by the OpenSSL project, please upgrade. The next version of cryptography will drop support for it.
20.08.20 14:09:52 (+0200) <screenly-viewer>   utils.CryptographyDeprecationWarning
20.08.20 14:09:56 (+0200) <screenly-viewer> Generating asset-list...
20.08.20 14:09:56 (+0200) <screenly-viewer> Loading browser...
20.08.20 14:09:56 (+0200) <screenly-viewer> Browser loading None. Running as PID 55.
20.08.20 14:10:00 (+0200) <screenly-viewer> current url is http://screenly-server:80/splash-page
20.08.20 14:11:00 (+0200) <screenly-viewer> current url is file:///tmp/screenly_html/black_page.html
20.08.20 14:11:02 (+0200) <screenly-viewer> Generating asset-list...
```

### Environment

* Raspberry Pi Hardware Version: 3
* Raspberry Pi Network Setup: built-in WiFi
* Screenly OSE Version: Sprint 10.1
* Managed by Balena Cloud

ealmonte32, 2020-09-13:
is it only doing it with that image? is every URL/asset you add creating the crash? try enabling one by one until you get the one that causes the error/crash.

ealmonte32, 2020-09-17:
also, if you can, would you mind sending me the assets you are adding that are crashing it? i would test and see if it is an encoding issue.

closed 2020-12-20 by stale[bot]