Screenly/Anthias #1361: Weather App shows wrong location after editing

Funkem, 2020-08-28:
Screenly Weather App shows wrong location after editing play time.

#### Reproduction Steps
- my weather url with coordinates: https://weather.srly.io/?lat=51.31126851400651&lng=12.220533177482004
- insert in a new asset
- activate the asset
- the city is Markranstadt in Saxony
- now edit the asset play time, for example to "Forever" and save. Maybe also other setting such duration and so on.
- after that is the location wrong and set to munic equal to the url without coordinates: https://weather.srly.io
- it is reproducable on 2 devices

### Environment
* Raspberry Pi Hardware Version: Model 3B+ Revision: 1.3
* Raspberry Pi Network Setup: (Ethernet)
* Screenly OSE Version: production@d12df55m, updated to this version today

Please note, the error is not new in this version. I see that error earlier in this year. In my opinion first in June...

vpetersson, 2020-08-28:
It seems to be render fine for me in my browser. Is this *just* an issue in the Weather App + Screenly?

Funkem, 2020-08-28:
Yes, it is only a problem in Screenly as described. The error occurs not directly after insert in to the Screenly asset.  The location will be wrong after editing date and maybe after editing some other settions and saving the asset setting. It seems to be a problem with the saving function of the asset, which modified the url to an invalid. If I dont edit the asset, the Screenly App is working fine.

vpetersson, 2020-09-01:
Interesting. Good find. 

What is most likely going on here is that the GET parameters are being dropped on edit. @rusko124 Any thoughts on this?

ealmonte32, 2020-09-13:
@vpetersson 
I have open PR for the longest as a fix for this, it is probably because the URL contains the `&` character between the longitude and latitude. If he were to show us logs from system info, I am 99% sure that it will show the URL as this: 
`https://weather.srly.io/?lat=51.31126851400651&amp;lng=12.220533177482004`

PR fix for this: https://github.com/Screenly/screenly-ose/pull/1344

screenshot of location of his URL with my fix:
[image]

screenshot of the URL of his location without the fix:
[image]

qbicdesign, 2020-10-01:
i just posted a similar bug for Google Slides
try generating the URL with a URL shortener like bit.ly 
I think Screenly isn't parsing the switches after first ampersand  &

ealmonte32, 2020-10-02:
@qbicdesign 
My PR fix fixes all this..
I am waiting on approval from viktor.

[image]
[image]
[image]
[image]
[image]

ealmonte32, 2020-11-25:
@Funkem this issue can be closed since this was fixed in master branch a few weeks ago.

LangenT, 2021-11-01:
Hello,
same problem here. The location is always wrong after editing the Asset.
I already tried the bit.ly but after that I have two screens with the right and the wrong location. 
Best regards L.

ealmonte32, 2021-11-05:
@seifrob  @LangenT 

Can you guys give more details on the URL (look at my previous posts from many months ago), also, the log that shows the actual URL being requested would help, there is only one explanation for incorrect location, and that is the URL not having the correct parameters sent during the request which is how the website knows what city to display the info for.

ealmonte32, 2022-03-01:
@Funkem 
Can you close this issue?
Trying to cleanup resolved and stale issues.
This issue is resolved in the development branch long ago.

Funkem, 2022-03-03:
Closed, because it was solved in the dev branch. Not reviewed by myself.

closed 2022-03-03 by Funkem