Screenly/Anthias #1362: Installation from bash script doesn't work

mflage, 2020-09-10:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

Installing using 'bash <(curl -sL https://www.screenly.io/install-ose.sh)' method doesn't yield a working Screenly installation after finishing. The reason I need to use this installation method is that we need to use the Pi for something else than just Screenly.

The errors from the installation included below:

```
+ sudo -u pi ansible localhost -m git -a 'repo=https://github.com/screenly/screenly-ose.git dest=/home/pi/screenly version=production'
/usr/local/lib/python2.7/dist-packages/ansible/parsing/vault/__init__.py:41: CryptographyDeprecationWarning: Python 2 is no longer supported by the Python core team. Support for it is now deprecated in cryptography, and will be removed in a future release.
  from cryptography.exceptions import InvalidSignature
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: OSError: [Errno 13] Permission denied: '.'
localhost | FAILED! => {
    "msg": "Unexpected failure during module execution.", 
    "stdout": ""
}
+ cd /home/pi/screenly/ansible
/dev/fd/63: line 171: cd: /home/pi/screenly/ansible: No such file or directory
+ sudo -E ansible-playbook site.yml --skip-tags enable-ssl
/usr/local/lib/python2.7/dist-packages/ansible/parsing/vault/__init__.py:41: CryptographyDeprecationWarning: Python 2 is no longer supported by the Python core team. Support for it is now deprecated in cryptography, and will be removed in a future release.
  from cryptography.exceptions import InvalidSignature
ERROR! the playbook: site.yml could not be found
```

Upon reboot I see no running screenly instance.

#### Reproduction Steps

### Environment

* Raspberry Pi Hardware Version: Pi3B
* Raspberry Pi Network Setup: Ethernet
* Screenly OSE Version: The one provided by the script

Noschvie, 2020-09-10:
Did you test running "Raspbian Stretch / Lite" ?
https://github.com/Screenly/screenly-ose/issues/1354#issuecomment-662348138
https://github.com/Screenly/screenly-ose/issues/1355#issuecomment-663628978

mflage, 2020-09-10:
Yes, I used this one:

Raspberry Pi OS (32-bit) Lite
Minimal image based on Debian Buster

Noschvie, 2020-09-10:
> Raspberry Pi OS (32-bit) Lite
> Minimal image based on Debian Buster

Use Stretch instead of Buster

mflage, 2020-09-10:
I attempted to run on Stretch, but I'm then getting:

```
fatal: [localhost]: FAILED! => {"changed": false, "cmd": ["/usr/local/bin/pip2", "install", "-r", "/home/pi/screenly/requirements/requirements.txt"], "msg": "stdout: Collecting Mako==0.7.3\n  Downloading Mako-0.7.3.tar.gz (401 kB)\nCollecting MarkupSafe==1.0\n  Downloading MarkupSafe-1.0.tar.gz (14 kB)\nCollecting ansible==2.8.8\n  Downloading ansible-2.8.8.tar.gz (12.7 MB)\nCollecting celery==4.3.0\n  Downloading celery-4.3.0-py2.py3-none-any.whl (413 kB)\nCollecting certifi==2018.4.16\n  Downloading certifi-2018.4.16-py2.py3-none-any.whl (150 kB)\nCollecting cffi==1.13.2\n  Downloading cffi-1.13.2.tar.gz (460 kB)\nCollecting click==6.7\n  Downloading click-6.7-py2.py3-none-any.whl (71 kB)\nCollecting configparser==3.5.0\n  Downloading configparser-3.5.0.tar.gz (39 kB)\nCollecting cryptography==2.8\n  Downloading cryptography-2.8.tar.gz (504 kB)\n  Installing build dependencies: started\n  Installing build dependencies: still running...\n  Installing build dependencies: finished with status 'done'\n  Getting requirements to build wheel: started\n  Getting requirements to build wheel: finished with status 'done'\n    Preparing wheel metadata: started\n    Preparing wheel metadata: finished with status 'done'\nCollecting Flask-Cors==3.0.4\n  Downloading Flask_Cors-3.0.4-py2.py3-none-any.whl (13 kB)\nCollecting flask-restful-swagger-2==0.35\n  Downloading flask-restful-swagger-2-0.35.tar.gz (11 kB)\nCollecting flask-swagger-ui==3.0.12\n  Downloading flask-swagger-ui-3.0.12.tar.gz (447 kB)\nCollecting flask==1.0.2\n  Downloading Flask-1.0.2-py2.py3-none-any.whl (91 kB)\nCollecting flask_restful==0.3.6\n  Downloading Flask_RESTful-0.3.6-py2.py3-none-any.whl (25 kB)\nCollecting futures==3.0.5\n  Downloading futures-3.0.5-py2-none-any.whl (14 kB)\nCollecting gevent-websocket==0.10.1\n  Downloading gevent-websocket-0.10.1.tar.gz (18 kB)\nCollecting gevent==1.2.2\n  Downloading gevent-1.2.2.tar.gz (3.1 MB)\nCollecting gunicorn==19.8.1\n  Downloading gunicorn-19.8.1-py2.py3-none-any.whl (112 kB)\nCollecting hurry.filesize==0.9\n  Downloading hurry.filesize-0.9.tar.gz (2.8 kB)\nCollecting jinja2==2.10.1\n  Downloading Jinja2-2.10.1-py2.py3-none-any.whl (124 kB)\nCollecting kombu==4.6.3\n  Downloading kombu-4.6.3-py2.py3-none-any.whl (181 kB)\nCollecting mixpanel==4.3.2\n  Downloading mixpanel-4.3.2-py2.py3-none-any.whl (8.1 kB)\nRequirement already satisfied: netifaces==0.10.4 in /usr/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements/requirements.txt (line 23)) (0.10.4)\nCollecting psutil==5.6.7\n  Downloading psutil-5.6.7.tar.gz (448 kB)\nCollecting pyasn1==0.1.8\n  Downloading pyasn1-0.1.8.tar.gz (75 kB)\nCollecting pydbus==0.6.0\n  Downloading pydbus-0.6.0-py2.py3-none-any.whl (19 kB)\nCollecting python-dateutil==2.4.2\n  Downloading python_dateutil-2.4.2-py2.py3-none-any.whl (188 kB)\nCollecting pytz==2012d\n  Downloading pytz-2012d.tar.bz2 (166 kB)\nCollecting pyzmq==16.0.2\n  Downloading pyzmq-16.0.2.tar.gz (1.0 MB)\nCollecting pyOpenSSL==19.1.0\n  Downloading pyOpenSSL-19.1.0-py2.py3-none-any.whl (53 kB)\nCollecting redis==3.2.0\n  Downloading redis-3.2.0-py2.py3-none-any.whl (65 kB)\nCollecting requests[security]==2.22.0\n  Downloading requests-2.22.0-py2.py3-none-any.whl (57 kB)\nCollecting sh==1.08\n  Downloading sh-1.08.tar.gz (25 kB)\nCollecting six==1.11.0\n  Downloading six-1.11.0-py2.py3-none-any.whl (10 kB)\nCollecting uptime==2.0.2\n  Downloading uptime-2.0.2.tar.gz (6.3 kB)\nCollecting urllib3==1.25.7\n  Downloading urllib3-1.25.7-py2.py3-none-any.whl (125 kB)\nCollecting werkzeug==0.15.3\n  Downloading Werkzeug-0.15.3-py2.py3-none-any.whl (327 kB)\nCollecting youtube_dl>=2017.7.2\n  Downloading youtube_dl-2020.9.6-py2.py3-none-any.whl (1.8 MB)\nRequirement already satisfied: PyYAML in /usr/local/lib/python2.7/dist-packages (from ansible==2.8.8->-r /home/pi/screenly/requirements/requirements.txt (line 3)) (5.3.1)\nCollecting billiard<4.0,>=3.6.0\n  Downloading billiard-3.6.3.0.tar.gz (155 kB)\nCollecting vine>=1.3.0\n  Downloading vine-1.3.0-py2.py3-none-any.whl (14 kB)\nRequirement already satisfied: pycparser in /usr/local/lib/python2.7/dist-packages (from cffi==1.13.2->-r /home/pi/screenly/requirements/requirements.txt (line 6)) (2.20)\nRequirement already satisfied: enum34; python_version < \"3\" in /usr/local/lib/python2.7/dist-packages (from cryptography==2.8->-r /home/pi/screenly/requirements/requirements.txt (line 9)) (1.1.10)\nRequirement already satisfied: ipaddress; python_version < \"3\" in /usr/local/lib/python2.7/dist-packages (from cryptography==2.8->-r /home/pi/screenly/requirements/requirements.txt (line 9)) (1.0.23)\nCollecting itsdangerous>=0.24\n  Downloading itsdangerous-1.1.0-py2.py3-none-any.whl (16 kB)\nCollecting aniso8601>=0.82\n  Downloading aniso8601-8.0.0-py2.py3-none-any.whl (43 kB)\nCollecting greenlet>=0.4.10\n  Downloading greenlet-0.4.16.tar.gz (60 kB)\nRequirement already satisfied: setuptools in /usr/local/lib/python2.7/dist-packages (from hurry.filesize==0.9->-r /home/pi/screenly/requirements/requirements.txt (line 19)) (44.1.1)\nCollecting amqp<3.0,>=2.5.0\n  Downloading amqp-2.6.1-py2.py3-none-any.whl (48 kB)\nCollecting chardet<3.1.0,>=3.0.2\n  Downloading chardet-3.0.4-py2.py3-none-any.whl (133 kB)\nCollecting idna<2.9,>=2.5\n  Downloading idna-2.8-py2.py3-none-any.whl (58 kB)\nBuilding wheels for collected packages: Mako, MarkupSafe, ansible, cffi, configparser, cryptograph [... cut ...]  onerror)\n  File \"/usr/lib/python2.7/shutil.py\", line 247, in rmtree\n    rmtree(fullname, ignore_errors, onerror)\n  File \"/usr/lib/python2.7/shutil.py\", line 241, in rmtree\n    fullname = os.path.join(path, name)\n  File \"/usr/lib/python2.7/posixpath.py\", line 73, in join\n    path += '/' + b\nUnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 13: ordinal not in range(128)\n"}
```

An insane blob og error. Seems like it's attempting to delete some folders and failing.

mflage, 2020-09-10:
I commented out that task in the ansible playbook, let's see if it moves past that now.

javad87, 2020-09-10:
Can someone let us know if OSE is working in stable manner on Raspbeery pi4?

I received this response from Screenly support:

Hi
Thanks for reaching out.

We are, unfortunately, unable to provide one-on-one support for Screenly OSE as it is a free product. This help page will point you towards available resources:
https://support.screenly.io/hc/en-us/categories/360002606694-OSE

You should note that the Raspberry Pi 4 board has had some well publicised issues with video and wifi; with this in mind, if reliability is important for your signs, we advise that you use the 3B+ variant. This is low cost and very reliable.

Best, Nick

--
Nick Bell
Screenly, Inc.

ealmonte32, 2020-09-13:
> The directory '/home/pi/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory.

Can you remove that directory? `rm -rf /home/pi/.cache` ?

also, can you edit your `/home/pi/screenly/bin/install.sh` and find this line and add that force=yes at the end:
```
sudo -u pi ansible localhost -m git -a "repo=$REPOSITORY dest=/home/pi/screenly version=$BRANCH force=yes"
```
then run that install.sh file.

mflage, 2020-09-14:
> Can you remove that directory? `rm -rf /home/pi/.cache` ?
>
> also, can you edit your `/home/pi/screenly/bin/install.sh` and find this line and add that force=yes at the end:
>
> ```
> sudo -u pi ansible [...]

Nope, that didn't help. Same problem. The only way to get past that is by commenting out the task in the playbook.

ealmonte32, 2020-09-15:
> Nope, that didn't help. Same problem. The only way to get past that is by commenting out the task in the playbook.

You had to comment out this task: `TASK [screenly : Install pip dependencies]` for it to work? then something def out of whack happening, and it is most likely not with ansible..

Which branch are you selecting when running the install script?

Something is not allowing the packages to install properly, before running the installer again just to test, can you remove all pip packages by running this line: `pip freeze | xargs sudo pip uninstall -y` ?

or if starting fresh, can you confirm this install script being ran is on a fresh raspbian stretch image from: http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/
?
did something run or install before you did the install script? because it's not normal for your user `pi` to not have permissions to delete that cache directory if `pi` is the owner and is the user who ran the installer previously...
[image]

if you decide to just start fresh with raspbian stretch lite image and then running the bash install script, note down the exact steps you are taking with details in case it doesnt work so that i can try to reproduce the issue.

mflage, 2020-09-15:
> You had to comment out this task: `TASK [screenly : Install pip dependencies]` for it to work? then something def out of whack happening, and it is most likely not with ansible..

Correct. If I let it run once, then it has already performed the pip installation and I can then just omit that step. I'm also wondering why the ansible task is taking several minutes, while running ```pip install -r /home/pi/screenly/requirements/requirements.txt``` manually only takes a few seconds. What else is ansible doing in each of these tasks?

> Which branch are you selecting when running the install script?

Most "stable like" setup. No to everything.

> or if starting fresh, can you confirm this install script being ran is on a fresh raspbian stretch image from: http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/?

This is the one I was using for the pi3: 2019-04-08-raspbian-stretch-lite.zip

> if you decide to just start fresh with raspbian stretch lite image and then running the bash install script, note down the exact steps you are taking with details in case it doesnt work so that i can try to reproduce the issue.

I'll see what I can do. I am currently using pi4 to install now, but I'll see if I can make another stab at the 3.

mflage, 2020-09-15:
The exact same thing happens when installing on a Pi4. This is with the most recent buster image. Just booted and gotten the Pi online and then run: ```bash <(curl -sL https://www.screenly.io/install-ose.sh)``` .

So it should be very easily reproducible.

ealmonte32, 2020-09-16:
> Correct. If I let it run once, then it has already performed the pip installation and I can then just omit that step. I'm also wondering why the ansible task is taking several minutes, while running [...]

ansible doesnt do anything except the task at hand, ansible is installing all of the packages in the requirements.txt file for the first time, remember that upon first installing of python packages pip installs the source distributions and most dont have wheels and need to be compiled so your installation will need to build wheels for each one that is not in the form of a wheel package to begin with..
when you ran the pip install manually, were those packages already installed on the system? i will assume yes, if so, that is obviously going to be faster than installing from scratch (packages are already cached).

btw, how are you getting the logs to show what pip is installing during the ansible task if that doesnt normally show or even in journalctl? i dont think you are doing a normal installation, you are leaving out exact steps you are taking for your installation, and you mentioned this RPi will be used for more than just screenly so which part are you doing first, your screenly-ose install or your other thing first?

---

> I'll see what I can do. I am currently using pi4 to install now, but I'll see if I can make another stab at the 3.

I dont have an rpi4 to test with but I dont think production branch or even exp/dev branch is compatible without some modifications to the installer and the dependencies/packages.. 

---

So I grabbed one of my test rpi3b and flashed the stretch lite image, and set up localization with sudo raspi-config because i always do that first and foremost, and then i ran the installer script. (well [mine](https://raw.githubusercontent.com/Screenly/screenly-ose/526db9a96fe493c1aef188900bf5d7baf32d48d3/bin/install.sh) but its basically the same with just the two small PR changes I've requested.. but you said you tried the force option and it didn't make a difference so doesnt matter..
Anyway my results for installation and building wheels took more than 25+ mins as well so like i said that's normal for the first time you run the installer.
After the installation finished I went on to the web dashboard to add an asset because without an asset nothing will display since you are already connected via ethernet.. i forgot but i dont think the sample assets start playing automatically, i think that happens in the master (development) branch, not production...

anyway after adding a website the screen started to display it and worked as expected, so I will assume again that you are not doing the normal steps to install screenly-ose.

mflage, 2020-09-21:
This is the COMPLETE output of the installation. The command to install screenly is literally the first command entered (after bootstrapping the installation with the WPA SSID and PSK and enabling ssh headless):

https://paste.ubuntu.com/p/hJrg7KSKwY/

ealmonte32, 2020-09-21:
On your original post you said you were using rpi3b, and then later on mentioned you tried raspbian stretch lite, and that you had the same issue, and I thought we were testing on the same thing, these lines from your pastebin tell me you're using newer kernel (most likely rpi4 hardware) and debian buster, and I have not been testing on that, I have been testing on debian stretch and rpi3b...

```
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
```

If you were to try on rpi3b again with debian stretch lite, I am sure the installer would work perfectly fine.

Now, if you want to help with rpi4/buster installation then you would need to edit your own installation script and fork the screenly repo because I cannot test with rpi4 hardware and others have opened issues that cannot get the screenly image to work with rpi4 specifically as far as i know.

I will do some tests with buster and let you know.

ealmonte32, 2020-09-21:
@mflage 
So I got screenly-ose working fine on debian buster lite, can you do me a favor when you have a chance and try my forked installer and repo to see if that helps on rpi4?

`bash <(curl -sL https://bit.ly/3637FqO)`

Please report back any of the error logs like you previously did on pastebin, that would help.

diditopher, 2020-09-23:
> Can someone let us know if OSE is working in stable manner on Raspbeery pi4?

I got OSE running on Pi 4 about a year ago. Back then I used Raspbian and manually installed the development version of OSE by running `bash <(curl -sL https://www.screenly.io/install-ose.sh)`

It was running pretty stable, but I had a hell of a time getting it to work in the first place. Things may have changed since then, but I can't test, because that Pi4 has been stolen and wasn't replaced.

ealmonte32, 2020-10-02:
@mflage 
For you and anyone wanting to install screenly-ose on RPi4 please see my comment here which reports say it fixes it/works:

https://github.com/Screenly/screenly-ose/issues/1355#issuecomment-696388611

vpetersson, 2020-10-20:
I believe most if not all of these issues have been addressed in the master (development) branch already. For Pi 4/Buster specific issues, see #1389.

closed 2020-10-20 by vpetersson