Screenly/Anthias #1363: Can't add/use url asset: "Server Error: Exceeded 30 redirects."

gardar, 2020-09-15:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

Trying to display a dashboard from https://xmatters.com in screenly results in the screenly viewer crashing, I've tried both the stable/dev branches as well as the experimental branch.
I did not record the error logs the stable/dev branches caused but the one from the experimental branch is: "Server Error: Exceeded 30 redirects."
The experimental branch doesn't even let me add the asset as it doesn't have the option to skip the asset check.

You can see for yourself by using the urls: https://demo.xmatters.com and https://demo.xmatters.com/xmatters/signOn.do

I tried modifying the lib/utils.py to raise the 30 redirects "limit" to 100 but that just resulted in the same error (Server Error: Exceeded 100 redirects.) So I guess the redirects are not the root of the issue.
Did a quick google search and found out that the redirect error could be caused because of missing headers from the request, so I tried adding some valid headers but that didn't change anything.

Any thoughts?

FYI the changes I did are as follows:

```diff
--- utils.orig.py       2020-09-15 15:18:49.768538567 +0000
+++ utils.new.py        2020-09-15 15:17:53.118911433 +0000
@@ -154,26 +154,34 @@
 
     headers = {
         'User-Agent': 'Mozilla/5.0 (X11; Linux armv7l) AppleWebKit/538.15 (KHTML, like Gecko) Version/8.0 Safari/538.15',
+        'Accept-Encoding': 'gzip, deflate, sdch',
+        'Accept-Language': 'en-US,en;q=0.8',
+        'Upgrade-Insecure-Requests': '1',
+        'Accept': 'text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8',
+        'Cache-Control': 'max-age=0',
+        'Connection': 'keep-alive',
     }
+    session = requests.Session()
+    session.max_redirects=100
     try:
         if not validate_url(url):
             return False
 
-        if requests.head(
+        if session.head(
             url,
             allow_redirects=True,
             headers=headers,
-            timeout=10,
-            verify=verify
+            timeout=100,
+            verify=False,
         ).status_code in HTTP_OK:
             return False
 
-        if requests.get(
+        if session.get(
             url,
             allow_redirects=True,
             headers=headers,
-            timeout=10,
-            verify=verify
+            timeout=100,
+            verify=False,
         ).status_code in HTTP_OK:
             return False
 
@@ -217,4 +225,3 @@
     if isinstance(value, str):
         return value.decode('utf-8')
     return unicode(value)
-

```

ealmonte32, 2020-09-17:
@gardar that file is not where the redirects are checked, the file you are looking for is:
`/usr/local/lib/python2.7/dist-packages/requests/sessions.py`

But something else needs to be edited, there is almost no website I know that creates 100 redirects, some loop is occurring and not properly landing at the correct redirect location.

ealmonte32, 2020-09-17:
i think that website headers are an issue because look at the redirect location.... it literally loops back at itself... on both..
I understand that on your browser it loads fine, but the requests session is programmed to check for redirects and be allowed by default, which maybe like on your situation is not ideal, so read on to exempt config to allow this to work for you.

[image]

if you really want to disable redirects then simply change the lib/utils to false as shown below.. then restart services like so:
`sudo systemctl restart screenly-*`

[image]

closed 2020-12-20 by stale[bot]