Screenly/Anthias #1364: Stuck on boot

FedericoFedeFede, 2020-09-20:
The rpi doesn't boot after the installation through downloading the [custom image](https://support.screenly.io/hc/en-us/articles/360042164173-Getting-started-with-Screenly-OSE)

[image]

### Environment

* Raspberry Pi Hardware Version: RPI 2 Model B V1.1
* Raspberry Pi Network Setup: Ethernet
* Screenly OSE Version: [Sprint 10.1 Release](https://github.com/Screenly/screenly-ose/releases/tag/v0.18.3)

ealmonte32, 2020-09-21:
@FedericoFedeFede 

How about some logs? run command `journalctl` 
or
To see if there is an issue with the sprint 10.1 release image, could you instead please do the following:

download the official raspbian stretch lite image:
http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/2019-04-08-raspbian-stretch-lite.zip

burn that to the SD card..
then, boot it up and in the terminal, run the bash install script:
`bash <(curl -sL https://www.screenly.io/install-ose.sh)`

select Y for the development/master branch, and Y for the full system upgrade, just in case..
let me know how it goes..

vpetersson, 2020-10-20:
Closing out due to inactivity.

closed 2020-10-20 by vpetersson