Screenly/Anthias #1365: (Recurring) 502 Bad Gateway - nginx

swebervna, 2020-09-22:
Overview of the Issue:
My Screenly OSE instances keep encountering this issue around a week after installing Screenly OSE. For the longest time everything worked, but lately, this issue keeps popping up, even after countless reinstallations of Screenly OSE (I install via the disk image, onto my 2 Raspberry Pi's). Screenly OSE is working just fine, it's just the web interface that I can't access.

Here's what I get when I try accessing the web interface:
83897335-7a8d5c80-a723-11ea-93e5-5135ef26dcbd

Reproduction Steps:
Very simple for my case: start the Raspberry Pi. Nothing specific from my input causes it, it's just stuck anytime the Pi runs.

What I've tried:

Formatting Pi SD cards and flashing the disk image again (multiple times) to install
Updating Screenly OSE (every time I update, I get this error & block of code in the middle of the update process):
sudo -u pi ansible localhost -m git -a 'repo=https://github.com/screenly/screenly-ose.git dest=/home/pi/screenly version=production'
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: OSError: [Errno 13] Permission denied: '.'
localhost | FAILED! => {
"msg": "Unexpected failure during module execution.",
"stdout": ""
}
Environment:
Raspberry Pi Hardware Version: Model 3B Revision: 1.2 Ram: 1 GB Sony UK
Raspberry Pi Network Setup: Ethernet
Screenly OSE Version: master@fcf603e

ealmonte32, 2020-09-28:
@swebervna 
if you take a look at the current requirements file here https://github.com/Screenly/screenly-ose/blob/master/requirements/requirements.txt#:~:text=psutil

you can see that psutil is called upon to be installed by the ansible task, so i am pretty sure you are running into something that almost no one else is running into.. 
if you run the install script and select master branch, you should be able to get screenly-ose running normally.. although some changes are pending in pull requests, you should still be able to not get that bad gateway error, not really sure how you got it in the first place since like i mentioned psutil is part of the requirements..

swebervna, 2020-09-28:
> @swebervna
> if you take a look at the current requirements file here https://github.com/Screenly/screenly-ose/blob/master/requirements/requirements.txt#:~:text=psutil
>
> you can see that psutil is [...]

It's beyond me why I'm encountering this. I have two Raspberry Pi's and they both experience this issue after a day or two of running a fresh install of Screenly-OSE. Also, our Pi's worked for years and it was earlier this year that this started happening. My guess is some sort of incompatibility and somehow that's causing psutil to uninstall?

ealmonte32, 2020-09-28:
> It's beyond me why I'm encountering this. I have two Raspberry Pi's and they both experience this issue after a day or two of running a fresh install of Screenly-OSE. Also, our Pi's worked for years [...]

I am not sure either but this is why I usually recommend using the bash install script on a fresh [raspbian stretch lite or buster lite image](http://downloads.raspberrypi.org/raspbian_lite/images/?C=M;O=D), that way you are for sure getting the latest installation because remember any image released with packaged software becomes outdated within days already because you can never test everything before releasing it.

closed 2020-12-20 by stale[bot]