Screenly/Anthias #1366: Looking for development document

vdj69, 2020-09-27:
Hi there!
Thank you for this beautiful project and making it opensource.
I'm new to this project and I'm looking for a codebase document or application flow diagram for depth understanding and development?

ealmonte32, 2020-10-04:
@vdj69 I think this is a big task because it requires so much documentation and explanation of every part of the program, that it might not be something feasible to obtain, many moving parts.

This might be a start:
https://github.com/ealmonte32/screenly-ose/blob/master/docs/qa-checklist.md

Maybe you can just browse around the folders and see if you could learn about each part, for example start with the install script in the `bin` folder and see where that leads you to.. you will learn as you go, the install script sets up the moving parts, then you should learn how each moving part connects with each other from the viewer to the server to the processes it calls, etc.. you would need to dig deep into the code of server.py and viewer.py along with the ansible tasks that need to happen in order for everything to come together.

Kipjr, 2020-10-16:
Count me in. I would like to contribute as well. The main things I would like to know / understand:

-  How many versions are currently supported / possible when starting the installation script and which one uses which part?

- What is the relationship between the docker containers and the scripts

vpetersson, 2020-10-16:
Hi @vdj69 and @Kipjr,

Thank you for expressing interest in contributing to Screenly OSE. 

As our long-time contributor @ealmonte32 provided some great pointers above, and let me expand and elaborate on where we are and where we need help at the moment.

* Finalize the migration to Docker with the new browser that replaces UZBL
* Fix the Rasbperry Pi 4 Model B support
* Start the migration to Python 3

Before zooming in, let's start with an overview of the components:

[image]

In short, screenly-viewer is what drives the screen, and the screenly-server is what the user interacts with (vastly simplified).

Historically, all of these components were installed on the Raspberry Pi using [Ansible](https://www.ansible.com/), but the goal is to remove a lot of this logic and instead move the complexity into Docker containers to make provisioning faster and easier.

So, let's dive into the required work.

## Finalize the migration to Docker with the new browser that replaces UZBL

As @vdj69 correctly pointed out, there is no easy way to understand the architecture of the project at this point in time. The above document should help to get a big-picture overview.

To give some context, Screenly OSE started its life using a browser called [UZBL](https://www.uzbl.org/). At the time, it was a fantastic little browser. Fully scriptable and came with a full API that we could integrate with to control it. Unfortunately, it has more or less be abandoned ([last release in 2016](https://github.com/uzbl/uzbl/releases)). This unfortunately means that it is unable to keep up with modern web pages. 

To remedy this, we began the work a few years ago migrating to a new browser. After evaluating lots of options, we ended up with writing a simple wrapper on top of QT's WebView. This work can be found [here](https://github.com/Screenly/screenly-ose-webview). This allowed us to achieve full control over the browser (something like Chromium/Chrome et al wouldn't).

The work has been done, and the browser works. During the installation, you have the option to install the "experimental" version of Screenly OSE. This is how you can currently test this out. This switches to the [experimental git branch](https://github.com/Screenly/screenly-ose/tree/experimental).

The next goal is to merge this into the master, such that it will be the default installer. There is a [Pull Request](https://github.com/Screenly/screenly-ose/pull/1239) to do just this, but it has a few merge conflicts and requires a bit of testing. So, if someone wants to take on this task, that would be amazing.

## Fix the Rasbperry Pi 4 Model B support

As many people have [pointed out on the forum](https://forums.screenly.io/t/does-screenly-work-on-the-raspberry-pi-4/100/27), the Raspberry Pi 4 Model B support is broken. This is on me. I did some testing on this, but clearly not sufficient amount due to time constraints. 

I would however strongly suggest that we tackle the above issue before going down this, as it will be easier to move between Raspbian/Raspberry Pi OS once we have finalized the migration to Docker (due to less system dependencies)

##  Start the migration to Python 3

Last, but not least, we need to begin the migration to Python 3 as Python 2.7 has now reached end-of-life. This is somewhat of a PITA, but is work that needs to be done. If not, things will break. I've already had issues with dependencies failing on Python 2.7 in other projects.

## Other questions

@Kipjr let me address your questions directly:

> How many versions are currently supported / possible when starting the installation script and which one uses which part?

I'm not sure I understand the questions, but let me take a stab at it. There are three versions of Screenly OSE, and they're all tracked based on the Git branches:

| Version       | Branch     | Comment    |
| :------------- | :----------: | -----------: |
|  Developer | master   | This is where we test things    |
|  Production | production   | This is the branch disk images are built from and should be properly tested    |
|  Experimental | experimental   | This is the branch with the experimental browser version mentioned above    |

>  What is the relationship between the docker containers and the scripts

Historically, the [Bash script](https://github.com/Screenly/screenly-ose/blob/master/bin/install.sh) was the only thing needed to run Screenly OSE. As things got more complex, we added Ansible to the mix. Currently, this script does some magic, but mostly is just used to initiate an [Ansible playbook](https://github.com/Screenly/screenly-ose/blob/master/ansible/site.yml).

In a perfect world, we would be able to get rid of Ansible entirely and only use the [Docker Compose](https://github.com/Screenly/screenly-ose/blob/master/docker-compose.yml) file, but I'm not sure that is realistic. Hence the goal is rather to try to remove things from the Ansible playbook and move all the necessary logic to the Docker containers and have the entire runtime environment isolated there.

I hope all of this makes sense, and I'm happy to answer additional questions.

Also @ealmonte32 we should probably extract some of this out to a more generic developer documentation rather than hide it in this ticket :)

ealmonte32, 2020-10-16:
@vpetersson 

> Fix the Rasbperry Pi 4 Model B support
As many people have pointed out on the forum, the Raspberry Pi 4 Model B support is broken. This is on me. I did some testing on this, but clearly not sufficient amount due to time constraints.

I think creating a Sprint 10.2 release simply using Buster instead of Stretch would be a quick fix for this so that it stops issues being created about it. This would also allow more users to report RPi4+Buster issues specifically.

.

> Also @ealmonte32 we should probably extract some of this out to a more generic developer documentation rather than hide it in this ticket :)

Sounds like a plan. I'll add it to my to-do list and create a PR with separate information and markdown/readme files from your post, which btw is 👍🏻  .

Kipjr, 2020-10-17:
@vpetersson 

> The work has been done, and the browser works. During the installation, you have the option to install the "experimental" version of Screenly OSE. This is how you can currently test this out. This [...]

I've got a Raspberry Pi 3B and a Raspberry Pi 4 (4GB). What would be the recommended device to test this?

vpetersson, 2020-10-17:
@ealmonte32 

> I think creating a Sprint 10.2 release simply using Buster instead of Stretch would be a quick fix for this so that it stops issues being created about it. This would also allow more users to report RPi4+Buster issues specifically.

So this is slightly easier said than done. We're using a fork of the official tool that is used for building Raspbian images called pi-gen. The fork can be found [here](https://github.com/Screenly/pi-gen). @rusko124 was the one that set this up and I haven't really touched it myself. 

What I presume needs to be done is to merge in the changes upstream, solve all merge conflicts, and then making sure the images build properly. Since this fork also includes the full installation, I anticipate there being some kind of installation issues as we haven't fully tested the installation on Buster. 

Thus, the best step forward is likely the following:

* Download the Buster disk image and verify that the installation works well there. If not, create a PR for the issues discovered (with backward compatibility in mind)
* Merge in the upstream changes into the pi-gen repo

> Sounds like a plan. I'll add it to my to-do list and create a PR with separate information and markdown/readme files from your post, which btw is 👍🏻 .

Excellent. Thanks!

@Kipjr 

> I've got a Raspberry Pi 3B and a Raspberry Pi 4 (4GB). What would be the recommended device to test this?

Perhaps the best step is to manually install Buster on either the Pi 3 or Pi 4 and try to run the installer on there as mentioned below. It might make sense to use the Pi 4 as it will give you quicker feedback given the added CPU resources.

Kipjr, 2020-10-17:
Raspberry Pi 3 Model B Rev 1.2 (Raspbian Buster)
Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux

PLAY RECAP 
localhost                  : ok=76   changed=45   unreachable=0    failed=0    skipped=24   rescued=0    ignored=0

# QA Checklist

(This is a partial list, as we lost the official list after Forgett closed down.)

1. Run device with an enabled ethernet and wait for the splash page.✅
2. Make sure that the splash page looks correct.✅
3. Use the provided link and make sure the link forwards to web UI.✅
4. Run device w/out ethernet and wait for the hotspot page.❌(untested)
5. Make sure that the hotspot page looks correct.❌(untested)
6. Connect to exists Screenly access point and wait for the splash page.❌(untested)
7. See step 2 and step 3❌(untested)
8. Run device with an enabled ethernet and wait for the splash page.✅
2. Make sure that the splash page looks correct.✅
3. Use the provided link and make sure the link forwards to web UI.✅
4. Run device w/out ethernet and wait for the hotspot page.❌(untested)
5. Make sure that the hotspot page looks correct.❌(untested)
6. Connect to exists Screenly access point and wait for the splash page.❌(untested)
7. See step 2 and step 3❌(untested)

## Content page

8. Add assets(image, video, webpage, stream) and make them as active..✅(jpeg, mp4, google.nl) ❌(stream untested, youtube not working)
9. Make sure that these assets are shown on a screen.✅
10. Disable the assets and make sure that the screen in standby mode.✅
11. Change a duration for any asset and make sure that the screen displays it during the correct time.✅
12. Change a start date and an end date and make sure that the asset displays in correct time. ✅
13. Turn on some assets and change their order(with drag and drop). Make sure that the assets display with the correct order.✅
14. Try to change a name of any asset.✅
15. Turn on some assets. Click on Previous asset and on Next asset. Make sure that the screen changes assets on.✅
16. Click on a download button near any asset. The asset should be downloaded.✅

## Settings page

17. Setup a player name✅
18. Change a default value for the Default duration input and upload any asset. Make sure that the duration value is correct.✅
19. Change a default value for the Default streaming duration input and upload any stream. Make sure that the duration value is correct. ❌(stream untested)
20. Turn on the Show splash screen and reboot device. Make sure that the splash screen is skipped.❌(Returned Blue Screenly Logo and after that the assets.)
21. Turn on the Default assets and make sure that the assets are added to content. Turn off the Default assets and make sure that the assets are deleted from content. ❌option missing
22. Turn on the Shuffle playlist. Activate some assets and make sure that the shuffle is working.❌(untested)
23. Turn on the Use 24-hour clock. Go to the content page and make sure that a time field uses correct format.✅
24. Activate any video file with sounds and choose HDMI for the Audion output. Make sure that the sound works.✅
25. Activate any video file with sounds and choose 3.5mm jack for the Audion output. Make sure that the sound works.❌(untested)
26. Choose any format for the Date format. Go to the content page and make sure that a date field uses correct format.❌(option missing)
27. Turn on Basic auth. Reload page and make sure that works.✅
28. Click on the Get backup. Delete all assets. Click on Upload and Recover and make sure that the assets are restored.❌

- "Server" webpage unresponsivef. Required reboot. Additionally some JS Console stuff:

- Failed to load resource: the server responded with a status of 504 (Gateway Time-out)

- settings:1 Uncaught SyntaxError: Unexpected token < in JSON at position 0
    at JSON.parse (<anonymous>)
    at Function.parseJSON (jquery-1.9.1.min.js:3)
    at HTMLInputElement.fail (settings.js:67)
    at e.<computed>.<computed>._trigger (jquery-ui-1.10.1.custom.min.js:6)
    at e.<computed>.<computed>._onFail (jquery.fileupload.js:673)
    at e.<computed>.<computed>._onFail (jquery-ui-1.10.1.custom.min.js:6)
    at Object.<anonymous> (jquery.fileupload.js:715)
    at c (jquery-1.9.1.min.js:3)
    at Object.fireWith [as rejectWith] (jquery-1.9.1.min.js:3)
    at k (jquery-1.9.1.min.js:5)

29. Generate file for the USB assets. Put it and some assets on any usb stick. Turn on the stick to device. Make sure that the assets are displays on the content page. Turn on some assets and make sure that the assets display correct. Turn off the USB stick. Make sure that the assets are deleted from the content page and make sure that the screen doesn't display them. ❌ (untested)
30. Open the Upgrade Screenly and choose the next params:
*  Production
* Turn on a manage your network
* Turn off a full system upgrade

and then start the upgrade. Make sure that passed w/out errors. ❌(untested)

31. Try to reboot device and Shutdown device from the settings page. ❌(missing option)
32. Go to the System Info page and make sure that all information are correct. ❌ (not much info, would love to have a log over there)
33. Make sure that all links in the footer is correct. ✅ looks ok
34. Go to the settings page. Click on the Reset wifi and reboot your device. Make sure that the hotspot page displays. ❌(untested)

35. Connect to the device by ssh. Run `./bin/enable_ssl.sh` script. Make sure the site url uses the ssl. ✅ screenly.local certificate

vpetersson, 2020-10-17:
Good stuff @Kipjr -- looks like we are in better shape than i thought. The next step then is to try to merge in the upstream changes into pi-gen then. I'll see if i can carve out some time next week to look at this (assuming nobody beats me to it :)

Kipjr, 2020-10-18:
I've got some questions related to my earlier question about versions:

1. What is the difference between master (most recent commits) and Experimental (new browser / docker)? 

1. What is the difference between rusko124/webkit_instead_uzbl and experimental?

1. What is the most up-to-date version of the docker version?

It is not completely clear to me

vpetersson, 2020-10-18:
@Kipjr 

> What is the difference between master (most recent commits) and Experimental (new browser / docker)?

The main difference should be the new browser. There are likely changes in master that are yet to be merged into the experimental branch.

> What is the difference between rusko124/webkit_instead_uzbl and experimental?

I can't seem to find this branch. That said, experimental should be the relevant branch. It's likely an older branch.

> What is the most up-to-date version of the docker version?

That should be master.

Kipjr, 2020-10-18:
The branch in https://github.com/rusko124/screenly-ose

Kipjr, 2020-10-18:
I cloned the pull-request with #1239 and merged it locally with master. That gave me a whole other view. "Experimental" and  #1239 are completely different as far as I can see. I think it might have to do with the pull of the docker files from dockerhub.com.
[image]

vpetersson, 2020-10-19:
> The branch in https://github.com/rusko124/screenly-ose

Right, so that's likely either work in progress or abandoned. 

> I cloned the pull-request with #1239 and merged it locally with master. That gave me a whole other view. "Experimental" and #1239 are completely different as far as I can see. I think it might have to do with the pull of the docker files from dockerhub.com. This is what confused me above.
> Below are my two versions (the one of yesterday and the PR)

Right. So the last screenshot (purple) is the "new look." This is the interface you can see on the [demo site](https://ose.demo.screenlyapp.com/).

My guess is that #1239 requires a bit of work. Were you able to cleanly merge this with the upstream master? Looking at the PR, there appears to be merge conflicts (but haven't looked into how complex they are).

vpetersson, 2020-10-20:
I've also spent some cycles on writing out the [roadmap](https://github.com/Screenly/screenly-ose/wiki/Roadmap).

ealmonte32, 2020-10-20:
@Kipjr 
Just to chime in and give some light to what you noticed between branches..
More than a year ago when I was doing some troubleshooting with uzbl and trying to find workaround for uzbl and issues with rendering of modern pages, rusko suggested the experimental branch in order to try the compiled qt webview browser he was working on, but one of the things that is not explicitly mentioned is that the experimental branch has very very basic setup and just focused on the changes of the browser rather than everything else, so things such as static files, helper files, libs, etc, were not updated to reflect changes in the master branch, so there are many many commits behind between the master and the experimental since the focus was specifically to just get the browser to work.

Just thought I'd mention this so that you understand more about what happened there..

As @vpetersson mentioned, all those merge conflicts need to be carefully looked at to see which commits are specifically targeting the browser changes and which are just commits to the overall system so that the experimental branch can have both: the latest overall commits in master, and the new web browser changes, but of course combined.
This way a contributor such as yourself can do work on the experimental branch and specifically deal with the new web browser part, while at the same time not worrying about affecting the underlying system or the way it looks, etc.

vpetersson, 2020-10-20:
@ealmonte32 Good pointers. Just one fairly important point though -- the browser is tracked in a [separate repository](https://github.com/Screenly/screenly-ose-webview). While it has been a while since I looked at this, IIRC the setup script for the experimental branch simply pulls down a [release](https://github.com/Screenly/screenly-ose-webview/releases) for the browser, which should include everything needed. Thus at least that part should be fairly straight forward.

vpetersson, 2020-10-20:
I also had some spare cycles to day to try the Pi 4. I was able to get it working without issues with the master branch. There are some minor issues left, but I think we're making good progress here.

The work is tracked here:
https://github.com/Screenly/screenly-ose/issues/1389

If anyone wants to take a look at the splashscreen issue, that would be great.

ealmonte32, 2020-10-20:
>I also had some spare cycles to day to try the Pi 4. I was able to get it working without issues with the master branch.

Just to make sure some users understand, that this is because the underlying Raspbian OS version is Buster, and not current Sprint 10.1 image of Stretch.

..
>If anyone wants to take a look at the splashscreen issue, that would be great.

I dont have a Pi4 to test but which issue is this referring to? is there an open issue somewhere giving logs about this? thx.

ealmonte32, 2020-10-20:
> @ealmonte32 Good pointers. Just one fairly important point though -- the browser is tracked in a [separate repository](https://github.com/Screenly/screenly-ose-webview). While it has been a while [...]

Gotcha.
Good to know and to put on dev docs wiki so that contributors know which parts tie into which.
Thx.

vpetersson, 2020-10-21:
> I dont have a Pi4 to test but which issue is this referring to? is there an open issue somewhere giving logs about this

I am yet to test this on Pi 3 as I've focused my effort on Pi 4. Let's use [this issue](https://github.com/Screenly/screenly-ose/issues/1389) to track it. Could you try installing the development version on Buster on a Pi 3 and see if you get the same issue there? I'm curious if it is a Buster or Pi 4 issue.

The logs (so far) didn't really show anything useful and the behavior is basically a black screen instead of the splash screen.

vpetersson, 2020-10-21:
Also, if anyone wants to help out, we could really need a hand answering questions here:
https://www.raspberrypi.org/forums/viewtopic.php?f=41&t=12396

ealmonte32, 2020-10-21:
> Could you try installing the development version on Buster on a Pi 3 and see if you get the same issue there? I'm curious if it is a Buster or Pi 4 issue.

I did this already multiple times on Pi3 and Pi Zero W, Buster and dev/master branch works every time.

vpetersson, 2020-10-21:
Yep, i was able to reproduce this too. Seems to be Pi 4 related. Still no answer on the forum tho.

ealmonte32, 2020-10-21:
> Also, if anyone wants to help out, we could really need a hand answering questions here:
> https://www.raspberrypi.org/forums/viewtopic.php?f=41&t=12396

There's a lot of comments there (45 pages) which really gets confusing to answer some.

Did you mean to post this URL: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=288832&p=1745888#p1745888 ?

vpetersson, 2020-10-21:
> There's a lot of comments there (45 pages) which really gets confusing to answer some.

Yeah so this is the reason why we created the Screenly Forum. Everything is in one thread, so it gets very confusing and hard to keep track of. But yes, that is the thread. I take it only the last five or so pages are relevant. Previous to that, I think it's under control.

ealmonte32, 2020-10-21:
> Yeah so this is the reason why we created the Screenly Forum. Everything is in one thread, so it gets very confusing and hard to keep track of. But yes, that is the thread. I take it only the last [...]

I was literally writing the same thing about why we didnt post on that raspberrypi forums a link to screenly forums better so we keep track of posts and topics separately, but then I noticed you meant to send users to a different topic so I erased what i was writing 😆 

With regards to that splash screen situation, did you check the plymouth script default settings themes etc? what does the /boot/cmdline.txt and /boot/config.txt say on the RPi4 vs RPi3? Ugh i'm probably going to have to buy an RPi4, I hate not being able to do my own troubleshooting.

Kipjr, 2020-10-22:
> I was literally writing the same thing about why we didnt post on that raspberrypi forums a link to screenly forums better so we keep track of posts and topics separately, but then I noticed you [...]

I've got one RPi3 and RPi4 (4gb) here. Anything I can do? On the 3 I've got PINN installed so I can choose any OS I want

vpetersson, 2020-10-22:
@ealmonte32 
> I was literally writing the same thing about why we didnt post on that raspberrypi forums a link to screenly forums better so we keep track of posts and topics separately, but then I noticed you meant to send users to a different topic so I erased what i was writing 😆

Haha. Yeah, I have been inclined to send users to the Screenly forum too, but it can backfire for two reasons: 

* It might upset the Foundation and/or users
* The official forum drives a lot of traffic

> With regards to that splash screen situation, did you check the plymouth script default settings themes etc? what does the /boot/cmdline.txt and /boot/config.txt say on the RPi4 vs RPi3? Ugh i'm probably going to have to buy an RPi4, I hate not being able to do my own troubleshooting.

Yeah, so what I did before [posting on the regular Raspbery Pi forum](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=288832&p=1745888#p1745888) about this I tried reproducing it on a clean Buster Lite image with *just* Plymouth installed and was able to reproduce it there. Thus my gut feeling is that something is broken in Buster for the Pi 4 with regards to the splash screen.

vpetersson, 2020-10-22:
> I've got one RPi3 and RPi4 (4gb) here. Anything I can do? On the 3 I've got PINN installed so I can choose any OS I want

PINN/NOOBS et al tends to cause a mess with regards to partitioning, so I wouldn't necessarily recommend testing with it. 

That said, I would encourage more testing of the experimental branch. There's work to be done there, but just a more thoroughly test would be handy.

ealmonte32, 2020-10-22:
> Thus my gut feeling is that something is broken in Buster for the Pi 4 with regards to the splash screen.

I will definitely look into this when I start testing with the coming Pi4.

..

> That said, I would encourage more testing of the experimental branch. There's work to be done there, but just a more thoroughly test would be handy.

Agree, we should be testing this branch most no matter what Pi version we have because the most important thing right now is Dockerizing + Qt WebView browser.

vpetersson, 2020-12-20:
@Kipjr Please pull down the latest changes and give it a spin.

Kipjr, 2020-12-20:
@vpetersson , I will

closed 2022-01-09 by stale[bot]