Screenly/Anthias #1367: Boot and display assets instead of IP?

gohanman, 2020-09-30:
#### On boot screenly shows its IP rather than active asset(s)

Currently when I boot up a Screenly device it shows a screen with its IP address & stays there until I go to the web interface and manually click "Next Asset". Ideally I'd like to get rid of the IP address entirely and go straight to active assets, or at minimum start showing the active assets after some timeout without manual intervention. I have four Pis running the safe version of Screenly and they all do this so I think it's unlikely to be a hardware problem.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Install Screenly (cURL + shell script on Raspbian (buster))
* Add an asset
* Press "On"
* Reboot Pi

### Environment

* Raspberry Pi Hardware Version: Model 3B+ Revision: 1.3
* Raspberry Pi Network Setup: built-in WiFi
* Screenly OSE Version: production@d12df55

ealmonte32, 2020-10-02:
@gohanman 
Do you have the show splash screen checked off?

[image]

ealmonte32, 2020-11-25:
@gohanman can you close this issue if the show splash suggestion i made worked..

closed 2021-01-08 by vpetersson