Screenly/Anthias #1368: docker-compose.dev.yml does not build

querry43, 2020-09-30:
#### Overview of the Issue

The dev docker container does not build due to python errors.

```
Successfully installed Flask-Cors-3.0.4 Mako-0.7.3 MarkupSafe-1.0 PyYAML-5.3.1 amqp-2.6.1 aniso8601-8.0.0 ansible-2.8.8 billiard-3.6.3.0 celery-4.3.0 certifi-2018.4.16 cffi-1.13.2 chardet-3.0.4 click-6.7 configparser-3.5.0 cryptography-2.8 enum34-1.1.10 flask-1.0.2 flask-restful-0.3.6 flask-restful-swagger-2-0.35 flask-swagger-ui-3.0.12 futures-3.0.5 gevent-1.2.2 gevent-websocket-0.10.1 greenlet-0.4.17 gunicorn-19.8.1 hurry.filesize-0.9 idna-2.8 ipaddress-1.0.23 itsdangerous-1.1.0 jinja2-2.10.1 kombu-4.6.3 mixpanel-4.3.2 psutil-5.6.7 pyOpenSSL-19.1.0 pyasn1-0.1.8 pycparser-2.20 pydbus-0.6.0 python-dateutil-2.4.2 pytz-2012d pyzmq-16.0.2 redis-3.2.0 requests-2.22.0 sh-1.8 six-1.11.0 uptime-2.0.2 urllib3-1.25.7 vine-1.3.0 werkzeug-0.15.3 youtube-dl-2020.9.20
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/cli/main.py", line 75, in main
    return command.main(cmd_args)
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/cli/base_command.py", line 121, in main
    return self._main(args)
  File "/usr/lib/python2.7/contextlib.py", line 24, in __exit__
    self.gen.next()
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/cli/command_context.py", line 28, in main_context
    yield
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/contextlib2.py", line 479, in __exit__
    _reraise_with_existing_context(exc_details)
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/contextlib2.py", line 353, in _reraise_with_existing_context
    exec("raise exc_type, exc_value, exc_tb")
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/contextlib2.py", line 468, in __exit__
    if cb(*exc_details):
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/contextlib2.py", line 396, in _exit_wrapper
    return cm_exit(cm, *exc_details)
  File "/usr/lib/python2.7/contextlib.py", line 24, in __exit__
    self.gen.next()
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/utils/temp_dir.py", line 46, in global_tempdir_manager
    _tempdir_manager = old_tempdir_manager
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/contextlib2.py", line 479, in __exit__
    _reraise_with_existing_context(exc_details)
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/contextlib2.py", line 353, in _reraise_with_existing_context
    exec("raise exc_type, exc_value, exc_tb")
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/contextlib2.py", line 468, in __exit__
    if cb(*exc_details):
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/contextlib2.py", line 396, in _exit_wrapper
    return cm_exit(cm, *exc_details)
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/utils/temp_dir.py", line 175, in __exit__
    self.cleanup()
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/utils/temp_dir.py", line 199, in cleanup
    rmtree(ensure_text(self._path))
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/retrying.py", line 49, in wrapped_f
    return Retrying(*dargs, **dkw).call(f, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/retrying.py", line 212, in call
    raise attempt.get()
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/retrying.py", line 247, in get
    six.reraise(self.value[0], self.value[1], self.value[2])
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/retrying.py", line 200, in call
    attempt = Attempt(fn(*args, **kwargs), attempt_number, False)
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/utils/misc.py", line 139, in rmtree
    onerror=rmtree_errorhandler)
  File "/usr/lib/python2.7/shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "/usr/lib/python2.7/shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "/usr/lib/python2.7/shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "/usr/lib/python2.7/shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "/usr/lib/python2.7/shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "/usr/lib/python2.7/shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "/usr/lib/python2.7/shutil.py", line 241, in rmtree
    fullname = os.path.join(path, name)
  File "/usr/lib/python2.7/posixpath.py", line 73, in join
    path += '/' + b
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 4: ordinal not in range(128)
ERROR: Service 'screenly-server' failed to build: The command '/bin/sh -c curl -s https://bootstrap.pypa.io/get-pip.py | python &&     pip install -r /tmp/requirements.txt &&     pip install -r /tmp/requirements.dev.txt' returned a non-zero code: 1
```

#### Reproduction Steps

1. check out HEAD (489b59aea9989c714d8e40e8a3ac11db3689cedf at the time)

2. docker-compose -f docker-compose.dev.yml up

### Environment

* Docker version 19.03.8, build afacb8b
* Screenly OSE Version: 489b59aea9989c714d8e40e8a3ac11db3689cedf

ealmonte32, 2020-10-04:
@querry43 
Pls try your git clone and build from my fork which is a current working testing grounds for buster release and balena..
https://github.com/ealmonte32/screenly-ose
So far all is working except for some reason redis service at some point gets denied access from celery which causes the viewer to just be black screen.
Aside from that, there is nothing else on the logs that give any signs of problems.

ealmonte32, 2020-10-14:
*update*
Actually, I was testing and modifying the docker compose and dockerfiles for buster image which worked but I am trying to reduce the image sizes now so do not run your git clone of the fork yet, if you want to manually modify your own just change jessie to buster and change python imaging to python-pil, i forgot what other changes need to be made for buster but when building the image you will see it in the logs.

vpetersson, 2020-12-20:
Should be fixed now. Pull down the latest changes and try again.

closed 2020-12-20 by vpetersson