Screenly/Anthias #1369: Screenly doesn't obey switches in Google slides publish to web URL

qbicdesign, 2020-10-01:
#### Overview of the Issue

Screenly doesn't obey switches in publish to web URL, specifically URL ends with   `pub?start=true&loop=true&delayms=10000` Everything after & doesn't seem to work

#### Reproduction Steps

Steps to reproduce this issue:

* Make a presentation in Google slides
* then publish to web (File->Publish to web)
* Set timings for slides to anything other than default 3 seconds
* Set slideshoiw to loop
* Publish and copy URL into screenly assets
* watch slideshow advance every 3 seconds (instead of what you defined)
* watch slideshow get stuck on last slide and not loop - this is irregular because sometimes it does loop, but sometimes not.

### Environment

* Raspberry Pi Hardware Version: Model 3B+ Revision: 1.3 Ram: 1 GB Sony UK
* Raspberry Pi Network Setup: built-in WiFi
* Screenly OSE Version: production@d12df55
--

The issue is not with the slideshow itself, as it works perfectly in a browser.
It simply looks like screenly doesn't pass any of the switches to the browser

I found that if I use a URL shortener like bit.ly  the slideshow works fine. Loops and delays are correct.

ealmonte32, 2020-10-02:
If you want to fix yourself until this gets approved, head over to this PR https://github.com/Screenly/screenly-ose/pull/1344 and make those changes on that file to your server.py file or you can just use `wget https://raw....github..etc -O ./screenly/server.py` to download the file in the PR and replace the one in your RPi..

ealmonte32, 2020-11-25:
@qbicdesign this issue should be closed since the fix was implemented in master branch few weeks ago.

closed 2022-01-09 by stale[bot]