Screenly/Anthias #1373: Sometimes Screenly doesn't start

jcsanchezrb, 2020-10-06:
#### Overview of the Issue

Sometimes, more than I wish, when I start or reboot the Raspberry Pi, either from the web or from ssh, the Screenly services don't start and I find the request login from tty1.

The only solution is restart the raspian os from ssh.

This has happened to me on more than one device.

Why this happens?

### Environment

* Raspberry Pi Hardware Version: Model 3B Revision: 1.2 Ram: 1 GB Embest
* Raspberry Pi Network Setup: WiFi / Ethernet
* Screenly OSE Version: production@d12df55
* Raspberry Pi OS (32-bit) Lite (Version: August 2020) + bash <(curl -sL https://www.screenly.io/install-ose.sh)

ealmonte32, 2020-10-06:
@jcsanchezrb Can you give the master/dev branch a try instead? else we would need to see logs and comb through and that might be time consuming.. instead just loading up the master branch brings up any and all improvements and fixes that have been done since..  one problem is that master branch has some pending PR fixes that unless you manually apply yourself, you will end up with URL's being escaped and the `&` character not working unless you apply the PR for `server.py` fix, and unless you apply the `viewer.py` memory limits, your pi might become unreachable if you load many assets that are very heavy in terms of memory consumption because of oom_killer might not work as quick in some instances.
We need more contributors and testers so in my case I have a fork that some users can use to try fixes that are still officially pending, but would not mind trying in order to verify the fix works, etc. Just a suggestion.

jcsanchezrb, 2020-11-05:
First of all I would like to thank you very much for the effort you are making with this project.

I see that when this happens, there aren't any info under "Latest Viewer Logs" in the tab "System Info", and I wonder how I would check that this is happening and then force a new reboot of the device

ealmonte32, 2020-11-06:
Well, one of the only ways to see this and troubleshoot would be to get access to the Pi either physically or via SSH.
If the system logs are not being shown, that means the service is not running, which then requires the review of logs to see exactly why it isn't starting. I am sure it is something very simple that just needs to be adjusted.
Do you know how to access the Pi via SSH? or do you have access to it via physical keyboard?

jcsanchezrb, 2020-11-06:
Below I attach the information that both systemctl and journalctl show on screenly when this happens.

$ systemctl | grep screenly
screenly-celery.service                                                                      loaded active       running            Screenly celery worker
screenly-viewer.service                                                                      loaded deactivating stop-sigterm start Screenly Viewer
screenly-web.service                                                                         loaded active       running            Screenly Web UI
screenly-websocket_server_layer.service                                                      loaded active       running            Websocket Server layer

$ journalctl | grep screenly
nov 06 06:00:24 TBC-SRLYOSE-RECANTEO celery[571]: [2020-11-06 06:00:24,134: INFO/MainProcess] worker@screenly ready.
nov 06 06:01:58 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Start-pre operation timed out. Terminating.
nov 06 06:01:58 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Control process exited, code=killed, status=15/TERM
nov 06 06:03:29 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: State 'stop-sigterm' timed out. Killing.
nov 06 06:03:29 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Killing process 673 (xset) with signal SIGKILL.
nov 06 06:03:29 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Control process exited, code=killed, status=9/KILL
nov 06 06:03:29 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Failed with result 'timeout'.
nov 06 06:03:29 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Service RestartSec=100ms expired, scheduling restart.
nov 06 06:03:29 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Scheduled restart job, restart counter is at 1.
nov 06 06:04:59 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Start-pre operation timed out. Terminating.
nov 06 06:04:59 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Control process exited, code=killed, status=15/TERM
nov 06 06:06:29 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: State 'stop-sigterm' timed out. Killing.
nov 06 06:06:29 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Killing process 1015 (xset) with signal SIGKILL.
nov 06 06:06:29 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Control process exited, code=killed, status=9/KILL
nov 06 06:06:29 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Failed with result 'timeout'.
nov 06 06:06:30 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Service RestartSec=100ms expired, scheduling restart.
nov 06 06:06:30 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Scheduled restart job, restart counter is at 2.
nov 06 06:08:00 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Start-pre operation timed out. Terminating.
nov 06 06:08:00 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Control process exited, code=killed, status=15/TERM
nov 06 06:09:30 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: State 'stop-sigterm' timed out. Killing.
nov 06 06:09:30 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Killing process 1357 (xset) with signal SIGKILL.
nov 06 06:09:30 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Control process exited, code=killed, status=9/KILL
nov 06 06:09:30 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Failed with result 'timeout'.
nov 06 06:09:30 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Service RestartSec=100ms expired, scheduling restart.
nov 06 06:09:30 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Scheduled restart job, restart counter is at 3.
nov 06 06:11:01 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Start-pre operation timed out. Terminating.
nov 06 06:11:01 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Control process exited, code=killed, status=15/TERM
nov 06 06:12:31 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: State 'stop-sigterm' timed out. Killing.
nov 06 06:12:31 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Killing process 1697 (xset) with signal SIGKILL.
nov 06 06:12:31 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Control process exited, code=killed, status=9/KILL
nov 06 06:12:31 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Failed with result 'timeout'.
nov 06 06:12:31 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Service RestartSec=100ms expired, scheduling restart.
nov 06 06:12:31 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Scheduled restart job, restart counter is at 4.
nov 06 06:14:01 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Start-pre operation timed out. Terminating.
nov 06 06:14:01 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Control process exited, code=killed, status=15/TERM
nov 06 06:15:32 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: State 'stop-sigterm' timed out. Killing.
nov 06 06:15:32 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Killing process 2049 (xset) with signal SIGKILL.
nov 06 06:15:32 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Control process exited, code=killed, status=9/KILL
nov 06 06:15:32 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Failed with result 'timeout'.
[... cut ...]
nov 06 06:42:38 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Killing process 5218 (xset) with signal SIGKILL.
nov 06 06:42:38 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Control process exited, code=killed, status=9/KILL
nov 06 06:42:38 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Failed with result 'timeout'.
nov 06 06:42:39 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Service RestartSec=100ms expired, scheduling restart.
nov 06 06:42:39 TBC-SRLYOSE-RECANTEO systemd[1]: screenly-viewer.service: Scheduled restart job, restart counter is at 14.
............

ealmonte32, 2020-11-06:
Using grep will filter out all python logs which might be showing what the problem is.
Is there a way you can instead just re-run the installer and select the development branch? You wont lose any of your asset data.

in console type: `bash <(curl -sL https://www.screenly.io/install-ose.sh)`

Select 2 for the dev/master branch... after all is done, your RPi should be functional with screenly running.

vpetersson, 2020-11-06:
My gut feeling tells me that it's one out of two things:

* Broken SD card (that can lead to random timeouts)
* A heavy web asset

I'd start by ruling out those two.

jcsanchezrb, 2020-11-06:
I would discarded both. It occurs with more than one raspberry pi and I have replace some SD cards and the issue carry on, and it isn't an heave web asset.

This problem only occurs when the device is rebooted (either via ssh or by clicking the "Restart screen" button or because at some point it has run out of power), never when it is working.

If you tell me which files would help to show what the problem is, I'll get them.

Isn't the produccion branch the one to use in a production environment?

I will install the development branch on another device to check if the problem takes place.

ealmonte32, 2020-11-06:
@jcsanchezrb , unless we can see all logs (maybe use https://pastebin.com/ to post all and link to us here), there is no way to know what might be. I have experienced a long time ago the no logs and nothing showing on screen on production after heavy assets makes RPi consume all the memory and after oom_killer cannot help anymore because it has restarted the process too many times, it just stops the python process that would show the viewer..

Yes, please give the development branch a try, that is the recommended action for now.

jcsanchezrb, 2020-11-15:
So, i'll use development branch for now.

closed 2020-11-15 by jcsanchezrb