Screenly/Anthias #1375: Better font for passwords

shaqaruden, 2020-10-06:
Password for wifi on the hotspot splash screen are difficult to read. I would recommend switching to a monospaced font like Fira Code, DejaVu Sans Mono, or Source Code Pro.

Though a issue was resolved for password readability before still not knowing is l is a upper case i or a lower case L is an issue and the characters are too close together

ealmonte32, 2020-10-07:
@shaqaruden 
Just wondering if you could test this real quick, if you could get to the terminal or ssh into your pi, would you mind editing this file on the screenshot and adding that line highlighted in red.. then restart the pi and see if it changes the font of the resin wifi hotspot info.. I just want to see how much difference it makes, so attach a before and after if you can.

[image]

ealmonte32, 2020-10-14:
@shaqaruden 
The lowercase letter `i` was removed from the ppp_letters list of available characters, thus there should no longer be any need to change fonts since there is no other character that has similarities with another.
I think this issue can be closed.

ealmonte32, 2020-11-03:
@shaqaruden just a reminder to close this, thx

shaqaruden, 2020-11-03:
\

closed 2020-11-03 by shaqaruden