Screenly/Anthias #1376: Reduce size of screenly containers on balena

ealmonte32, 2020-10-07:
I know there is a current work in progress on balena side to reduce their busybox/node/host memory consumption, it is crazy that with all containers/services stopped, the host OS on balena takes up 400mb+ of memory .... that is very inefficient and causes the Pi to run slower than it normally does on Pi not running on balena which in turn causes the viewer to restart itself after too much memory has been taken up with fewer assets than normally. Not sure if this is only occurring on certain versions of BalenaOS or not (probably), but I still think that regardless of that, the containers are very big.

To alleviate some of the burden on the Pi, I think some of the services/containers should be built with alpine, such as redis, websocket, and celery (i know as of right now it is only available on the testing/edge version of alpine), as this contributor started to do already: https://github.com/Screenly/screenly-ose/pull/1299 , but as you can see, it went stale, and understandably due to only one person working on it.

If anyone can help tackle this for screenly-ose container size reduction with using multistage builds and alpine, it would be greatly appreciated. As of right now, I am working and testing on my forked version so as to not mess with the current official master branch. If anyone knows how to work with multistage docker container builds, alpine, and has successfully implemented python on it and can help, either comment here or make some PR requests over [here](https://github.com/ealmonte32/screenly-ose/tree/master/docker) at my fork so that it could be tested prior to making the PR here on official master/dev branch. So far I think I got redis on alpine working simply because is the easiest to implement, current redis container image size on pi zero is 80MB and pi3 68MB.

Thanks in advance for any/all contributions.

[image]

[image]

[image]

escoand, 2020-10-07:
@ealmonte32 I can see you are quite busy with implementing new and optimizing existing features. But do I understand you right, that you're also not able to merge your work? This was my problem with the mentioned stale PR and also an other. At some point I had the feeling nobody was here to answer questions and help to get it into master. So I stopped working on it...

ealmonte32, 2020-10-07:
@escoand , while it is unusual for PR requests to take months to be reviewed and merged, remember this year was very bad (still is) and resources are scarce so the priorities have shifted. I'm not 100% but I think rusko is mainly focusing on python3 and newer browser version of screenly-ose, and I am sure that there is an explanation for the lack of response so I will just continue to make edits on my fork so that whenever things do get back to normal, all the code updates/changes don't go down the drain.

escoand, 2020-10-07:
I do understand this completely. I was wondering who is responsible for the project at the end of the day. If it depends on one person (who has to do what his employer pays him for) the whole thing could stop just because of this one person. And this was my impression so far...

ealmonte32, 2020-10-07:
At the end of the day it all falls on the founder vpetersson.
As a volunteer support I have not experienced such delay in merging PRs that are vetted, tested, and good to go like this before, but again, resources during Covid-19 have shifted so I guess we need to be understanding for a while.. hopefully the responsibilities of testing OSE PRs could be delegated to others who are already part of the team and want to help because I think many potential clients for the commercial version do want to give screenly OSE a test drive first and see the potential it has for their business, etc, so OSE is sort of important, not to mention a testing platform for features and ideas that could be implemented on commercial one.

vpetersson, 2020-10-08:
Hi @ealmonte32 and @escoand. 

First, let me just say thank you to your both for your ongoing support. We greatly appreciate it. Let me also be the first one to say that I'm not happy about the delay it takes to get PRs merged. I want this to be a pleasant and swift experience that makes it easy for new contributors to get involved.

However, as @ealmonte32 correctly pointed out, this year has been...interesting. Our goal earlier this year was to bring on a new (paid) developer to work at least part-time on OSE. 

Unfortunately due to the uncertainty and COVID, we had to prioritize additional resources to the commercial version. We hope that over the coming months, there will be resources in the budget available for bringing someone onboard to OSE.  Realistically, that might end up being delayed until Q1. 

As @ealmonte32 also correctly pointed out, OSE is strategically important to us. The goal has all along been that we use this as a testing ground where we can learn together with the community, while also contributing back with a useful product.

That said, we have still made some improvements to OSE over the last few months, so it's not fully stale.

I hope this all makes sense.

Now to the questions...

> but I still think that regardless of that, the containers are very big.

Yes, I agree. They need to be slimmed down. However, the first step is to move everything to Docker, and only then we will start optimizing. It's very common in software project to set up ambitious goals, only to be overwhelmed by the scope and give up. Thus I want to take baby steps towards the end goal instead of trying to do it all in one big step.

> To alleviate some of the burden on the Pi, I think some of the services/containers should be built with alpine, such as redis, websocket, and celery

Yes, agreed. This is the plan.

> If anyone knows how to work with multistage docker container builds, alpine, and has successfully implemented python on it and can help,

I have decent experience working with multi-stage Docker builds. It's a bit messy to do with Python, as you need copy over the binaries from all libraries (rather than having a single binary like in Go). That said, it is possible, as we're doing this in the commercial version of Screenly for the back-end.

> So far I think I got redis on alpine working simply because is the easiest to implement, current redis container image size on pi zero is 80MB and pi3 68MB.

For Redis, it probably makes more sense to use the [official Redis image](https://hub.docker.com/_/redis/), as it already is available as an Alpine version.

Thanks!

ealmonte32, 2020-10-18:
Closing this since we now know that the main focus is on compartmentalizing the "images" as docker containers instead.

closed 2020-10-18 by ealmonte32