Screenly/Anthias #1381: Sprint 10.1 Not working with RPI4 Model B

shaqaruden, 2020-10-08:
#### Overview of the Issue

When attempting to boot a sd card with Sprint 10.1 but nothing happens no display. Boot sd card with raspbian lite and it works fine.

#### Reproduction Steps

* Download Sprint 10.1
* Flash SD card with Sprint 10.1 using Etcher
* Insert SD card into pi
* Attempt to boot

### Environment

* Raspberry Pi Hardware Version: RPI4 Model B
* Raspberry Pi Network Setup: built-in WiFi
* Screenly OSE Version: Sprint 10.1

ealmonte32, 2020-10-08:
This might be related to a missing module, could you paste the results from this cmd:
`systemctl status screenly-*`

And also, can you send some logs from `journalctl` , I explicitly did not filter the results for python or systemd beause I want to see all errors. Maybe use https://pastebin.com/ so that we keep the post clean.

shaqaruden, 2020-10-09:
The device doesn’t boot at all.

ealmonte32, 2020-10-09:
I think I know what the issue is, please try the instructions I give here: https://github.com/Screenly/screenly-ose/issues/1355#issuecomment-696388611

Let me know how it goes after that, it should work, which is why we think Sprint 10.1 compatibility with RPi4 needs to be updated.

If that doesnt work, I know at least the OS will be running, so you could get logs that way, so you need to either SSH (placing an empty file named `ssh` into the SD cards /boot/ directory) or press ctrl+alt+f1 on the keyboard attached to the Pi, to access the console/terminal in order to run commands whether screenly boots or not, if the device in general doesnt boot (meaning the operating system) then that does not deal with screenly, although I doubt this is an actual OS issue.

shaqaruden, 2020-10-10:
I will try this on Monday. I currently don't have access to any RPI4s

shaqaruden, 2020-10-10:
Isn't 10.1 still Stretch? If it is then that's why it does work because Stretch is not supported by the RPI4. https://en.wikipedia.org/wiki/Raspberry_Pi_OS#Version_history

ealmonte32, 2020-10-14:
@shaqaruden and @bd8392 
This is correct, Sprint 10.1 is a Stretch release, I am wondering why the sprint release notes say it works on RPi4 when it officially shouldn't without some tinkering around with the boot partition, and other stuff..

bd8392, 2020-10-14:
> @shaqaruden and @bd8392
> This is correct, Sprint 10.1 is a Stretch release, I am wondering why the sprint release notes say it works on RPi4 when it officially shouldn't without some tinkering around with the boot partition, and other stuff..

Exactly :-). Are there any plans to upgrade this to buster? Because otherwise it will not work with pi4...also it's in most cases better to run the latest software I think :-).

ealmonte32, 2020-10-18:
> Exactly :-). Are there any plans to upgrade this to buster? Because otherwise it will not work with pi4...also it's in most cases better to run the latest software I think :-).

Yes a buster release image is in the works, but since this is the open-source edition it is mostly worked on by contributors and volunteers so if you need screenly-ose to work on RPi4, you currently have to do the manual install and select the master branch as I see that you did on another post. Thx.

bd8392, 2020-10-18:
I'm ready to help testing :-) Someone helps codeing? :-)

vpetersson, 2020-10-20:
Closed in favor of #1389

closed 2020-10-20 by vpetersson