Screenly/Anthias #1389: Fix Raspberry Pi 4 Model B support

vpetersson, 2020-10-20:
It has become evident that the support for Pi 4 is broken. I'm going to use this ticket to track the remaining efforts to get this across the finishing line.

The status below reflects the current status of a manual installation on Raspbian Buster (Lite) on a Raspberry Pi 4 Model B.

## Here are the outstanding issues:

- [x] Installation (development branch)
- [x] Web interface
- [x] Display of webpages
- [x] Display of images
- [x] Display of videos
   - Unclear if h265 works at the moment
- [ ] Splash screen isn't working

## Tasks once QC/QA has been performed:

- [ ] Create new disk image ([this pull request](https://github.com/Screenly/pi-gen/pull/25) needs to be merged before)
- [ ] Merge master to production

bd8392, 2020-10-20:
Display of webpages was not working on my pi 4 (not properly rendered)

vpetersson, 2020-10-20:
Not what you're testing with @bd8392 but it works fine for me. I have Hacker News loaded on the screen in my lab without issues. If you're talking about some kind of complex page, then that is out of the scope of this issue. This is purely about getting the Pi 4 to work, not about extending the functionality of the browser.

bd8392, 2020-10-20:
> Not sure what you're testing with @bd8392 but it works fine for me. I have [Hacker News](https://news.ycombinator.com/) loaded on the screen in my lab without issues. If you're talking about some kind of complex page, then that is out of the scope of this issue. This is purely about getting the Pi 4 to work, not about extending the functionality of the browser.

Oh okay I tested two wordpress-sites....both were not working but Google worked...

vpetersson, 2020-10-21:
Just a note on the splash screen. As I mentioned to @ealmonte32 [here](https://github.com/Screenly/screenly-ose/issues/1366#issuecomment-713122407), it's currently unclear if the issue is related to the Pi 4 or Buster.

bd8392, 2020-10-21:
Can you deacitvate overscan in boot.txt by default image? Because on 99% of screens it produces unwanted black-bars...

vpetersson, 2020-10-21:
No, this is about splash screen. It has nothing to do with overscan.

bd8392, 2020-10-21:
> No, this is about splash screen. It has nothing to do with overscan.

OHHH okay I'm sorry =). The Black bars really afffect just splash...Just tested one =)

ealmonte32, 2020-10-21:
> > No, this is about splash screen. It has nothing to do with overscan.
> 
> OHHH okay I'm sorry =). The Black bars really afffect just splash...Just tested one =)

You can do this yourself manually, just go into the `/boot/config.txt` file and make the changes there. Read over the comments on that file or even on raspberry pi website, and you will see it is very easy to configure overscan and adjust for your tv/monitor. You can even do it when you have the SD card inserted in your computer, you can edit the config.txt file directly from there.

shaqaruden, 2020-10-22:
Not sure why the splash screen isn't working. I have like 10+ RPI 4 Model Bs running the sprint 10 with no issues

ealmonte32, 2020-10-28:
@vpetersson 
This is regarding the splash screen issue.
I found these instructions [1](https://shop.sb-components.co.uk/blogs/posts/customising-splash-screen-on-your-raspberry-pi) and [2](https://yingtongli.me/blog/2016/12/21/splash.html), and without plymouth or themes, splash screen is working by using framebuffer image viewer `fbi` only.

Raspbian Buster Lite on RPi4.

The `splashscreen.service` file:
```
pi@pi4blk:~ $ cat /etc/systemd/system/splashscreen.service 
[Unit]
Description=Screenly splash screen
DefaultDependencies=no
After=local-fs.target

[Service]
ExecStart=/usr/bin/fbi -d /dev/fb0 --noverbose -a /home/pi/screenlylogo.png
StandardInput=tty
StandardOutput=tty

[Install]
WantedBy=sysinit.target
```

The `/boot/cmdline.txt` file:
```
pi@pi4blk:~ $ cat /boot/cmdline.txt 
console=serial0,115200 console=tty1 root=PARTUUID=53c6e06f-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait logo.nologo quiet
```

vpetersson, 2020-10-28:
So we used to use `fbi` in the early days but we moved away from it  because the upstream Raspbian adopted plymouth. Thus I'm not sure it would make sense to move back.

shaqaruden, 2020-10-28:
From my research using FBI seems to be the common method for Raspbian Lite

vpetersson, 2020-10-28:
Yes, this is true. That's why we started using it. However, it has a number of drawbacks over Plymouth.

ealmonte32, 2020-10-28:
> So we used to use `fbi` in the early days but we moved away from it because the upstream Raspbian adopted plymouth. Thus I'm not sure it would make sense to move back.

Gotcha, so we keep plymouth.
I started troubleshooting, so far I can see all of the default themes showing splash screen after installing `plymouth` and `plymouth-themes` , then running `sudo plymouth-set-default-theme -R themename` , as you know to see the list we use `plymouth-set-default-theme -l` .

I will continue troubleshooting

ealmonte32, 2020-10-29:
Ok so it is working for me:
```
pi@pi4blk:~ $ sudo plymouth-set-default-theme -R screenly
pi@pi4blk:~ $ sudo update-initramfs -u
pi@pi4blk:~ $ sudo reboot
```
..

I compared ansible splashscreen files like screenly.plymouth and screenly.script, and they are identical... 

These are the packages I installed during testing:
```
pi@pi4blk:~ $ dpkg -l | grep -i "plymouth"
ii  libplymouth4:armhf             0.9.4-1.1+rpt2                      armhf        graphical boot animation and logger - shared libraries
ii  plymouth                       0.9.4-1.1+rpt2                      armhf        boot animation, logger and I/O multiplexer
ii  plymouth-label                 0.9.4-1.1+rpt2                      armhf        boot animation, logger and I/O multiplexer - label control
ii  plymouth-theme-breeze          5.14.5-1                            armhf        Breeze theme for Plymouth
ii  plymouth-theme-hamara          1.0                                 all          boot animation, logger and I/O multiplexer - Hamara theme
ii  plymouth-themes                0.9.4-1.1+rpt2                      armhf        boot animation, logger and I/O multiplexer - themes
```

..

This is the `/boot/cmdline.txt`
```
pi@pi4blk:~ $ sudo cat /boot/cmdline.txt 
console=serial0,115200 console=tty1 root=PARTUUID=53c6e06f-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait logo.nologo quiet plymouth.ignore-serial-consoles loglevel=0 splash
```

shaqaruden, 2020-10-29:
did you have to modify cmdline.txt?

ealmonte32, 2020-10-29:
> did you have to modify cmdline.txt?

yes, I believe the original cmdline.txt file didnt have these: 
```
logo.nologo quiet plymouth.ignore-serial-consoles loglevel=0 splash
```
loglevel=0 is not necessary I just left it there during testing, it can be safely removed.

vpetersson, 2020-10-29:
Good job @ealmonte32! So the missing step was just `sudo update-initramfs -u`?

shaqaruden, 2020-10-29:
I attempted this myself but I was not able to get it working. Here's what I did

1. Flash latest Raspberry Pi OS Lite
2. sudo apt update && sudo apt upgrade
3. Installing Screenly from the development branch
4. Install all missing packages from @ealmonte32's list
5. Edit cmdline.txt to reflect what @ealmonte32's is (minus log)
6. sudo plymouth-set-default-theme -R screenly
7. sudo update-initramfs -u
8. sudo reboot

vpetersson, 2020-10-29:
Interesting. And this was on a Pi 4 @shaqaruden?

shaqaruden, 2020-10-29:
@vpetersson yes

ealmonte32, 2020-10-29:
@shaqaruden 
I am going to test again with documenting each step since by the end result there were so many back and forth changes that I did not keep track of every single one because i just wanted to see if it was actually capable of showing the splash screen.

Is your buster image this one: 
http://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-08-24/

shaqaruden, 2020-10-29:
Yes I am using that image

shaqaruden, 2020-10-29:
@ealmonte32 yes I am using that image

ealmonte32, 2020-10-29:
@shaqaruden 

I got it to work again after much trial and error..

i forgot to tell you to install:
`sudo apt install rpd-plym-splash`
but dont do it yet, try this first:

```
pi@raspberrypi4:~ $ sudo plymouth-set-default-theme -r
pi@raspberrypi4:~ $ sudo plymouth-set-default-theme -R screenly
pi@raspberrypi4:~ $ sudo plymouth --show-splash
pi@raspberrypi4:~ $ sudo update-initramfs -u

pi@raspberrypi4:~ $ sudo nano /etc/plymouth/plymouthd.conf 

these are the contents of my conf file: notice the ShowDelay=1

# Administrator customizations go in this file
#[Daemon]
#Theme=futureprototype
#ShowDelay=0
[Daemon]
Theme=screenly
ShowDelay=1
```
Then reboot, let me know how it goes..

ealmonte32, 2020-10-29:
working splash screen:

[image]

vpetersson, 2020-10-29:
Awesome! So now it's just a matter of porting that to Ansible.

ealmonte32, 2020-10-29:
I am trying to reproduce it now though, again starting with fresh install, trying to narrow down the exact commands or whatever makes it work, having a hard time doing it, but it does work as proven twice.

ealmonte32, 2020-10-29:
@shaqaruden it seems like the delay seconds in file `/etc/plymouth/plymouthd.conf` does in fact have some sort of affect on this, I just changed from 0 to 2 and was able to see the splash screen, then changed it to 0, 1, nothing, put 3, see splash.. can you try some of this?

ealmonte32, 2020-10-30:
So I read that we can get log messages from kernel paramters by adding: `plymouth.debug=stream:/run/plymouth.log` to the `/boot/cmdline.txt` file.

After doing that, I saw messages like:
```
[main.c:295]                                 load_settings:Trying to load /etc/plymouth//plymouthd.conf
[ply-key-file.c:171]                       ply_key_file_load_group:trying to load group Daemon
[ply-key-file.c:259]                      ply_key_file_load_groups:key file has no more groups
[main.c:309]                                 load_settings:Checking if /run/plymouth/themes/screenly/screenly.plymouth exists
[main.c:312]                                 load_settings:/run/plymouth/themes/screenly/screenly.plymouth not found, fallbacking to /usr/share/plymouth/themes/
[main.c:325]                                 load_settings:Splash delay is set to 2.000000
[ply-key-file.c:352]                        ply_key_file_get_value:key file does not have entry for key 'DeviceTimeout'
[ply-key-file.c:352]                        ply_key_file_get_value:key file does not have entry for key 'DeviceScale'
[main.c:446]                    find_system_default_splash:System configured theme file is '/usr/share/plymouth/themes/screenly/screenly.plymouth'
[main.c:295]                                 load_settings:Trying to load /run/plymouth/plymouthd.defaults
[ply-key-file.c:83]                        ply_key_file_open_file:Failed to open key file /run/plymouth/plymouthd.defaults: No such file or directory
[main.c:456]              find_distribution_default_splash:failed to load /run/plymouth/plymouthd.defaults, trying /usr/share/plymouth//
[main.c:295]                                 load_settings:Trying to load /usr/share/plymouth//plymouthd.defaults
[ply-key-file.c:171]                       ply_key_file_load_group:trying to load group Daemon
[ply-key-file.c:259]                      ply_key_file_load_groups:key file has no more groups
[main.c:309]                                 load_settings:Checking if /run/plymouth/themes/futureprototype/futureprototype.plymouth exists
[main.c:312]                                 load_settings:/run/plymouth/themes/futureprototype/futureprototype.plymouth not found, fallbacking to /usr/share/plymouth/themes/
[main.c:337]                                 load_settings:Device timeout is set to 5.000000
[ply-key-file.c:352]                        ply_key_file_get_value:key file does not have entry for key 'DeviceScale'
[main.c:463]              find_distribution_default_splash:Distribution default theme file is '/usr/share/plymouth/themes/futureprototype/futureprototype.plymouth'
[main.c:873]           plymouth_should_show_default_splash:checking if plymouth should show default splash
[main.c:901]           plymouth_should_show_default_splash:using default splash because kernel command line has option "splash"
... its a very long log file.. but mostly similar entries with details..
```

..that show weird errors, I think they are build errors like when plymouth was compiled because a lot of those errors are conflicting the directories and what they are trying to load..

the first thing i did was install the plymouth-themes and plymouth-label packages, those made those specific `[ply-utils.c:658] ` and `[main.c:309]                                 load_settings:Checking if /run/plymouth/themes/futureprototype/futureprototype.plymouth exists` errors go away..

but basically for me to know exactly what was the issue i would have to again for a third time start from scratch and have this kernel logging for plymouth on right from the initial steps, then I can probably tell which error and part specifically is causing the splash screen from showing.

vpetersson, 2020-10-30:
Excellent research. Thanks, @ealmonte32

shaqaruden, 2020-11-02:
I will try this again as well and I will document the results along with the steps I took.

shaqaruden, 2020-11-03:
Ok here are all the steps I took this time

1. Flash latest Raspberry Pi OS Lite
2. sudo apt update && sudo apt upgrade
3. Reboot
4. Installing Screenly from the development branch(No Network Management, no full system upgrade, no reboot)
5. `sudo apt install -y libplymouth4:armhf plymouth plymouth-label plymouth-theme-breeze plymouth-theme-hamara plymouth-themes`
6. Edit cmdline.txt to reflect what @ealmonte32's is (minus log)
7. sudo plymouth-set-default-theme -r
8. sudo plymouth-set-default-theme -R Screenly
9. sudo plymouth --show-splash
10. sudo update-initramfs -u
11. Add `ShowDelay=1` to /etc/plymouth/plymouth.conf
*No Splash*
11. Change `ShowDelay=1 => ShowDeplay=2` to /etc/plymouth/plymouth.conf
*See Splash*

shaqaruden, 2020-11-03:
[image]

shaqaruden, 2020-11-03:
[image]

shaqaruden, 2020-11-03:
[image]

Raziers, 2020-11-05:
I followed  @shaqaruden ´s guide. but when i get to this part. I get the no such file or directory problem. What do i do wrong?

pi@raspberrypi:/boot $ sudo plymouth-set-default-theme -R Screenly
pi@raspberrypi:/boot: No such file or directory

pelicanmedia, 2020-11-05:
> I followed @shaqaruden ´s guide. but when i get to this part. I get the no such file or directory problem. What do i do wrong?
> 
> pi@raspberrypi:/boot $ sudo plymouth-set-default-theme -R Screenly
> pi@raspberrypi:/boot: No such file or directory
> /usr/share/plymouth/themes/Screenly/Screenly.plymouth does not exist
> 
> when i check. it seems it down not create the Screenly folder

I had the same issue

vpetersson, 2020-11-05:
@Raziers / @pelicanmedia - Is this *after* you've installed the latest updates to the master branch?

Raziers, 2020-11-05:
> 
> 
> @Raziers / @pelicanmedia - Is this _after_ you've installed the latest updates to the master branch?

Assuming that is what step 5 does, then yes.

vpetersson, 2020-11-05:
No, those are just some dependencies. You need to run `bash <(curl -sL https://www.screenly.io/install-ose.sh)` and select the development release.

Raziers, 2020-11-05:
@vpetersson 

Now i got that worked out

But now it seems i dont have plymouth.conf in /etc/plymouth

I only got plymouthd.conf

is that the same file?

pi@raspberrypi:/etc/plymouth $ ls
plymouthd.conf

ealmonte32, 2020-11-06:
Guys, plymouth themes seem to be case sensitive, so you will get an error if you do:
`sudo plymouth-set-default-theme -R Screenly` because Screenly should be lowercase like `screenly`

@Raziers , `/etc/plymouth/plymouthd.conf` is the correct file to have.
type `cat /etc/plymouth/plymouthd.conf` and show me what comes up.

Raziers, 2020-11-09:
Did something get updated?  I ran through @shaqaruden guide again, but only got to installing Screenly from the dev branch. After that, it seems to work fine now on the raspberry 4

ealmonte32, 2020-11-10:
Well yes many many changes/commits are on the dev/master branch vs the production branch.

MichaliQQ, 2020-11-17:
I am testing screenly ose and I some problems. I have raspberry 4 model B 4GB, when I install on raspbian lite by bash command and I choose development release the installation is successful but when I want display the page i just see white screen. I saw other issues and i know that i have to choose experimental release but I got error on task `[screenly : Run docker-compose]` . Are there any plans to fix this problem?

ealmonte32, 2020-11-17:
@MichaliQQ , do not use experimental for this purpose, development branch should be the one.
Can you explain more about the website you are adding? I have Pi4B here and can test this, it could simply be the website/uzbl related.

MichaliQQ, 2020-11-18:
@ealmonte32 Website just uses socket.io and jquery. I cant show you because site is created locally. Screenly only load from this website image of logo (the rest of the page is white), in screenly browser console returns error in index with 4 line `let socket = io();`

Monomaniak-Tek, 2020-11-18:
I tried varius methods of installation with Pi 4 (on this topic and little more). Unfortunately uzbl seem not work very well (no javascript, no css?).
Is it possible to change browse or install webkit1 ? 
Thanks !

ealmonte32, 2020-11-18:
> @ealmonte32 Website just uses socket.io and jquery. I cant show you because site is created locally. Screenly only load from this website image of logo (the rest of the page is white), in screenly browser console returns error in index with 4 line `let socket = io();`

Oh ok.. well, so based on this info then yes the reason for webpage not displayed properly is UZBL browser, which is being replaced on the next version of Screenly.

ealmonte32, 2020-11-18:
Hi @Monomaniak-Tek 

> I tried varius methods of installation with Pi 4 (with this topic and little more). Unfortunately, in the best way, uzbl seem not workink very well (no javascript, no css?).
> Is it possible to change browser or install webkit1 ?

As mentioned to other user above and in other posts, UZBL has been the culprit of all these issues with displaying of web assets incorrectly and it is being replaced in the next version of Screenly.
The new browser should have WebKit built-in as part of the Qt Web View browser app that will be used.

Since UZBL has not been updated in many years (in the official debian packages repo) , you would need to compile it manually and install it yourself (instructions are on their github repo). This might allow more web pages to work but in the end, the browser is not updated frequently so for example the last version they officially released as source is: `v0.9.1
@keis keis released this on Oct 27, 2016` , so as you can see, 4 years old, which is still better than the current version on raspbian apt sources of `0.0.0~git.20120514-1.2+b18` , which is from 2012...
If i can compile this newer version and have it working without doing so many workarounds, I will post here what to do, but I dont want to promise anything.

> If anyone have a nice installation of scrennly OSE on Pi 4, is it possible to share image file of SD card?

But installing it on Pi 4 does not change the browser outcome, so I am not sure what you are asking for an image file.. this is not good, you should never install non-official images of entire OS+applications on your device.. Do other webpage assets work on your Pi4 with latest version of Screenly OSE (dev/master branch)?

vpetersson, 2020-11-23:
Update from last week:

* I've been doing [heavy refactoring](https://github.com/Screenly/screenly-ose/pull/1409) on the experimental branch to make it more manageable going forward. Some things I did in the update:
   * Consolidate Balena and non-Balena `docker-compose.yml` files (with overrides)
   * Move WebViewer into main repo
   * Introduced a base image to reduce space (much more can be done here)
   * Introduced multi-stage builds for viewer
   * Documented and verified how QT Base is built
   * Refactored almost all Dockerfiles
   * Scripted docker build (and prepared it for GitHub Actions)
   * ...and probably a lot more that I've forgotten about already :)

Other outstanding issues:

* Fix Raspberry Pi 4 Model B support
  * As things stand right now, the viewer component does not work on Pi 4 on the Experimental branch. This is possibly due to lack of Pi 4 support in QT 5.9.
* Fix IP on intro page
  * Because we're running inside a docker container, we cannot detect the IP of the host.
* Fix SSL support
  * With the refactoring of the Nginx code, I broke the SSL support. This should be relatively easy to fix.
* Validate that the WiFi Hotspot code works

ealmonte32, 2020-11-25:
@vpetersson 
Just to confirm, to work on some of the outstanding issues we have to do the troubleshooting under the experimental branch + pi3?

vpetersson, 2020-11-25:
Yes, exactly. I'm working with @rusko124 to get the Experimental branch to support Pi 4. However, in the meantime, it would be helpful to kick the tires on a Pi 3 as we've pushed a lot of changes to the branch recently.

Monomaniak-Tek, 2020-11-26:
I (with a friend..) can solve uzbl display on Pi4 for me :
If this can help anyone...
What to do:
--------------------------------------------------------------------
1) Download uzbl-next:
        git clone --depth=1 https://github.com/uzbl/uzbl -b next
        cd uzbl

2) Install the dependancies:
        apt-get install libwebkit2gtk-4.0-dev libgnutls28-dev

3) Build uzbl-core:
        export CFLAGS=" `pkg-config --cflags --libs webkit2gtk-4.0 glib-2.0 gtk+-3.0 libsoup-2.4 gnutls`"
        make uzbl-core

4) Install the binary:
        mv /usr/bin/uzbl-core{,.orig} 
        cp uzbl-core /usr/bin/
        
5) Fix the last line of /usr/bin/uzbl-browser, replacing "@" by
127.0.0.1:8080/splash-page --config - --print-events, so the line
becomes: exec uzbl-core 127.0.0.1:8080/splash-page --config -
--print-events ${config_file:+--config "$config_file"} --connect-socket
$DAEMON_SOCKET

6) Restart the daemon (or reboot):
        systemctl restart screenly-viewer.service

vpetersson, 2020-11-27:
@Monomaniak-Tek Thanks, but we've already moved away from `uzbl` in the Experimental branch that we hope to get out in the coming weeks. Uzbl has more or less been abandoned upstream so we're moving to a more modern browser based on WebEngine.

javad87, 2020-11-30:
> Guys, plymouth themes seem to be case sensitive, so you will get an error if you do:
> `sudo plymouth-set-default-theme -R Screenly` because Screenly should be lowercase like `screenly`
> you can check the exact theme name you need to write by typing: `sudo plymouth-set-default-theme -l`
> 
> @Raziers , `/etc/plymouth/plymouthd.conf` is the correct file to have.
> type `cat /etc/plymouth/plymouthd.conf` and show me what comes up.

For this Error:  "no such file or directory problem" when executing :sudo plymouth-set-default-theme -R Screenly,
I guess the problem is not with case sensitivness, I did both ways but still had the Error, after checking folders in /etc/systemd/system I could not find the screenly related services and also in /usr/.... path also did not find screenly folder and related files, although screenly installation via bash script provided, finished successfully and it asked for reboot ....

I changed pi password to default password of pi which is "raspberry" and run the installation script again, this time  services and screenly files installed correctly and not receiving the error....

So, if you changed pi default password, return it back to raspberry and run bash <(curl -sL https://www.screenly.io/install-ose.sh) again...

consider that during installation u will be asked to enter pi password which u have to enter default one and also later for security reason asking to change it...I did not change it u can change it later ...first check if it boots correctly...

U can change password by raspberry config menu: sudo raspi-config

Anyone knows how can I move from splash screen to main X window os environment (rasbian) or opening shell or new tty?
I want to turn my wifi on/off or doing some configuration....
by the way in screenly splash screen when I connected vi WiFi it says can not resolve IP...I found my IP via my access point admin pannel..

Thanks guys for ur help, hope all be well and healthy and wealthy

Monomaniak-Tek, 2020-11-30:
> @Monomaniak-Tek Thanks, but we've already moved away from `uzbl` in the Experimental branch that we hope to get out in the coming weeks. Uzbl has more or less been abandoned upstream so we're moving to a more modern browser based on WebEngine.

Ok, but on Pi 4, I only can install developement branch  (as explain by shaqaruden) however PI 4 crash with Experimental branch).

Monomaniak-Tek, 2020-11-30:
On Pi 4 and Development branch

- [X]  web + js  (with webkit2) 

- [X]  Vidéo with audio

- [X] Photo (I had to modify the "viewer.py" file -> I don't know if i can share the file and where I can put it because I am a new user?)

javad87, 2020-12-01:
I did the same things that mentioned above, everything is ok except when it comes to showing video assets, screen will be black for a moment till moving to a next asset which is a pic and it shows pic....

I increased GPU to 256, file is around 2 minutes and 13MB , mp4 codec H264, omxplayer Version: 20190723+gitf543a0d-1,I use buster image, Rapi 4 model B

What should I do to solve black screen problem?

screenly log:

Dec 01 17:41:57 raspberrypi python[3119]: Displaying video /home/pi/screenly_assets/9f0e815ed52840199fb48dcebd2722e1 for 1
Dec 01 17:41:57 raspberrypi python[3119]: current url is file:///tmp/screenly_html/black_page.html
Dec 01 17:44:50 raspberrypi python[3119]: Last update: 2020-12-01 10:18:41.020324
Dec 01 17:44:50 raspberrypi python[3119]: get_next_asset
Dec 01 17:44:50 raspberrypi python[3119]: refresh_playlist
Dec 01 17:44:50 raspberrypi python[3119]: refresh: counter: (0) deadline (2020-11-29 22:07:00) timecur (2020-12-01 14:14:5
Dec 01 17:44:50 raspberrypi python[3119]: update_playlist
Dec 01 17:44:50 raspberrypi python[3119]: Generating asset-list...
Dec 01 17:44:50 raspberrypi python[3119]: generate_asset_list deadline: 2020-11-29 22:07:00
Dec 01 17:44:50 raspberrypi python[3119]: get_next_asset after refresh
Dec 01 17:44:50 raspberrypi python[3119]: get_next_asset counter 0 returning asset 1 of 4
Dec 01 17:44:50 raspberrypi python[3119]: Showing asset Pasargad.jpg (image)
Dec 01 17:44:50 raspberrypi python[3119]: Asset URI /home/pi/screenly_assets/55de0f08b0ff465e9441164d3dff66dc
Dec 01 17:44:50 raspberrypi python[3119]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Dec 01 17:44:50 raspberrypi python[3119]: Sleeping for 10
Dec 01 17:45:00 raspberrypi python[3119]: Last update: 2020-12-01 10:18:41.020324
Dec 01 17:45:00 raspberrypi python[3119]: get_next_asset
Dec 01 17:45:00 raspberrypi python[3119]: refresh_playlist
Dec 01 17:45:00 raspberrypi python[3119]: refresh: counter: (0) deadline (2020-11-29 22:07:00) timecur (2020-12-01 14:15:0
Dec 01 17:45:00 raspberrypi python[3119]: update_playlist
Dec 01 17:45:00 raspberrypi python[3119]: Generating asset-list...
Dec 01 17:45:00 raspberrypi python[3119]: generate_asset_list deadline: 2020-11-29 22:07:00
Dec 01 17:45:00 raspberrypi python[3119]: get_next_asset after refresh
Dec 01 17:45:00 raspberrypi python[3119]: get_next_asset counter 0 returning asset 2 of 4
Dec 01 17:45:00 raspberrypi python[3119]: Showing asset Hafez.jpg (image)
Dec 01 17:45:00 raspberrypi python[3119]: Asset URI /home/pi/screenly_assets/4456bd4409aa4b6f8f114a11babe420b
Dec 01 17:45:00 raspberrypi python[3119]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Dec 01 17:45:00 raspberrypi python[3119]: Sleeping for 10
Dec 01 17:45:10 raspberrypi python[3119]: Last update: 2020-12-01 10:18:41.020324
Dec 01 17:45:10 raspberrypi python[3119]: get_next_asset
Dec 01 17:45:10 raspberrypi python[3119]: refresh_playlist
Dec 01 17:45:10 raspberrypi python[3119]: refresh: counter: (0) deadline (2020-11-29 22:07:00) timecur (2020-12-01 14:15:1
Dec 01 17:45:10 raspberrypi python[3119]: update_playlist
Dec 01 17:45:10 raspberrypi python[3119]: Generating asset-list...
Dec 01 17:45:10 raspberrypi python[3119]: generate_asset_list deadline: 2020-11-29 22:07:00
Dec 01 17:45:10 raspberrypi python[3119]: get_next_asset after refresh
Dec 01 17:45:10 raspberrypi python[3119]: get_next_asset counter 0 returning asset 3 of 4
Dec 01 17:45:10 raspberrypi python[3119]: Showing asset h264-low.mp4 (video)
Dec 01 17:45:10 raspberrypi python[3119]: Asset URI /home/pi/screenly_assets/902596d7542f4de59a5d929307e97e5e
Dec 01 17:45:10 raspberrypi python[3119]: Displaying video /home/pi/screenly_assets/902596d7542f4de59a5d929307e97e5e for 1
Dec 01 17:45:10 raspberrypi python[3119]: current url is file:///tmp/screenly_html/black_page.html

[6 more replies]

closed 2020-12-02 by vpetersson