Screenly/Anthias #1405: log files are very large

lArtl, 2020-11-03:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

Log files clogged all the space on the Sd card. For 6 days of work, 10gb files have accumulated
OSE worked fine for 6 days.
I had to delete all log files from the var / log directory.
Is it normal that files are being created?
A paragraph or two about the issue you're experiencing.

#### Environment

* Raspberry Pi Hardware Version: pi3
* Raspberry Pi Network Setup: (Ethernet)
* Screenly OSE Version:Sprint 10.1 Release

lArtl, 2020-11-03:
[image]

vpetersson, 2020-11-03:
There must be something wrong with the setup if you're getting that big log files. Could you look through the logs to see if there is any particular error that is repeated for instance?

lArtl, 2020-11-03:
Tell me where I can find the logs you are interested in?

vpetersson, 2020-11-03:
I'm not sure. You mentioned that there are 10G of logs, so i guess start poking around in `/var/logs` to see what files are growing.

lArtl, 2020-11-03:
Unfortunately, I deleted all large logs. And I only know their names.
The largest log file was called kern.log1 syslog syslog.1 kern.log messages.1 demon.log.1

lArtl, 2020-11-03:
2020-10-27 12:37:01 status unpacked base-files:armhf 9.9+rpi1+deb9u13
2020-10-27 12:37:01 status unpacked base-files:armhf 9.9+rpi1+deb9u13
2020-10-27 12:37:01 status unpacked base-files:armhf 9.9+rpi1+deb9u13
2020-10-27 12:37:01 status half-configured base-files:armhf 9.9+rpi1+deb9u13
2020-10-27 12:37:01 status installed base-files:armhf 9.9+rpi1+deb9u13
2020-10-27 12:37:02 startup archives unpack
2020-10-27 12:37:02 upgrade e2fslibs:armhf 1.43.4-2+deb9u1 1.43.4-2+deb9u2
2020-10-27 12:37:02 status triggers-pending libc-bin:armhf 2.24-11+deb9u4
2020-10-27 12:37:02 status half-configured e2fslibs:armhf 1.43.4-2+deb9u1
2020-10-27 12:37:02 status unpacked e2fslibs:armhf 1.43.4-2+deb9u1
2020-10-27 12:37:02 status half-installed e2fslibs:armhf 1.43.4-2+deb9u1
2020-10-27 12:37:02 status half-installed e2fslibs:armhf 1.43.4-2+deb9u1
2020-10-27 12:37:02 status unpacked e2fslibs:armhf 1.43.4-2+deb9u2
2020-10-27 12:37:02 status unpacked e2fslibs:armhf 1.43.4-2+deb9u2
2020-10-27 12:37:03 startup packages configure
2020-10-27 12:37:03 configure e2fslibs:armhf 1.43.4-2+deb9u2 <none>
2020-10-27 12:37:03 status unpacked e2fslibs:armhf 1.43.4-2+deb9u2
2020-10-27 12:37:03 status half-configured e2fslibs:armhf 1.43.4-2+deb9u2
2020-10-27 12:37:03 status installed e2fslibs:armhf 1.43.4-2+deb9u2
2020-10-27 12:37:03 startup archives unpack
2020-10-27 12:37:03 upgrade e2fsprogs:armhf 1.43.4-2+deb9u1 1.43.4-2+deb9u2
2020-10-27 12:37:03 status half-configured e2fsprogs:armhf 1.43.4-2+deb9u1
2020-10-27 12:37:03 status unpacked e2fsprogs:armhf 1.43.4-2+deb9u1
2020-10-27 12:37:03 status half-installed e2fsprogs:armhf 1.43.4-2+deb9u1
2020-10-27 12:37:04 status half-installed e2fsprogs:armhf 1.43.4-2+deb9u1
2020-10-27 12:37:04 status unpacked e2fsprogs:armhf 1.43.4-2+deb9u2
2020-10-27 12:37:04 status unpacked e2fsprogs:armhf 1.43.4-2+deb9u2
2020-10-27 12:37:04 startup packages configure
2020-10-27 12:37:04 configure e2fsprogs:armhf 1.43.4-2+deb9u2 <none>
2020-10-27 12:37:04 status unpacked e2fsprogs:armhf 1.43.4-2+deb9u2
2020-10-27 12:37:04 status unpacked e2fsprogs:armhf 1.43.4-2+deb9u2
2020-10-27 12:37:04 status half-configured e2fsprogs:armhf 1.43.4-2+deb9u2
2020-10-27 12:37:04 status installed e2fsprogs:armhf 1.43.4-2+deb9u2
2020-10-27 12:37:05 status triggers-pending initramfs-tools:all 0.130
2020-10-27 12:37:05 startup archives unpack
2020-10-27 12:37:05 upgrade libperl5.24:armhf 5.24.1-3+deb9u6 5.24.1-3+deb9u7
2020-10-27 12:37:05 status half-configured libperl5.24:armhf 5.24.1-3+deb9u6
2020-10-27 12:37:05 status unpacked libperl5.24:armhf 5.24.1-3+deb9u6
2020-10-27 12:37:05 status half-installed libperl5.24:armhf 5.24.1-3+deb9u6
2020-10-27 12:37:10 status half-installed libperl5.24:armhf 5.24.1-3+deb9u6
2020-10-27 12:37:10 status unpacked libperl5.24:armhf 5.24.1-3+deb9u7
2020-10-27 12:37:10 status unpacked libperl5.24:armhf 5.24.1-3+deb9u7
2020-10-27 12:37:10 upgrade perl:armhf 5.24.1-3+deb9u6 5.24.1-3+deb9u7
2020-10-27 12:37:10 status half-configured perl:armhf 5.24.1-3+deb9u6
2020-10-27 12:37:10 status unpacked perl:armhf 5.24.1-3+deb9u6
2020-10-27 12:37:10 status half-installed perl:armhf 5.24.1-3+deb9u6
2020-10-27 12:37:11 status half-installed perl:armhf 5.24.1-3+deb9u6
2020-10-27 12:37:11 status unpacked perl:armhf 5.24.1-3+deb9u7
2020-10-27 12:37:11 status unpacked perl:armhf 5.24.1-3+deb9u7
2020-10-27 12:37:11 upgrade perl-base:armhf 5.24.1-3+deb9u6 5.24.1-3+deb9u7
2020-10-27 12:37:11 status half-configured perl-base:armhf 5.24.1-3+deb9u6
2020-10-27 12:37:11 status unpacked perl-base:armhf 5.24.1-3+deb9u6
2020-10-27 12:37:11 status half-installed perl-base:armhf 5.24.1-3+deb9u6
2020-10-27 12:37:13 status half-installed perl-base:armhf 5.24.1-3+deb9u6
2020-10-27 12:37:13 status unpacked perl-base:armhf 5.24.1-3+deb9u7
2020-10-27 12:37:14 status unpacked perl-base:armhf 5.24.1-3+deb9u7
2020-10-27 12:37:14 startup packages configure

vpetersson, 2020-11-03:
Hm, my gut feeling is that it's due to a busted SD card. That can cause the logs to spew a lot of errors. Try rotating it out to see if that solves the problem.

lArtl, 2020-11-05:
greetings to all. After two days of showing, Screenly was able to clean up the space inside the SD itself. After my cleaning the logs it was 8gb space, Screenly now made 11gb. If you look at journalctl, there I saw Scheduler: Sending due task cleanup (server.cleanup)
Sending due task cleanup (server.cleanup_usb_assets)
These commands clean up unnecessary files, right?

lArtl, 2020-11-05:
Most likely my first problem is due to the wrong time on the server.
This script (cleanup) is executed every hour.
After rebooting Screenly, the time became incorrect, even though I set my time zone. It would be cool if in the OSE version on the web you can configure the time and time zone.

closed 2020-11-09 by lArtl